esphome/issues #3351: Raspberry Pi UART `/dev/ttyAMA0` is not available for flashing

dokmic, 2022-06-07:
### The problem

The UART port is not available for installation when running ESPHome dashboard standalone on Raspberry Pi with Raspberry Pi OS (not as a HomeAssistant addon).

## Reproduction Path
1) Run ESPHome using Docker on Raspberry Pi OS.
2) Create a device.
3) Click `Install`.
4) Pick `Plug into the computer running ESPHome Dashboard`.

## Expected
There should be `/dev/ttyAMA0` available for flashing.

## Actual
There are no ports available for flashing.

## Root Cause
The problem is in the [`get_serial_ports`](https://github.com/esphome/esphome/blob/dev/esphome/util.py#L286) from `util.py` implementation. If we try to run the following command:
```bash
$ python3 -c "from serial.tools.list_ports import comports; print([(port, info) for (port, description, info) in comports(include_links=True)])"
```
The output will be:
```
[('/dev/ttyAMA0', 'fe201000.serial')]
```
It won't pass the following [condition](https://github.com/esphome/esphome/blob/dev/esphome/util.py#L293):
```python
if "VID:PID" in info:
    result.append(SerialPort(path=port, description=desc))
```
Since the string `fe201000.serial` has no `VID:PID` in it.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

OttoWinter, 2022-06-14:
Indeed, that's intentionally disabled because it can get confusing. For example my computer has a serial port somewhere inside (and is not exposed via USB, so has no VID). Showing that runs the risk of confusing the user since for 99% of cases it's not what they want to use. I don't think there's a good solution here for that problem that doesn't sacrifice the usability for the 99% cases.

dokmic, 2022-06-17:
@OttoWinter But in the [code](https://github.com/esphome/esphome/blob/dev/esphome/dashboard/dashboard.py#L358-L359) calling the `get_serial_ports` function, that port is expected. Raspberry Pi is quite a popular platform, and there is plenty of information online on how to flash an ESP device via GPIO, so this seems to me like a bug.

What if we introduce an environment variable like `ESPHOME_DASHBOARD_NON_USB_SERIAL_PORTS` that will override this `VID:PID` check? This will be disabled by default so that it won't be a breaking change.

I am happy to help and open a PR if we find an acceptable solution.

OttoWinter, 2022-06-20:
> @OttoWinter But in the [code](https://github.com/esphome/esphome/blob/dev/esphome/dashboard/dashboard.py?rgh-link-date=2022-06-17T20%3A02%3A14Z#L358-L359) calling the get_serial_ports function, that port is expected.

Ah yes, I remember. That was there first until the exact problem I noted happened, and I got a bunch of reports from confused users (most often a user would think their USB cable was plugged in correctly, since an option was seen on the frontend; in reality the USB was power-only and the option was there all the time)

Ok yeah an env variable should also do the trick (however, it's hard to document something like that in a way that users who actually want to use it can also find it).

closed 2022-11-03 by github-actions[bot]

QGB, 2023-03-02:
│··