esphome/issues #3353: Nextion integration throws an error if names have underscores

rmarskell, 2022-06-09:
### The problem

Names with underscores are supported through the Nextion Editor when creating HMI/TFT files. However, when you try to use them with the ESPHome, it says you're not allowed to use the underscore character:
[image]

This is because of the validation code here:
https://github.com/esphome/esphome/blob/3a9ab50dd28ff72d61118fecb9d0465add778b36/esphome/components/nextion/base_component.py#L33-L43

I can submit a PR for this, but I was wondering if there was a reason this wasn't included originally that I'm not aware of. I see that the original file was added by @SenexCrenshaw so perhaps they can provide some insight on the reasoning.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP32

### Board

NSPanel

### Example YAML snippet

```yaml
- platform: nextion
  name: Save Button
  page_id: 0
  component_name: save_button
```

edwardtfn, 2024-05-07:
This was probably fixed by #5979 as I cannot duplicate the issue.

rmarskell, 2024-05-08:
That does look like it solved the issue. Thanks

closed 2024-05-08 by rmarskell