esphome/issues #3357: Adafruit Feather ESP32 V2 not accepted as board type

critchards, 2022-06-10:
### The problem

Im trying to commission a new Adafruit Feather ESP32 V2. It adopts fine using a generic ESP32DEV board type but when I attempt to set the board to `adafruit_feather_esp32_v2` from the platformio registry I get a `target_platform` error. 

Im very new at this so I may be missing something obvious but im guesing something didnt get updated when the feather v2 was added.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.7

### What platform are you using?

ESP32

### Board

adafruit_feather_esp32_v2

### Component causing the issue

platform

### Example YAML snippet

```yaml
esphome:
  name: "weather-feather"

esp32:
  board: adafruit_feather_esp32_v2
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/weather-feather.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 909, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 972, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 827, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 815, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 742, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 124, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 498, in run
    validated = schema(self.conf)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/esphome/esphome/voluptuous_schema.py", line 118, in validate_mapping
    not isinstance(key.default, vol.Undefined)
  File "/esphome/esphome/config_validation.py", line 1445, in default
    if CORE.is_esp8266:
  File "/esphome/esphome/core/__init__.py", line 590, in is_esp8266
    return self.target_platform == "esp8266"
  File "/esphome/esphome/core/__init__.py", line 586, in target_platform
    return self.data[KEY_CORE][KEY_TARGET_PLATFORM]
KeyError: 'target_platform'
```

majuss, 2022-06-11:
Try different ESP32 boards. Spoiler, almost every board is broken that way... fresh venv on macOS or Debian. Can't compile any ESP32 board.

mmakaay, 2022-06-17:
@majuss You error doesn't really seem related to the issue report.
It looks like you're managing a local ESPHome installation.
Did you run `./script/setup` after upgrading ESPHome?

As for the original report, on the [platformio Adafruit Feather ESP32 v2 page](https://docs.platformio.org/en/latest/boards/espressif32/adafruit_feather_esp32_v2.html), the board id is `adafruit_feather_esp32_v2`, so that bit ought to be right.

I will try to reproduce the issue and track it down when I can.

majuss, 2022-06-17:
I did:
```
python3 -m venv venv
source venv/bin/activate
pip install esphome
esphome compile example.yaml
```

mmakaay, 2022-06-17:
@majuss Then please create a new issue for it, since it really doesn't look like it's related to the original issue here. Include a configuration that fails for you so we can try to reproduce the issue.

@critchards I can reproduce your issue locally, so I'll see what I can find.

devoh747, 2022-06-18:
I upgraded my docker container to 2022.6.1 and still have the same errors as @critchards .  It does not seem to be effecting all boards as if I changed the board to esp32dev that compiles. It's interesting that the editor will pick it up as an error.. so it is actually erroring before compiling. The editor shows the red "x" on the first line and says 'target_platform'.
[image]

mmakaay, 2022-06-18:
The list of boards in ESPHome needs updating. I already did that on my dev environment, but now I'm getting a follow-up error from it seems platformio about the board not being valid. 

I tried upgrading platformio to see if that was the issue (it's a relatively new board), but it broke my dev env, so setting up a new one.

mmakaay, 2022-06-18:
BTW, this is the error that I see on build now:

[image]

devoh747, 2022-06-19:
but isn;t it saying it's using PIO core 5.2.5 and NOT 6.0.2 ?  Would it
even have been defined back in 5.2.5?  It is a new board.

mmakaay, 2022-06-19:
No, that's why it' was not building for me, but upgrading to core 6 failed for me so couldn't test that.
This screenshot was to show that the updated ESPHome code did fix the primary error.

I can commit the code so far if you want to give it a spin.

Next to the list update possibly fixing the issue, I also am looking into why the ugly error comes up to begin with. When using an unknown board, it should cleany notify about that instead of dumping a stack trace

mmakaay, 2022-06-19:
Got my platformio working, there were some file permission issues on my dev host. But there seem to be multiple follow-up issues.

When trying to build the code in the ESPHome build dir manually, I get:

[image]

So that's platformio 6.0.2, complaining about not knowing the board.
But another problem is that ESPHome itself is not yet using 6.0.2, but 5.2.5. I quickly tried updating the requirements for ESPHome to make it use 6.0.2, but the platformio API changed. Because of that, ESPHome can't talk to it anymore, and breaks on:

```
ModuleNotFoundError: No module named 'platformio.commands.run
```

So for this to work:
- platformio ought to work to begin with
- ESPHome must be taught how to talk to platformio API v6.x.x

mmakaay, 2022-06-19:
Ah, the Feather ESP32 V2 board was added in [platform-espressif32 v4.3.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.3.0)
ESPHome is currently using v3.5.0.

After tweaking some things manually in the build dir and kicking off a build, things looked promising, but in the end, this resulted in some low level build errors that require changes in the ESPHome code to get the build working.

My conclusion:
This is not going to be a quick fix.
To build code for this device, using a different board is the best option for now. A generic `esp32dev` would work, like you already found out. However, when you require the use of the board's PSRAM, then `esp32cam` might be a feasible option too.

mmakaay, 2022-06-19:
> Still the same with 6.0, everything is broken. Not only ESP32 but also ESP8266 is broken af

@majuss I was able to reproduce your issue. It's an issue with platformio, and I found a [community post about this](https://community.platformio.org/t/init-unexpected-argument-uri-when-building-project/27390/18)

I could fix things by doing:
```
$ pip uninstall platformio   (removes the v6 version)
$ script/setup               (from the ESPHome code, installs the v5 version)
$ rm -fR ~/.platformio       (cleans up any resources)
```
After this, I could start a compilation using `esphome compile`, platformio downloaded the required resources and the compilation was successful.

majuss, 2022-06-19:
> Got my platformio working, there were some file permission issues on my dev host. But there seem to be multiple follow-up issues.
>
> When trying to build the code in the ESPHome build dir manually, I [...]

Just alter line 23 in `lib/python3.9/site-packages/esphome/platformio_api.py` and change "run" in "lib". Everything works then.

devoh747, 2022-06-20:
> Ah, the Feather ESP32 V2 board was added in [platform-espressif32 v4.3.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.3.0) ESPHome is currently using v3.5.0.
>
> After tweaking [...]

Using the esp32dev board definition instead of adafruit_feather_esp32_v2 gives other issues as it does not allow i2c SCL to use ping 20

majuss, 2022-06-21:
@mmakaay I'am really sorry for my tone, I was stressed out that day. Yes I'am using esphome in a local python venv. I had various problems and could't compile anything with version 2022.4, 2022.5 and now 2022.6.

So how to get ESPhome working:
```
python3 -m venv venv
source venv/bin/activate
pip install esphome
pio upgrade
sed -i 's/platformio.commands.run/platformio.commands.lib/g' venv/lib/python3.9/site-packages/esphome/platformio_api.py
```

Then ESPhome is working as expected.

mmakaay, 2022-06-21:
No problem, we all get stressed out at timen 😉

You're using platformio 6 then, right? ESPHome is on version 5.2.5, as configured in its dependencies. I think that on a clean system, this recipe would fail therefore, but cool that you got it to work and that the platformio API didn't dramatically change.

majuss, 2022-06-21:
When I try it with 5.2.5 I get the inital error in post #1 for any board I try.

critchards, 2022-07-01:
> Ah, the Feather ESP32 V2 board was added in [platform-espressif32 v4.3.0](https://github.com/platformio/platform-espressif32/releases/tag/v4.3.0) ESPHome is currently using v3.5.0.
>
> After tweaking [...]

Thank you @mmakaay. As @devoh747 pointed out this raises another issue of GPIO20 being locked out before compile. My use case is a series of workshops for new to hardware participants. We will be using the STEMMAQT connector on the Feather V2 to connect to a BME688 sensor and I was hoping to introduce the participants to Home Assistant and ESPHome. Can you guess which pin the STEMMAQT SCL line is? ;) Is there anyway to bypass the GPIO20 restriction?

majuss, 2022-07-04:
@mmakaay I'am still getting the URI error. All my steps:

```
python3 -m venv esphomevenv
source esphomevenv/bin/activate
pip install esphome
esphome compile bettlich.yaml
```

My yaml:
```
substitutions:
    devicename: "bettlicht"
  
  esphome:
    name: ${devicename}
    platform: ESP8266
    board: d1
  
  web_server:
    port: 80
  
  
  light:
    - platform: neopixelbus
      type: BRG
      variant: WS2811
      pin: 14
      num_leds: 294
      name: "Bettlicht"
      effects:
        - addressable_rainbow:
            name: rainbow
            speed: 5
            width: 50

  api:
    password: ""
  
  ota:
    password: ""
  
  wifi:
    ssid: ""
    password: ""

  
  captive_portal:
  
```

mmakaay, 2022-07-04:
That issue is not really related to this issue report, so please keep it a bit clean in here.
I gave a [suggestion in this post](https://github.com/esphome/issues/issues/3357#issuecomment-1159673190). You didn't apply that fix option?

majuss, 2022-07-04:
What is this line: "script/setup"? Which setup from where?

mmakaay, 2022-07-04:
That's the setup script from the ESPHome directory. That script must be run for ESPHome to install its dependencies. When using pip to install, that is already taken care of. For your case, cleaning out the `.platformio` directory in your homedir is the important bit.

majuss, 2022-07-05:
```shell
rm -rf ~/.platformio
python3 -m venv venv
source venv/bin/activate
pip install esphome
esphome compile example.yaml
```

I still get:

[image]

mmakaay, 2022-07-05:
That's bad luck. I could solve it like that, after downgrading to platformio 5.2.5.
But please, I [hinted at it before](https://github.com/esphome/issues/issues/3357#issuecomment-1159141355), this issue you have is not on topic for the issue that you are running into. If you want to discuss further on this, please create a separate issue for it.

devoh747, 2022-07-13:
I am still getting the original error using the docker images from [2022.7.0-dev20220713] .  I was under the impression from other pulls that dev had it fixed and it would be pushed to stable for 2022.08 release.  

See https://github.com/esphome/esphome/pull/3566 "Bump platformio to 6.0.2"

mmakaay, 2022-07-14:
That is only a bump of the platformio version to 6.0.2.
Next to that, the ESP32 boards list needs updating, to have the board recognized by the ESPHome compiler.
I have worked on that, but saw a [PR #3564](https://github.com/esphome/esphome/pull/3564)  by @jesserockz that did the same thing, so I dropped working on that. This PR has not yet been merged, so that's why it doesn't work using dev yet.

devoh747, 2022-07-21:
I seem to be having some success by specifying the variant manually..

`esphome:
  name: weather-eps32

esp32:
  board: adafruit_feather_esp32_v2
  variant: esp32
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.0.0`

Didn't error on me using the esphome docker image for dev.  Now to see if I can build my weather station completely and flash it.

kj4dxk, 2022-08-15:
I am working with the adafruit qt py and this worked for me to get around the error:

esp32:
  board: adafruit_qtpy_esp32
  variant: esp32
  framework:
    type: arduino

However, the neopixel does not work:

output:
  - platform: gpio
    id: "neopixel_power"
    pin: GPIO8 # NEOPIXEL_POWER
    inverted: true

  This pin cannot be used on ESP32s and is already used by the flash interface (function: Flash Data 1).
  pin: GPIO8

kj4dxk, 2022-08-15:
esp32:
board: adafruit_qtpy_esp32
variant: esp32
framework:
type: arduino

Scratch that, I get this error: Error: Unknown board ID 'adafruit_qtpy_esp32'

closed 2022-12-14 by ssieb