esphome/issues #3358: DS18B20 temperature validity

Benoit3, 2022-06-10:
### The problem

With some conditions of Pull Up (10k, yes it is to much) and wire length (25 m), esphome read all bits as 0. The checksum is still valid , so the temperature send to HA is 0°C and not NAN

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32-EVB

### Component causing the issue

dallas

### Anything in the logs that might be useful for us?

```txt
19:42:58][V][mqtt:415]: Publish(topic='cesi/sensor/sensor_2/state' payload='0.0' retain=1)
[19:42:58][VV][scheduler:196]: Running timeout '0xb200000017ab4f28' with interval=750 last_execution=827051 (now=827835)
[19:42:58][VV][dallas.sensor:248]: Scratch pad: 00.00.00.00.00.00.00.00.00 (00)
[19:42:58][D][dallas.sensor:144]: 'Sensor 3': Got Temperature=0.0°C
[19:42:58][V][sensor:074]: 'Sensor 3': Received new state 0.000000
[19:42:58][D][sensor:125]: 'Sensor 3': Sending state 0.00000 °C with 1 decimals of accuracy
[19:42:58][V][mqtt:415]: Publish(topic='cesi/sensor/sensor_3/state' payload='0.0' retain=1)
```

Benoit3, 2022-06-11:
I propose a PR to solve it [https://github.com/esphome/esphome/pull/3545] :

- bits 0 to 4 of the CONFIG register of the fourth BYTE of scratchpad register are always '1'
- bit 7 is always 0
=> the solution is to check them to confirm scratchpad validity

nagyrobi, 2022-06-11:
I reproduced this too.

Misiu, 2022-06-11:
@Benoit3 does your PR solves the issue? You can read the temperature on 25m length cables?

Benoit3, 2022-06-11:
The problem is not that the temperature is unavailable (normal with out of spec pull up) but that a wrong value could be provided to other system in case of sensor connection issue. In case of automation , this is major issue.

To answer to comments, the PR is not solving temperature reading errors, just error detection.
And yes with 25m of twisted cable, a 3k pull up, an ESP 32 EVB board, I can read temperature.

Misiu, 2022-06-11:
@Benoit3 that you for clarification.
Did you connect only a single sensor? I had problem with reading from multiple sensors in a star topology, each one with around 20m cable, all connected to a single pin.

Benoit3, 2022-10-13:
Isuue solved with PR integrated in the 2022.8

closed 2022-10-13 by Benoit3