esphome/issues #3360: Splitting YAML files fails when using -platform: template.

gsieben, 2022-06-10:
### The problem

Hi guys,
I have splitted my configuration into single YAML files. Exactly as described on https://esphome.io/guides/configuration-types.html.
Works fine. The secrets.yaml problem I solved with a symbolic link. But as soon as a "- platform: template" section is in the splitted yaml, it doesn't work anymore. I have used for this the ebefalls in the documentation about the dew point. 

- platform: template
  name: ${acronym} Dew Point
  lambda: |-
    return (243.5*(log(id(bme_hum).state/100)+((17.67*id(bme_temp).state)/
    (243.5+id(bme_temp).state))/(17.67-log(id(bme_hum).state/100)-
    ((17.67*id(bme_temp).state)/(243.5+id(bme_temp).state))));
  unit_of_measurement: °C
  icon: "mdi:thermometer-alert"

It seems to be a bug. I have not found a workaround. 

### Which version of ESPHome has the issue?

ESPHome (2022.5.1)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.6.4

### What platform are you using?

ESP32

### Board

Immaterial

### Component causing the issue

Compiler

### Example YAML snippet

```yaml
Main YAML:
substitutions:
  devicename: hd-bad
  acronym: BAD

packages:
  device_base: !include modules/device_base.yaml
  wifi: !include modules/wifi.yaml

############ B U S ############ 
# I²C Bus
i2c:
  sda: GPIO19             # Normal 21 
  scl: GPIO22
  scan: True
  # frequency: 100kHz
  
# ONE Wire
dallas:
  - pin: GPIO16

#######################################
############ S E N S O R S ############ 
#######################################
sensor:
  - <<: !include modules/sensor/esp32.yaml
  - <<: !include modules/sensor/bme280.yaml

bme280.yaml:
- platform: bme280
  temperature:
    name: ${acronym} Temperature
    id: bme_temp
  pressure:
    name: ${acronym} Pressure
  humidity:
    name: ${acronym} Relativ Humidity
    id: bme_hum
  address: 0x76
  update_interval: 15s

- platform: template
  name: ${acronym} Absolute Humidity
  lambda: |-
    const float mw = 18.01534;    // molar mass of water g/mol
    const float r = 8.31447215;   // Universal gas constant J/mol/K
    return (6.112 * powf(2.718281828, (17.67 * id(bme_temp).state) /
      (id(bme_temp).state + 243.5)) * id(bme_hum).state * mw) /
      ((273.15 + id(bme_temp).state) * r); // in grams/m^3
  accuracy_decimals: 2
  update_interval: 15s
  icon: "mdi:water"
  unit_of_measurement: "g/m³"

- platform: template
  name: ${acronym} Dew Point
  lambda: |-
    return (243.5*(log(id(bme_hum).state/100)+((17.67*id(bme_temp).state)/
    (243.5+id(bme_temp).state)))/(17.67-log(id(bme_hum).state/100)-
    ((17.67*id(bme_temp).state)/(243.5+id(bme_temp).state))));
  unit_of_measurement: °C
  icon: "mdi:thermometer-alert"
```

RoboMagus, 2022-07-27:
Could you also please provide some actual logs of what is shown when compiling? 

e.g. what do you get as an output when running the following command through the Esphome CLI (since you've got an IDE setup I assume you should also be able to run this in your environment):
`esphome -v compile main.yaml` 
(Replacing `main.yaml` with the name of your actual config ofcourse.)

gsieben, 2022-07-28:
I'm not at all surprised that you're asking. I can no longer reproduce the error myself. It seems to have disappeared somehow :-D. So the matter is probably settled. Thank you very much.

gsieben, 2022-07-28:
It really drives me crazy. Once it worked. And now it doesn't work again. But probably has nothing to do with the lambda. Here is the compiler output. It is not different from what Visual Code reports anyway. I get the same message when I compile via the ESPHome interface. 

```
➜ esphome esphome -v compile hd-test.yaml
INFO Reading configuration hd-test.yaml...
ERROR Error while reading config: Invalid YAML syntax:

mapping values are not allowed here
  in "modules/esp32.yaml", line 3, column 12:
        filters:
```

I made it as simple as possible. Now it doesn't work if I just leave out the filter with the lambda. 
[image]
[image]

nielsnl68, 2022-07-28:
Should `filters:` not be put direct below name instead of your indented solution?

like:
``'yaml
sensor:
  - platform: uptime
    name: yourname
    filters:
       - lambda: return x/3600
    unit_of_measurement: "h"
```

gsieben, 2022-07-28:
Yes, of course! Oh, I'm blind. At least the error is out. But something still doesn't fit. I found out one small thing. 

There must be a space in front of the `<<: !include` .

```
sensor:
 <<: !include modules/esp32.yaml # the content of the file esp32.yaml is the following comment code
  ###################### contents of the file esp32.yaml ######################
#- platform: uptime
# name: ${acronym} Uptime
# filter:
# - lambda: return x / 3600;
# unit_of_measurement: "h"

#- platform: esp32_hall
# name: ${acronym} Hall
# update_interval: 15s

#- platform: wifi_signal
# name: ${acronym} WiFi strength
# update_interval: 15s
```
I am still checking. Maybe the "blank" is the problem. Maybe the logic of merging here should be made a little smarter.  I have a variety of different error messages here. Depending on what I try.

gsieben, 2022-07-28:
Yes. Seems to be a problem with the indentation. That makes the strangest error messages. 

This works as expected
[image]

This also works
[image]

But not if something else is added. No matter if as include or as code.
[image]

closed 2022-12-11 by github-actions[bot]