esphome/issues #3362: BH1750 fails with update interval < 1s

Stewie3112, 2022-06-11:
### The problem

As per title, bh1750 sensor (and maybe other components using time) just doesn't work without any warning in the logs when update_interval is set below 1s. I actually need faster response time and was using  200ms with the previous releases successfully.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

EugeneKorneev, 2022-08-11:
affected for me too v2022.6.3
i try change update_interval from 1s to 0.5s and get n/a

closed 2022-12-25 by github-actions[bot]

Garfield-Z, 2023-02-28:
ESPHome 2023.2.4 still suffers this issue, the minimal working interval I tried so far is 800ms, when reduced further to 600ms, stopped working thereafter.