esphome/issues #3363: ESP32-S3 or ESP32-C3 missing tool chanin

G6EJD, 2022-06-11:
### The problem

Rpi-4 based ESPHOME, when trying to build an S2 or C3 package, this occurs:

INFO Reading configuration /config/esphome/y.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing y (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'

### Which version of ESPHome has the issue?

2022.2.6 - March 2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant Core 2022.6.5, Home Assistant Supervisor 2022.05.3, Home Assistant OS 8.1

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

Any

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "a211c518650bf75374b0238d3e4e7d99"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "49WM0dYRpMFz"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing test (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'
```

### Additional information

Same results with the ESP32-S2 board type

boklubre, 2022-06-23:
Hi,
I have the same problems.
Any suggestions to fix it?
Cheers

Boris

mmakaay, 2022-06-23:
The only solution would be to compile the code using a system for which a toolchain is available.
I think Raspberry Pi using a 32-bit OS instead of 64-bit should work.
Or use for example a command line install on a PC.

mmakaay, 2022-06-23:
Closing this issue, since it's a duplicate.

Duplicate of #2722

closed 2022-06-23 by mmakaay