esphome/issues #3364: two sensors with same input not work with latest firmware

hidaba, 2022-06-11:
### The problem

I measure water with two different intervals of time for statistics, but with the latest firmware it work only one, 
in this case work WaterGiardino

 `sensor:
  - platform: pulse_counter
    pin: D8
    name: 'Utilizzo Acqua Giardino'
    update_interval: 5s
    icon: 'mdi:water'
    unit_of_measurement: 'L/min'
    id: water_usage
    filters:
      - lambda: return (x / 396); # 396 = 6,6 * 60
      
  - platform: pulse_counter
    pin: D8
    name: "WaterGiardino"
    update_interval: 60s
    icon: 'mdi:water'
    unit_of_measurement: 'L'
    filters:
      - lambda: return (x / 370 ); `

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.2

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

pulse

### Example YAML snippet

```yaml
sensor:
  - platform: pulse_counter
    pin: D8
    name: 'Utilizzo Acqua Giardino'
    update_interval: 5s
    icon: 'mdi:water'
    unit_of_measurement: 'L/min'
    id: water_usage
    filters:
      - lambda: return (x / 396); # 396 = 6,6 * 60
      
  - platform: pulse_counter
    pin: D8
    name: "WaterGiardino"
    update_interval: 60s
    icon: 'mdi:water'
    unit_of_measurement: 'L'
    filters:
      - lambda: return (x / 370 ); `
```

### Anything in the logs that might be useful for us?

```txt
with the latest firmware work only the last sensor, the first it always 0
with 2022.4 and older work correctly
```

nagyrobi, 2022-06-11:
Try using copy platform instead, if measuring from the same sensor. Make two copies and keep the pulse counter as internal.

bkaufx, 2022-06-12:
Looking at the changelog, I can't see anything obvious that would cause this.  I think the copy implementation would look something like this.  You need to change both the existing sensors into platform: copy instead of pulse counter.  Add a pulse counter with no filter, and then each individual sensor applies its own filter in its copy.

```sensor:
  - platform: pulse_counter
    pin: D8
    id: water_pulse
    update_interval: 5s

  - platform: copy
    source_id: water_pulse
    name: 'Utilizzo Acqua Giardino'
    icon: 'mdi:water'
    unit_of_measurement: 'L/min'
    id: water_usage
    filters:
      - lambda: return (x / 396); # 396 = 6,6 * 60
      
  - platform: copy
    source_id: water_pulse
    name: "WaterGiardino"
    update_interval: 60s
    icon: 'mdi:water'
    unit_of_measurement: 'L'
    filters:
      - lambda: return (x / 370 );
```

hidaba, 2022-06-13:
with 2022.5.0 works with 2022.5.1 don't work, I see also changelog but don't see anything.
I change with platform: copy and it works, the only little problem is that I can't change update_interval:
but I resolved to change the number of sample in statistics on homeassistant

thank you @bkaufx and @nagyrobi

RoboMagus, 2022-07-10:
I'm running into the same issue, in my case when using `pulse_counter` followed by `pulse_meter` on the same pin.
The copy platform might be a work around for 2 sensors of the same platform, but that would not work in my config.

ssieb, 2022-11-09:
Using the same pin for multiple components is not supported.  Possibly if someone wanted to do the work to allow sharing the pin, it might be accepted, but there are so many edge cases here.

closed 2022-11-09 by ssieb

nagyrobi, 2022-11-09:
What's wrong with using the copy component?

TekFan, 2022-11-26:
Well, how do you define a "duty_cycle" and "pulse_width" as a copy of a pulse meter ?
A platform can only be "copy" or "duty cycle" for instance but not both at the same time.
So how do you define a copy of a pulse_meter as "duty cycle" ?