esphome/issues #3371: Radon Eye RD200 UUID's no longer same as in module

smarttowers, 2022-06-14:
### The problem

I haven't confirmed that the code does not work. I have tried a python program that uses the same UUID's as used here and it does not work. 
This is the current module UUID's the same as the python code I tried.
00001523-1212-efde-1523-785feabcd123
00001524-1212-efde-1523-785feabcd123
00001525-1212-efde-1523-785feabcd123

My device which I received today has these UUID:
   -- SERVICE: 00001801-0000-1000-8000-00805f9b34fb [Generic Attribute]
   --   --> CHAR: 00002a05-0000-1000-8000-00805f9b34fb, Handle: 3 (0x0003) - INDICATE  - [Service Changed]
   -- SERVICE: 00001800-0000-1000-8000-00805f9b34fb [Generic Access]
   --   --> CHAR: 00002a00-0000-1000-8000-00805f9b34fb, Handle: 22 (0x0016) - READ  - [Device Name]
   --   --> CHAR: 00002a01-0000-1000-8000-00805f9b34fb, Handle: 24 (0x0018) - READ  - [Appearance]
   --   --> CHAR: 00002aa6-0000-1000-8000-00805f9b34fb, Handle: 26 (0x001a) - READ  - [Central Address Resolution]
   -- SERVICE: 00001523-0000-1000-8000-00805f9b34fb [1523]
   --   --> CHAR: 00001524-0000-1000-8000-00805f9b34fb, Handle: 42 (0x002a) - READ WRITE NO RESPONSE  - [1524]
   --   --> CHAR: 00001525-0000-1000-8000-00805f9b34fb, Handle: 44 (0x002c) - READ NOTIFY  - [1525]
   --   --> CHAR: 00001526-0000-1000-8000-00805f9b34fb, Handle: 47 (0x002f) - READ NOTIFY  - [1526]

I assumed that just changing to the new UUID would make it work but I always get a 0 reading from the python script I'm testing.
    uuidRead  = btle.UUID("00001525-0000-1000-8000-00805F9B34FB")
    #uuidRead  = btle.UUID("00001525-1212-efde-1523-785feabcd123")
    RadonEyeValue = RadonEyeService.getCharacteristics(uuidRead)[0]
    RadonValue = RadonEyeValue.read()
    RadonValue = struct.unpack('<f',RadonValue[2:6])[0]

Hoping someone can help me with this here as I suspect that it is also a problem with ESPHome on the current Radon Eye RD200 firmware.

### Which version of ESPHome has the issue?

Currently none

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

realumhelp, 2022-07-15:
This might be related from my ESPHome Log. I just got a Radon Eye today from Amazon Prime Day Sale. Does device firmware play a factor?

```
[21:11:33][I][radon_eye_rd200:015]: Connected successfully!
[21:11:35][W][radon_eye_rd200:030]: No sensor read characteristic found at service 00001523-1212-EFDE-1523-785FEABCD123 char 00001525-1212-EFDE-1523-785FEABCD123
```

tvories, 2022-07-15:
Here's a whole discussion on the newer Radon Eye models.  It appears that the matching may be different with the newer model in the ESPHome integration: https://community.home-assistant.io/t/radoneye-ble-interface/94962/133

drmulligan, 2022-08-19:
I too am encountering this.  Scanning for my RD200 returns name: 'FR:RU22201030471'
Once my esp32 connects I see logs exactly as in the comment by @realumhelp, then every few minutes I see '[W][radon_eye_rd200:141]: Connection in progress'

altmoola, 2022-09-14:
I'm seeing the same issue as @drmulligan 

`[22:25:02][W][radon_eye_rd200:141]: Connection in progress` messages every 1 minute. Bluetooth connection icon is shown on the device. Device name does not follow the FR:R20:SNXXXX naming scheme. It follows a similar scheme to @drmulligan.

Sensor does not report any values when using the examples from this: https://esphome.io/components/sensor/radon_eye_ble.html

arpia49, 2022-09-27:
Hello, the main problem is on the FW version, yes, the current code works with characteristic x-1212-x-x-x is for old models while x-0000-x-x-x seems to be working fine for my 6 month old device.

The GATT slightly changes (I have two python scripts one for old and one for new) old using read, new using subscribe.

Old:
```
device.char_write("00001524-1212-efde-1523-785feabcd123", bytearray([0x50]))
value = device.char_read("00001525-1212-efde-1523-785feabcd123")
```

New:
```
device.subscribe("00001525-0000-1000-8000-00805f9b34fb", callback=handle_data)
device.char_write("00001524-0000-1000-8000-00805f9b34fb", bytearray([0x50]))
```
I would love to make some changes localy and test but I'm new to ESPHome. What I can gladly do is to share my experience with old/new devices.

Please let me know if I can help, discovering radon at home was so unpleasant that if I can ease the detection for someone else, would be kind of _liberation_.

drmulligan, 2022-10-05:
@arpia49 I have tried exactly the same change you proposed before but it did not work.  How did you determine the the correct GATT for your firmware version?  It seems that the manufacturer loves to change these values, I wonder if there is a way to automate the discovery of them.

arpia49, 2022-10-18:
> How did you determine the the correct GATT for your firmware version?
Hello @drmulligan I saw the GATT info with bluetoothctl and saw the services available and realized it could only be the one I used. The workflow is different in my device since you need to subscribe and prepare a callback (instead of reading).

I've been out for a couple of days, if you could send me hte GATT info, I would gladly try to help.

drmulligan, 2022-10-18:
I will have to learn bluetoothctl and put a Pi near my RD200 in order to be able to check this.  Unfortunately I only have the Esp32 with EspHome near my RD200 right now.

tvories, 2022-10-18:
> I will have to learn bluetoothctl and put a Pi near my RD200 in order to be able to check this. Unfortunately I only have the Esp32 with EspHome near my RD200 right now.

I'm all set up with a pi and RD200.  What can I look up for you?

dimatx, 2022-12-19:
Wondering if anyone has RD200 FW 2.0.2 working with ESPHome and could share a working config.

HeedfulCrayon, 2022-12-19:
@dimatx I also have not been able to get mine working with ESPHome, however I got mine working with this on a raspberry pi zero w https://github.com/EtoTen/radonreader
In looking at the code, it looks like this was set up to handle two different models based off of the variable `rdDeviceType` (where the with the new version of the rd200 would be `rdDeviceType=1`
https://github.com/EtoTen/radonreader/blob/e659968321ad5d151e8cbbd1d6933f41c10bc53f/radon_reader_by_handle.py#L74

dimatx, 2022-12-19:
Yeah, I also did spot this PR that seems to have stalled... https://github.com/esphome/esphome/pull/3597 Not sure if it would also fix it our issue but it might... The main intent seems to support some RadonEye that also has temp and humidity, but I wonder if it uses the same radon reporting as ours. Hopefully when that one gets merged, it enables our units to use the 'p2'

HeedfulCrayon, 2022-12-19:
@dimatx that is actually a different model that has more features (including wifi I believe)

dimatx, 2022-12-20:
Went with the python on Raspberry Pi Zero W and things are working!

jdeath, 2022-12-25:
All, I just made a native Homeassistant integration for the FR:RU version (works with ESPHome BLE Gateway) 
https://github.com/jdeath/rd200v2
Not beautiful code, but working.

HeedfulCrayon, 2022-12-29:
> All, I just made a native Homeassistant integration for the FR:RU version (works with ESPHome BLE Gateway) https://github.com/jdeath/rd200v2 Not beautiful code, but working.

Any plans on making a PR into ESPHome?

jdeath, 2022-12-29:
No plans. Why, do you use ESPHome without homeassistant?

jeffeb3, 2022-12-30:
> All, I just made a native Homeassistant integration for the FR:RU version (works with ESPHome BLE Gateway) https://github.com/jdeath/rd200v2 Not beautiful code, but working.

If a better way gets merged into HA (and I think BLE gateway on ESPHome is much better), we can update the docs for this integration so it doesn't deter people who are looking for a solution. 

I mentioned it a few times above, but I don't have access to the hardware anymore. So I can't test any changes. But I am sure it is broken with the newer rd200's.

tvories, 2022-12-30:
> No plans. Why, do you use ESPHome without homeassistant?

I don't have Bluetooth available to my homeassistant server, so the esphome integration is preferred.

jeffeb3, 2022-12-30:
There is a (relatively new) feature in HA and EspHome to turn an esp32 into a ble gateway. The esphome will read from BLE devices and relay the info to HA. It uses a lot less memory on the esp32, and you can have many esp32s with this gateway/proxy set up. Home Assistant does all the work to decode these messages and they show up as new devices in Home Assistant Integrations:

https://esphome.io/components/bluetooth_proxy.html

I switched all my BLE over to this. I just added the proxy to 3 of my esp32s already spread around the house. It is much better to move all of that device soecific decoding to the more powerful HA server, and let the esp32 just be a protocol translator. 

The cost is just that the code to decode the messages needs to be moved into HA integrations, from the esphome code where it was before.

If this rd200 integration gets merged into the main HA, then I will probably deprecate this in esphome, or at least put a big banner pointing people there (so old builds won't break).

jdeath, 2022-12-30:
What? Huh? Just add these lines in your ESPHome config and it becomes your Bluetooth adapter. Can still use other esphome ble things if you have them setup, but many are natively supported

```
bluetooth_proxy:
  active: true
```
Opps. jeffeb3 wrote at same time. I probably won't to a PR into homeassistant, but maybe someone will take my code and do that.

tvories, 2022-12-30:
I was unaware that I could use my esp as a BLE gateway.  Thanks for the info!

altmoola, 2022-12-30:
I'm trying to use the integration that @jdeath made but I'm having a bit of trouble. I have an ESP32 board flashed with the bluetooth proxy enabled. I have HA running on a VM with HACS installed. I have the custom integration `rd200v2` that @jdeath made downloaded. Now what do I do?

jdeath, 2022-12-30:
restart homeassistant. It should find it automatically after a few minutes. If not, go to Settings->Integrations and add the RD200 BLE integration. It should find it. If have issues, make an issue in my repo or in the homeassistant thread.

altmoola, 2022-12-30:
HA didn't find it automatically but if I add the RD200 BLE integration I get `No devices found on the network`. I must be missing something simple. My ESP configuration is below, the MAC address is my RD200. I'm not totally understanding the workflow here. The RD200 is physically showing the connection icon so I believe it is paired with this ESP32. How does the integration find this ESP32 and therefore the RD200? Do I need to add the ESP32 as an integration as well? I also had this `iBeacon Tracker` integration show up when I added the ESP32 node.

YAML:

```
esphome:
  name: basement-radon-node
  platformio_options:
    board_build.flash_mode: dio
    
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

esp32_ble_tracker:

bluetooth_proxy:
  active: true

ble_client:
  - mac_address: ##:##:##:##:##:##
    id: radon_eye_ble_id

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  
  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: ""

```

jdeath, 2022-12-30:
You need to remove it from the ESPHome configuration or that will own it and it will not get to homeassistant
remove these lines from your esphome config:
```
ble_client:
  - mac_address: ##:##:##:##:##:##
    id: radon_eye_ble_id
```
I did the same thing when writing the integration and was so frustrated it wouldn't find it. Also, this only works for versions with serial  FR:RU and  FR:RE , just incase you have a different version.

altmoola, 2023-01-05:
> You need to remove it from the ESPHome configuration or that will own it and it will not get to homeassistant

What does this mean? Right now I'm flashing it by using ESPHome from HA plugin:

[image]

> this only works for versions with serial FR:RU and FR:RE

Yes, I have an FR:RU version

jdeath, 2023-01-05:
It means your config needs to look like what I posted above. The MAC address should not be in the ESPhome ble_client: section.

HAfrom8, 2023-02-19:
My Radoneye is an FR:RD version. Any way I can make this work for me? Currently I'm not able to get the integration to find the Radoneye.

jdeath, 2023-02-19:
No. The Rd serial number will not work. You need to use my custom component for home assistant and configure your esphome as a proxy https://github.com/jdeath/rd200v2

HAfrom8, 2023-02-19:
I thought I did that. I used an ESP32 and loaded it as a Bluetooth proxy using the website you provided. I then downloaded your custom component into HACS. Restarted home assistant. When i look for the integration it's there. However, when i open it, it says "No devices found on the network". I don't have the radoneye app open. 

Perhaps I'm confused as where to put this piece:

esp32_ble_tracker:
  scan_parameters:
    active: true

bluetooth_proxy:
  active: true

Does it go into the Bluetooth proxy ESP config? Or do I need a separate one for the radoneye?

jdeath, 2023-02-25:
Sorry, I've been out. FR:RD? Wow, they keep adding serial numbers. Are you in US? I'll add it to the list

flotux2, 2023-03-07:
I got yesterday my RadonEye RD200 and have problems with the latest version of ESPhome on my board az-delivery-devkit-v4.

The serial of my device starts with FR:RE22*. So I guess I have a new(er) version of the product. Firmware version is 2.0.2.

I changed the search string for the serial number so I could get the MAC ID in scanning mode with esp32_ble_tracker, 
radon_eye_ble. 

But then the readout of the sensor values is not working. I also noticed that the UUIDs are different compared to the definition in the .h file. I changed them with my values. But that does not solve the problem.
[image]

I read about the change [Added support for the RadonEye+2 #3597](https://github.com/esphome/esphome/pull/3597/commits/e592b3d1f71c1a9eef4dcfeb54326c4f19163c64). But that seems to be for the version with WLAN support that I do not have.

I do not use HomeAssistant but just use ESPhome as connection point to the sensor to send the data to my MQTT server.
Any idea how to get it working?

jdeath, 2023-03-07:
Most people have moved to homeassistant. Can you use this guy's script on a rapberry PI or something to read your device? `https://github.com/EtoTen/radonreader` . It would be much easier. If you really want ESPhome, you would need to develop yourself. As a rough outline: download the esphome source code, move the component/[radon_eye_ble](https://github.com/esphome/esphome/tree/dev/esphome/components/radon_eye_ble) and [radon_eye_ble](https://github.com/esphome/esphome/tree/dev/esphome/components/radon_eye_ble) into your esphome/mycomponenets compiling directory, Then edit both to swap out the serial number prefix, and then change the UUIDS, handle, and convert the byte differently as layed out in the above person's script. Then you need to go int your esphome config and force it to load your local copy of the two components.

I tried the above first, but I didn't get very far. I eventually just spent the time to learn how to do it in homeassistant.

flotux2, 2023-03-07:
Thanks for your fast answer @jdeath. I will try to adapt the mentioned source code. Homeassistant sounds very interesting. But all my devices and stuff are configured in ioBroker. A second big smart home solution would be overkill.

dimatx, 2023-03-08:
@flotux2 I used that python script and it worked great before the home assistant solution was available. I recommend it if you have a place to run python with Bluetooth proximity to the RadonEye.

flotux2, 2023-03-08:
@dimatx: I will give it a try. I guess you mean [radon_reader.py](https://github.com/EtoTen/radonreader)? I will also I take up the suggestion of @jdeath to derive from the Python the program logic to read out the sensor values via BLE and hopefully get it working within ESPHome.

dimatx, 2023-03-08:
> @dimatx: I will give it a try. I guess you mean [radon_reader.py](https://github.com/EtoTen/radonreader)? I will also I take up the suggestion of @jdeath to derive from the Python the program logic to read out the sensor values via BLE and hopefully get it working within ESPHome.

Yes, radon_reader.py

EtoTen, 2023-03-10:
@flotux2 Let me know if you need any help. I am the guy that wrote that updated script. 
I actually have mine running on the Host Raspbian OS of an RPI-4, with the home assistant running a docker container on the same device. This wont work if you just use the SD card image install for home assistant, as they lock you out of the host OS.

flotux2, 2023-03-10:
@EtoTen: Thanks for your offer to support me. But it will take some time until I can start with it. I plan to use a RPi Zero W or 2 W and they are currently either out of stock or offered with a ridiculous price. :-(

I use ioBroker with an integrated MQTT server, InfluxDB and Grafana. So I will use your script to read out the sensor data and send it to the MQTT server.

I wrote the hardware vendor if maybe he can support with the different hardware revisions or even contributing code. This is certainly rather unlikely, but perhaps worth a try.

For me still the best option would be a working ESPHome solution. Then I could enable MQTT support in the .yaml file and wouldn't need an own RPi just for reading the sensor values.

EtoTen, 2023-03-10:
@flotux2 ya I agree ESPHome native support could be cleaner. 
Interesting I never knew about ioBroker. Is it better than home assistant? 
Can you just run the script on whatever computer ioBroker is running on instead of getting an RPI Zero? Also you could get a Banana Pi BPI-M2 Zero instead. They also run Linux and are not bad.... https://www.banana-pi.org/en/banana-pi-sbcs/1.html

flotux2, 2023-03-10:
>  Interesting I never knew about ioBroker. Is it better than home assistant?
I olny know ioBroker so I cannot answer it. I'm happy with it but I also like the idea of ESPHome with its cool idea to integrate sensors into HA.

I will try to get a RPi with BLE support to test your script.

flotux2, 2023-04-06:
After trying for a long time, I have now given up. The connection establishment did not work:

`pi@iobroker:/usr/local/bin $ ./radon_reader.py -v -a XX:XX:XX:XX:XX:XX  -t 1
2023-03-11 13:28:10,479 - root - DEBUG - Re-trying connections attempts: 1'
2023-03-11 13:28:11,492 - root - DEBUG - Re-trying connections attempts: 2'
2023-03-11 13:28:12,505 - root - DEBUG - Re-trying connections attempts: 3'
2023-03-11 13:28:13,530 - root - DEBUG - Re-trying connections attempts: 4'
2023-03-11 13:28:14,531 - root - DEBUG - Sending payload (byte): b'P' To handle (int): 42
Helper not started (did you call connect()?)
2023-03-11 13:28:14,532 - root - DEBUG - Failed, trying again (1)...
2023-03-11 13:28:19,550 - root - DEBUG - Re-trying connections attempts: 1'
2023-03-11 13:28:20,565 - root - DEBUG - Re-trying connections attempts: 2'
2023-03-11 13:28:21,579 - root - DEBUG - Re-trying connections attempts: 3'
2023-03-11 13:28:22,608 - root - DEBUG - Re-trying connections attempts: 4'
2023-03-11 13:28:23,610 - root - DEBUG - Sending payload (byte): b'P' To handle (int): 42
Helper not started (did you call connect()?)
2023-03-11 13:28:23,611 - root - DEBUG - Failed, trying again (2)...
^CTraceback (most recent call last):
  File "/usr/local/bin/./radon_reader.py", line 113, in <module>
    GetRadonValue()
  File "/usr/local/bin/./radon_reader.py", line 62, in GetRadonValue
    mRadonValueBQ, mRadonValuePCi = radon_device_reader (mRdDeviceAddress , mRdDeviceType) #get data from the device
  File "/usr/local/bin/radon_reader_by_handle.py", line 91, in radon_device_reader
    p.writeCharacteristic(intHandle, bGETValues, True);
  File "/home/pi/.local/lib/python3.9/site-packages/bluepy/btle.py", line 542, in writeCharacteristic
    self._writeCmd("%s %X %s\n" % (cmd, handle, binascii.b2a_hex(val).decode('utf-8')))
  File "/home/pi/.local/lib/python3.9/site-packages/bluepy/btle.py", line 302, in _writeCmd
    raise BTLEInternalError("Helper not started (did you call connect()?)")
bluepy.btle.BTLEInternalError: Helper not started (did you call connect()?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/./radon_reader.py", line 122, in <module>
    sleep(5)
KeyboardInterrupt`

I was also in contact with the German distributor. End of the year there will be a new product that supports WiFi and Matter. So I guess it will be then easier to integrate it into a Smart Home solution. 

I've sent back my RD200 unit and have to wait some months. :-(

sormy, 2023-12-11:
There are two versions of radon reader. The second one I have at home and wrote a script for it to integrate into hass using mqt: https://github.com/sormy/radoneye-reader Works like a clock for a year plus.

EtoTen, 2023-12-11:
> After trying for a long time, I have now given up. The connection establishment did not work:
>
> ```
> pi@iobroker:/usr/local/bin $ ./radon_reader.py -v -a XX:XX:XX:XX:XX:XX  -t 1
> 2023-03-11 [...]

I think this means that bluepy-helper is not running. You are running this on the actual host, correct?

EtoTen, 2023-12-11:
> There are two versions of radon reader. The second one I have at home and wrote a script for it to run on rpi to integrate into hass using mqtt: https://github.com/sormy/radoneye-reader Works like a clock for a year plus.
> 
> Similarly support can be added in esphome. I will leave reverse engineered contract here if someone wanted to fix:
> 
> MTU: 517
> 
> Service: 00001523-0000-1000-8000-00805f9b34fb
> 
> Characteristics:
> 
>     * Command (write): 00001524-0000-1000-8000-00805f9b34fb
> 
>     * Status (read on notify): 00001525-0000-1000-8000-00805f9b34fb
> 
>     * History (read on notify): 00001526-0000-1000-8000-00805f9b34fb
> 
> 
> Commands:
> 
>     * Status: 0x40
> 
>     * History: 0x41
> 
>     * Beep: 0xA1, 0x11, 0x17, 0x0C, 0x0B, 0x01, 0x25, 0x28
> 
> 
> Status response format:
> 
> ```c
> typedef struct {
>     /* 00 */ uint8_t command; // supposed to be 0x40
>     /* 01 */ uint8_t size;    // supposed to be 0x42
>     /* 02 */ char serial_part2[6];
>     /* 08 */ char serial_part1[3];
>     /* 11 */ char serial_part3[4];
>     /* 15 */ uint8_t __unk1[1];
>     /* 16 */ char model[6];
>     /* 22 */ char version[6];
>     /* 28 */ uint8_t __unk2[5];
>     /* 33 */ uint16_t latest_bq_m3;
>     /* 35 */ uint16_t day_avg_bq_m3;
>     /* 37 */ uint16_t month_avg_bq_m3;
>     /* 39 */ uint8_t __unk3[12];
>     /* 51 */ uint16_t peak_bq_m3;
>     /* 53 */ uint8_t __unk4[15];
> } __attribute__((packed)) radoneye_value_response_t;
> ```
> 
> History response format:
> 
> ```c
> // usually whole struct is up to 504 bytes with 250 max values in response
> // if history is bigger than 250 data points, then multiple responses will be sent back
> typedef struct {
>     /* 00 */ uint8_t command; // supposed to be 0x41
>     /* 01 */ uint8_t response_count; // total number of responses (each response is separate event)
>     /* 02 */ uint8_t response_no; // number of response
>     /* 03 */ uint8_t value_count; // within response
>     /* 04 */ uint16_t values_bq_m3[?];
> } __attribute__((packed)) radoneye_history_response_t;
> ```
So is yours for Version 1 then? 
Mine should work with v1 and v2, although I only tested with v2.

sormy, 2023-12-11:
Above is for v2 (newer one).

EtoTen, 2023-12-12:
> Above is for v2 (newer one).

Cool

sormy, 2024-10-12:
By the way, full reversed spec:

V1: https://github.com/sormy/radoneye/blob/main/KNOWLEDGE_V1.md
v2: https://github.com/sormy/radoneye/blob/main/KNOWLEDGE_V2.md

package has unit tests with examples of payload for v1 and v2. I don't have access to any of radon readers anymore, so hope this will be useful to someone.