esphome/issues #3372: ESPHome + LilygoT5 + external component vbaksa

xcybplx, 2022-06-15:
### The problem

Hello,

Some time ago I wrote a yaml file using external component from vroland. It was working fine, no code was changed and when I tried to update the device using new ESPHome version I cannot install it.

I have this:

`INFO Reading configuration /config/esphome/kkk.yaml...
INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing lilygo-t52 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
|-- <epdiy> 1.0.0+sha.4ad5615
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/src/main.cpp.o
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/board/epd_board_v6.c.o
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/epd_driver.c.o
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/epd_temperature.c.o
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c: In function 'epd_board_init':
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:7: error: 'i2c_config_t {aka struct <anonymous>}' has no member named 'clk_flags'
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
       ^
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: error: 'I2C_SCLK_SRC_FLAG_FOR_NOMAL' undeclared (first use in this function)
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
                    ^
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: note: each undeclared identifier is reported only once for each function it appears in
*** [/data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/board/epd_board_v6.c.o] Error 1`

Any ideas how to fix that?

I'm using:

external_components:
  - source:
      type: git
      url: https://github.com/vbaksa/esphome
      ref: dev
    components: [lilygo_t5_47_display, lilygo_t5_47_battery]

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Latest

### What platform are you using?

ESP32

### Board

LilygoT5 with ESP32 on board.

neostim, 2022-07-17:
I had the same error, for what it's worth I swapped the external component to:

external_components:
  - source: github://ashald/esphome@lilygo-t5-47
    components:
      - lilygo_t5_47

and then it worked.

balbulator, 2022-08-06:
Exactly the same error I have, I'm installing for the first time

> I had the same error, for what it's worth I swapped the external component to:
> 
> external_components:
> 
> * source: github://ashald/esphome@lilygo-t5-47
>   components:
>   
>   * lilygo_t5_47
> 
> and then it worked.

This leads to many other errors that are related to the battery, etc.

NeilS1966, 2022-08-08:
I had the same error and then changed the external_components source to ashald as @balbulator did.
It then compiled without issue.

However, the problem returned:
`Compiling .pioenvs/lilygo-t5-47/libddd/epdiy/board/epd_board_v6.c.o
.piolibdeps/lilygo-t5-47/epdiy/src/epd_driver/board/epd_board_v6.c: In function 'epd_board_init':
.piolibdeps/lilygo-t5-47/epdiy/src/epd_driver/board/epd_board_v6.c:102:7: error: 'i2c_config_t {aka struct <anonymous>}' has no member named 'clk_flags'
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
       ^
.piolibdeps/lilygo-t5-47/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: error: 'I2C_SCLK_SRC_FLAG_FOR_NOMAL' undeclared (first use in this function)
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
                    ^
.piolibdeps/lilygo-t5-47/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: note: each undeclared identifier is reported only once for each function it appears in
Compiling .pioenvs/lilygo-t5-47/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pioenvs/lilygo-t5-47/FrameworkArduino/esp32-hal-gpio.c.o
Compiling .pioenvs/lilygo-t5-47/FrameworkArduino/esp32-hal-i2c.c.o
Compiling .pioenvs/lilygo-t5-47/FrameworkArduino/esp32-hal-ledc.c.o
*** [.pioenvs/lilygo-t5-47/libddd/epdiy/board/epd_board_v6.c.o] Error 1
========================== [FAILED] Took 2.45 seconds ==========================`

The strange thing is, I have two lilygo T5 boards, I used the same code on the other board and it compiles.
I'm not a programmer and have only limited knowledge, I'm hoping someone with better knowledge can resolve the problem.

manyos-robert, 2022-08-17:
I'm getting the same error with this def:

external_components:
  - source:
      type: git
      url: https://github.com/vbaksa/esphome
      ref: dev
    components: [lilygo_t5_47_display, lilygo_t5_47_battery]
    refresh: 1h

Did you find a workaround?

jesserockz, 2022-12-16:
Please dont create issues for external components. You need to create an issue on that repo. This issue tracker is for official esphome components only.

closed 2022-12-16 by jesserockz