esphome/issues #3373: Modbus CRC failure

RobertSmart, 2022-06-15:
### The problem

Hi, 

I am trying to return a couple of registers from a connected modbus device but I'm getting a CRC failed error.

I've tested my setup using a slave simulator and it all works ok through that (using a usb dongle and unslave modbus slave simulator)

attached logging shows error

The command and the response look OK, but the CRC check seems to be failing. 

Is there anything obvious that can cause this issue? I've tried another bit of software on another master and that seems to work OK.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

TTGO-ethernet with custom modbuc controller

### Component causing the issue

modbus

### Example YAML snippet

```yaml
esphome:
  name: heatpump_controller

esp32:
  board: pico32
  framework:
    type: arduino

#Enable logging
logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  # baud_rate: 0 #disable logging over uart
  hardware_uart: UART0

# Enable Home Assistant API
api:
  password: ""

uart:
  id: mod_bus
  tx_pin: 33
  rx_pin: 35
  baud_rate: 9600
  stop_bits: 1

modbus:
  flow_control_pin: 32
  uart_id: mod_bus
  id: modbus1

modbus_controller:
  - id: heatpump
    ## the Modbus device addr
    address: 156
    modbus_id: modbus1
    setup_priority: 1

# switch:
#   - platform: modbus_controller
#     modbus_controller_id: heatpump
#     id: SH_Pump_Status
#     register_type: coil
#     address: 0xBCB
#     name: "Space Heating Pump Status"
sensor:
  - platform: modbus_controller
    modbus_controller_id: heatpump
    id: HP_AC_Voltage
    name: "Heat Pump AC Voltage"
    address: 7021
    unit_of_measurement: "V" ## for any other unit the value is returned in minutes
    register_type: holding
    value_type: U_WORD
```

### Anything in the logs that might be useful for us?

```txt
[12:57:28][V][modbus_controller:036]: Sending next modbus command to device 156 register 0x1B6D count 2
[12:57:28][VV][uart.arduino_esp32:150]:     Flushing...
[12:57:28][V][modbus:184]: Modbus write: 9C.03.1B.6D.00.02.4F.7F (8)
[12:57:28][V][modbus_controller:487]: Command sent 3 0x1B6D 2
[12:57:28][V][modbus:058]: Modbus received Byte  0 (0X0)
[12:57:28][V][modbus:058]: Modbus received Byte  156 (0X9c)
[12:57:28][V][modbus:058]: Modbus received Byte  3 (0X3)
[12:57:28][V][modbus:058]: Modbus received Byte  4 (0X4)
[12:57:28][V][modbus:058]: Modbus received Byte  0 (0X0)
[12:57:28][W][modbus:096]: Modbus CRC Check failed! C158!=04
[12:57:28][V][modbus:058]: Modbus received Byte  236 (0Xec)
[12:57:28][V][modbus:058]: Modbus received Byte  20 (0X14)
[12:57:28][V][modbus:058]: Modbus received Byte  25 (0X19)
[12:57:28][V][modbus:058]: Modbus received Byte  185 (0Xb9)
[12:57:28][V][modbus:058]: Modbus received Byte  197 (0Xc5)
[12:57:28][V][modbus:058]: Modbus received Byte  0 (0X0)
[12:57:28][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[12:57:28][V][component:200]: Components should block for at most 20-30ms.
```

nagyrobi, 2022-06-15:
Do you have the 120ohm termination resistor attached to the transceiver at the ESP?

RobertSmart, 2022-06-15:
Hi,

I do at the esphome device end. I can't say if there is one at the other end on the slave device. I can check the resistance across it tomorrow with a meter and put a resistor across the a/b terminals if needed. 

As I said, it does work with some other kit, so I assumed it was OK. 

Thanks

nagyrobi, 2022-06-15:
Also try with and without the GND connected at one end.

RobertSmart, 2022-06-16:
Hi,

The measured resistance was 240ohm, so I put a 240ohm resistor across the terminals to lower it to 120ohm. This hasn't made any difference, it is still doing the same thing. I've tested the device with yet another modbus adapter and that is working. I'm going to try a few more things, but not ruling out hardware issue at the moment.

Any more ideas?

Thanks,

Rob Smart

RobertSmart, 2022-06-16:
Hi,

I've played around a little more today and have made a bit of progress. I am now seeing only intermittent CRC errors

[17:32:25][V][modbus_controller:126]: Range : 1B6D Size: 2 (3) skip: 0
[17:32:25][V][modbus_controller:036]: Sending next modbus command to device 156 register 0x1B6D count 2
[17:32:25][V][modbus:184]: Modbus write: 9C.03.1B.6D.00.02.4F.7F (8)
[17:32:25][V][modbus_controller:487]: Command sent 3 0x1B6D 2
[17:32:25][V][modbus:058]: Modbus received Byte  255 (0Xff)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][modbus:058]: Modbus received Byte  156 (0X9c)
[17:32:25][V][modbus:058]: Modbus received Byte  3 (0X3)
[17:32:25][V][modbus:058]: Modbus received Byte  4 (0X4)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][modbus:058]: Modbus received Byte  237 (0Xed)
[17:32:25][V][modbus:058]: Modbus received Byte  24 (0X18)
[17:32:25][V][modbus:058]: Modbus received Byte  170 (0Xaa)
[17:32:25][V][modbus:058]: Modbus received Byte  172 (0Xac)
[17:32:25][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:25][V][modbus_controller:036]: Sending next modbus command to device 156 register 0x1B6D count 2
[17:32:25][V][modbus:184]: Modbus write: 9C.03.1B.6D.00.02.4F.7F (8)
[17:32:25][V][modbus_controller:487]: Command sent 3 0x1B6D 2
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][modbus:058]: Modbus received Byte  156 (0X9c)
[17:32:25][V][modbus:058]: Modbus received Byte  3 (0X3)
[17:32:25][V][modbus:058]: Modbus received Byte  4 (0X4)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][W][modbus:096]: Modbus CRC Check failed! C158!=04
[17:32:25][V][modbus:058]: Modbus received Byte  237 (0Xed)
[17:32:25][V][modbus:058]: Modbus received Byte  24 (0X18)
[17:32:25][V][modbus:058]: Modbus received Byte  170 (0Xaa)
[17:32:25][V][modbus:058]: Modbus received Byte  172 (0Xac)
[17:32:25][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:25][V][modbus_controller:036]: Sending next modbus command to device 156 register 0x1B6D count 2
[17:32:25][V][modbus:184]: Modbus write: 9C.03.1B.6D.00.02.4F.7F (8)
[17:32:25][V][modbus_controller:487]: Command sent 3 0x1B6D 2
[17:32:25][V][modbus:058]: Modbus received Byte  255 (0Xff)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][modbus:058]: Modbus received Byte  156 (0X9c)
[17:32:25][V][modbus:058]: Modbus received Byte  3 (0X3)
[17:32:25][V][modbus:058]: Modbus received Byte  4 (0X4)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][modbus:058]: Modbus received Byte  237 (0Xed)
[17:32:25][V][modbus:058]: Modbus received Byte  24 (0X18)
[17:32:25][V][modbus:058]: Modbus received Byte  170 (0Xaa)
[17:32:25][V][modbus:058]: Modbus received Byte  172 (0Xac)
[17:32:25][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:25][V][modbus_controller:036]: Sending next modbus command to device 156 register 0x1B6D count 2
[17:32:25][V][modbus:184]: Modbus write: 9C.03.1B.6D.00.02.4F.7F (8)
[17:32:25][V][modbus_controller:487]: Command sent 3 0x1B6D 2
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][modbus:058]: Modbus received Byte  156 (0X9c)
[17:32:25][V][modbus:058]: Modbus received Byte  3 (0X3)
[17:32:25][V][modbus:058]: Modbus received Byte  4 (0X4)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][W][modbus:096]: Modbus CRC Check failed! C158!=04
[17:32:25][V][modbus:058]: Modbus received Byte  237 (0Xed)
[17:32:25][V][modbus:058]: Modbus received Byte  24 (0X18)
[17:32:25][V][modbus:058]: Modbus received Byte  170 (0Xaa)
[17:32:25][V][modbus:058]: Modbus received Byte  172 (0Xac)
[17:32:25][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[17:32:25][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:25][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:26][V][modbus_controller:036]: Sending next modbus command to device 156 register 0x1B6D count 2
[17:32:26][V][modbus:184]: Modbus write: 9C.03.1B.6D.00.02.4F.7F (8)
[17:32:26][V][modbus_controller:487]: Command sent 3 0x1B6D 2
[17:32:26][V][modbus:058]: Modbus received Byte  255 (0Xff)
[17:32:26][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:26][V][modbus:058]: Modbus received Byte  156 (0X9c)
[17:32:26][V][modbus:058]: Modbus received Byte  3 (0X3)
[17:32:26][V][modbus:058]: Modbus received Byte  4 (0X4)
[17:32:26][V][modbus:058]: Modbus received Byte  0 (0X0)
[17:32:26][V][modbus:058]: Modbus received Byte  237 (0Xed)
[17:32:26][V][modbus:058]: Modbus received Byte  24 (0X18)
[17:32:26][V][modbus:058]: Modbus received Byte  170 (0Xaa)
[17:32:26][V][modbus:058]: Modbus received Byte  172 (0Xac)
[17:32:26][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[... cut ...]
[17:32:26][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[17:32:26][V][component:200]: Components should block for at most 20-30ms.
[17:32:26][D][modbus_controller:032]: Modbus command to device=156 register=0x1B6D countdown=0 no response received - removed from send queu

But why does the system send 5 requests at once? this might be causing an issue as the remote device might not respond quick enough? 

Also, even though I get a few responses that don't show errors, it still doesn't update Home assistant with the values it got.

nagyrobi, 2022-06-16:
@martgras

RobertSmart, 2022-06-17:
Hi,

This was was a hardware issue. I've implemented the 'fail-safe' configuration and this have now started to work OK. 

Thanks for the help.

closed 2022-06-17 by RobertSmart

RobertSmart, 2022-06-20:
Hi,

So the issue seemed to be related to the voltage of the A/B lines. I used an RS485 circuit that I had previously used for a raspberry pi, which was using 5v logic levels. This time I wanted to be 3.3v tolerate for the ESP32, so I switched the MAX485 IC to a MAX3485, but I didn't change any of the bias resistor levels, so they were clamping the ideal voltage too low. I swapped these out for 680 ohm resistors to get the idle voltage to near 200mv. I also had the A line pulled low at idle, and the B line pulled high (via the bias resistors) which worked fine on the other board but didn't work on this (it worked in some instances but not others). So I swapped the lines and pulled the A line high (via the 680 ohm resistor) and the B line low (via 680 ohm resistor) and then it all worked fine. 

I hope that is enough detail, please let me know if any more is required.

nagyrobi, 2022-06-20:
I use this kind of MAX485 directly with ESP32, and has no issues:

[image]

RobertSmart, 2022-06-20:
I designed the board from scratch for our application so figured I would try a dedicated 3.3v tolerate IC as I've heard the MAX485 can be a little unstable at 3.3v, even though I too have used those modules at 3.3v without issue.