esphome/issues #3376: SGP4x unable to compile

TOBx2, 2022-06-16:
### The problem

Error message:

`src/esphome/components/sgp4x/sgp4x.cpp: In member function 'bool esphome::sgp4x::SGP4xComponent::measure_gas_indices_(int32_t&, int32_t&)':
src/esphome/components/sgp4x/sgp4x.cpp:172:37: error: cannot bind non-const lvalue reference of type 'float&' to an rvalue of type 'float'
  172 |     voc_algorithm_.get_states(this->voc_state0_, this->voc_state1_);
      |                               ~~~~~~^~~~~~~~~~~
In file included from src/esphome/components/sgp4x/sgp4x.h:8,
                 from src/esphome/components/sgp4x/sgp4x.cpp:1:
/data/esp-air-quality/.piolibdeps/esp-air-quality/Sensirion Gas Index Algorithm/src/VOCGasIndexAlgorithm.h:51:28: note:   initializing argument 1 of 'void VOCGasIndexAlgorithm::get_states(float&, float&)'
   51 |     void get_states(float& state0, float& state1);
      |                     ~~~~~~~^~~~~~
*** [/data/esp-air-quality/.pioenvs/esp-air-quality/src/esphome/components/sgp4x/sgp4x.cpp.o] Error 1
========================= [FAILED] Took 16.07 seconds ========================= `

### Which version of ESPHome has the issue?

2022.06

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.6

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

sgp4x

### Example YAML snippet

```yaml
- platform: sgp4x
    voc:
      name: "air-quality-voc"
    compensation:
      temperature_source: esp_air_quality_dht11_temp  # optional: can improve the sensor’s internal calculations
      humidity_source: esp_air_quality_dht11_humidity # optional: can improve the sensor’s internal calculations
    update_interval: 3min
```

TOBx2, 2022-06-16:
When removing the compensation the error persists.

IAMABANANA, 2022-06-16:
I have the same issue but the error message is a bit different:

### The problem
**Error message:**
`src/esphome/components/sgp4x/sgp4x.cpp: In member function 'bool esphome::sgp4x::SGP4xComponent::measure_gas_indices_(int32_t&, int32_t&)':
src/esphome/components/sgp4x/sgp4x.cpp:172:37: error: invalid initialization of non-const reference of type 'float&' from an rvalue of type 'float'
     voc_algorithm_.get_states(this->voc_state0_, this->voc_state1_);
                                     ^
In file included from src/esphome/components/sgp4x/sgp4x.h:8:0,
                 from src/esphome/components/sgp4x/sgp4x.cpp:1:
/data/basement/.piolibdeps/basement/Sensirion Gas Index Algorithm/src/VOCGasIndexAlgorithm.h:51:10: note:   initializing argument 1 of 'void VOCGasIndexAlgorithm::get_states(float&, float&)'
     void get_states(float& state0, float& state1);
          ^
Compiling /data/basement/.pioenvs/basement/src/esphome/core/component.cpp.o
*** [/data/basement/.pioenvs/basement/src/esphome/components/sgp4x/sgp4x.cpp.o] Error 1
========================= [FAILED] Took 10.47 seconds =========================`

**Which version of ESPHome has the issue?**
2022.6.0

**What type of installation are you using?**
Home Assistant Add-on

**Which version of Home Assistant has the issue?**
2022.6.6

**What platform are you using?**
ESP32

**Board**
esp32dev

**Component causing the issue**
sgp4x

**Example YAML snippet**
`  - platform: sgp4x
    voc:
      name: "basement_sgp40_VOC"
    nox:
      name: "basement_sgp40_NOx"
    compensation:
      humidity_source: basement_humidity
      temperature_source: basement_temperature`

**Anything in the logs that might be useful for us?**
No response

**Additional information**
I've tried it with and without compensation and nox

martgras, 2022-06-19:
The compile error is caused by a change in sensirions code : https://github.com/Sensirion/arduino-gas-index-algorithm/commit/dbf298208a9049454e0c4c80adf2efb1a21f45f9

martgras, 2022-06-20:
Fixed in https://github.com/esphome/esphome/pull/3581

closed 2022-06-20 by martgras

martgras, 2022-06-20:
I can't  repro the problem anymore when using the dev tree

my test YAML
```yaml
substitutions:
  updates: 10s

esphome:
  name: testm5

esp32:
  board: pico32
  framework:
    type: esp-idf

logger:
  level: DEBUG

i2c:
  - id: bus_a
    sda: 21
    scl: 22
    scan: True

sensor:
  - platform: sgp4x
    voc:
      name: "VOC Index"
      id: sgp40_voc_index
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    nox:
      name: "NOx"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230

    #    compensation:
    #      humidity_source: sht40_humidity
    #      temperature_source: sht40_temperature
    update_interval: ${updates}

```

climeoDev1, 2022-06-21:
Thats my docker-compose part:

`
  esphome:
    #image: esphome/esphome:beta
    image: esphome/esphome:dev
    #image: esphome/esphome:latest
    network_mode: host
    devices:
      - "/dev/ttyUSB1:/dev/ttyUSB1"
    #  - "/dev/bus/usb:/dev/bus/usb"
    # - "/dev/ttyS3:/dev/ttyUSB0"
    privileged: true
    container_name: esphome
    volumes:
      - ./config:/config:rw
      # Use local time for logging timestamps
      - /etc/localtime:/etc/localtime:ro
    # network_mode: host
    environment:
      - ESPHOME_DASHBOARD_USE_PING=true
    ports:
      - 6152:6052
      - 6123:6123
`

And the test YAML is exactly the same as yours...

`