esphome/issues #3377: http_request.get to https://script.google.com connection lost not response data

hmjvaline, 2022-06-17:
### The problem

When using action http_request.get , the normal https web page can be connected normally, but it is connected to  google app script web pages,like https://script.google.com, the URL is https://script.google.com/macros /s/JrZw8O3T_Lh85FNxTUNk8MBragLWo8DsdhKjJ0q6qeWgaWSalCtj3g6ZiFDC9xPg/exec, mainly returns json data, the data can be received normally in the chrome browser, but the error connection lost status_code will be received in esphome, the request seems to be able to be sent, and the google app script can be received , but the response data is 0 bytes

### Which version of ESPHome has the issue?

2022.6.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

http_request

### Example YAML snippet

```yaml
http_request :
   useragent : esphome/device 
   timeout : 5s

      - http_request.get:
         url: https://script.google.com/macros/s/JrZw8O3T_Lh85FNxTUNk8MBragLWo8DsdhKjJ0q6qeWgaWSalCtj3g6ZiFDC9xPg/exec
         headers:
          Content-Type: application/json
         verify_ssl: false
         on_response:
           then:
             - logger.log:
                 format: 'Response status: %d'
                 args:
                   - status_code
```

### Anything in the logs that might be useful for us?

```txt
[09:00:30][D][main:301]: Response status: -5
[09:00:30][W][http_request:081]: HTTP Request failed; URL: https://script.google.com/macros/s/JrZw8O3T_Lh85FNxTUNk8MBragLWo8DsdhKjJ0q6qeWgaWSalCtj3g6ZiFDC9xPg/exec; Error: connection lost
```

hmjvaline, 2022-07-26:
anyone has the same problem? or operate like me using?

Oliver84, 2022-09-30:
I'm having the same issue

trombik, 2022-10-13:
it seems that `POST` is also broken.

```
esphome Version: 2022.9.1
platform = platformio/espressif8266 @ 3.2.0
```

```console
[09:18:59][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[09:18:59][V][json:051]: Size after shrink 76 bytes
[09:18:59][HTTP-Client][begin] url: https://hooks.slack.com/services/XXX/XXX
[09:18:59][HTTP-Client][begin] host: hooks.slack.com port: 443 url: /services/XXX/XXX
[09:18:59][HTTP-Client][sendRequest] type: 'POST' redirCount: 0
[09:18:59][hostByName] request IP for: hooks.slack.com
[09:18:59][hostByName] Host: hooks.slack.com IP: 13.213.164.176
[09:18:59]BSSL:_connectSSL: start connection
[09:18:59]BSSL:CERT: 30 82 06 c4 30 82 05 ac a0 03 02 01 02 02 10 0b e7 85 7f e6 56 11 fa ef 4f d5 2a 4e af 87 91 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 30 4f 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 15 30 13 06 03 55 04 0a 13 0c 44 69 67 69 43 65 72 74 20 49 6e 63 31 29 30 27 06 03 55 04 03 13 20 44 69 67 69 43 65 72 74 20 54 4c 53 20 52 53 41 20 53 48 41 32 35 36 20 32 30 32 30 20 43 41 31 30 1e 17 0d 32 32 30 33 32 38 30 30 30 30 30 30 5a 17 0d 32 33 30 33 32 38 32 33 35 39 35 39 5a 30 71 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 13 30 11 06 03 55 04 08 13 0a 43 61 6c 69 66 6f 72 6e 69 61 31 16 30 14 06 03 55 04 07 13 0d 53 61 6e 20 46 72 61 6e 63 69 73 63 6f 31 21 30 1f 06 03 55 04 0a 13 18 53 6c 61 63 6b 20 54 65 63 68 6e 6f 6c 6f 67 69 65 73 2c 20 49 6e 63 2e 31 12 
[09:18:59]BSSL:CERT: 30 10 06 03 55 04 03 13 09 73 6c 61 63 6b 2e 63 6f 6d 30 82 01 22 30 0d 06 09 2a 86 48 86 f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a 02 82 01 01 00 ce ca 54 1e 38 c6 93 9a 00 ca a4 b5 46 d2 6c 39 b6 99 5c 15 5e d6 d8 e4 b1 9e d6 c9 f8 5d cc ce 42 2b 68 9e f7 db 00 18 24 3b 0e 12 64 8e e5 1e 16 68 66 be 60 0a 0e 5e cd 9b 7f 6f 90 34 45 36 68 15 c1 f7 92 47 4b 6b 81 fa 25 c7 40 62 55 bc d7 56 7e f5 f8 df c3 5f 2b 57 eb b7 97 32 f2 f1 76 b0 4a bd 26 8d 17 e8 4a 3f a6 af b7 ce 4c 19 7b c2 eb c2 f3 b1 51 ab 88 db 7e 1f 04 ec b1 ec 9e ce e1 d8 81 bc 7c ec 4c ad 70 db 69 4a 50 f4 36 70 a7 4a 43 29 84 69 9b f2 36 5f 52 0c 1b 47 63 c3 d7 54 1f 18 50 9e 0b aa 23 c5 9c 1c 4f 2e 60 15 b2 c2 22 6f 57 d3 90 d8 b9 ab ff 8d 63 3e 19 8f 8d 25 a7 ab fe 94 c3 5c a9 d7 61 
[09:18:59]BSSL:CERT: 76 85 3b 5b bc dc 04 44 06 cc 64 fa 04 51 31 c9 21 bd ca b6 bc 54 f5 4e ea 85 36 cd 54 37 cc 0d 5f d9 cb e6 dc 9b 7a 8a 8c d2 ac 6a 0d 4d dc bd 92 96 e5 02 03 01 00 01 a3 82 03 78 30 82 03 74 30 1f 06 03 55 1d 23 04 18 30 16 80 14 b7 6b a2 ea a8 aa 84 8c 79 ea b4 da 0f 98 b2 c5 95 76 b9 f4 30 1d 06 03 55 1d 0e 04 16 04 14 a8 fa b4 0c 4f b0 e0 87 15 a1 09 3e 15 a5 1b e5 c5 f0 82 b0 30 21 06 03 55 1d 11 04 1a 30 18 82 09 73 6c 61 63 6b 2e 63 6f 6d 82 0b 2a 2e 73 6c 61 63 6b 2e 63 6f 6d 30 0e 06 03 55 1d 0f 01 01 ff 04 04 03 02 05 a0 30 1d 06 03 55 1d 25 04 16 30 14 06 08 2b 06 01 05 05 07 03 01 06 08 2b 06 01 05 05 07 03 02 30 81 8f 06 03 55 1d 1f 04 81 87 30 81 84 30 40 a0 3e a0 3c 86 3a 68 74 74 70 3a 2f 2f 63 72 6c 33 2e 64 69 67 69 63 65 72 74 2e 63 6f 6d 
[09:18:59]BSSL:CERT: 2f 44 69 67 69 43 65 72 74 54 4c 53 52 53 41 53 48 41 32 35 36 32 30 32 30 43 41 31 2d 34 2e 63 72 6c 30 40 a0 3e a0 3c 86 3a 68 74 74 70 3a 2f 2f 63 72 6c 34 2e 64 69 67 69 63 65 72 74 2e 63 6f 6d 2f 44 69 67 69 43 65 72 74 54 4c 53 52 53 41 53 48 41 32 35 36 32 30 32 30 43 41 31 2d 34 2e 63 72 6c 30 3e 06 03 55 1d 20 04 37 30 35 30 33 06 06 67 81 0c 01 02 02 30 29 30 27 06 08 2b 06 01 05 05 07 02 01 16 1b 68 74 74 70 3a 2f 2f 77 77 77 2e 64 69 67 69 63 65 72 74 2e 63 6f 6d 2f 43 50 53 30 7f 06 08 2b 06 01 05 05 07 01 01 04 73 30 71 30 24 06 08 2b 06 01 05 05 07 30 01 86 18 68 74 74 70 3a 2f 2f 6f 63 73 70 2e 64 69 67 69 63 65 72 74 2e 63 6f 6d 30 49 06 08 2b 06 01 05 05 07 30 02 86 3d 68 74 74 70 3a 2f 2f 63 61 63 65 72 74 73 2e 64 69 67 69 63 65 72 74 2e 
[09:18:59]BSSL:CERT: 63 6f 6d 2f 44 69 67 69 43 65 72 74 54 4c 53 52 53 41 53 48 41 32 35 36 32 30 32 30 43 41 31 2d 31 2e 63 72 74 30 09 06 03 55 1d 13 04 02 30 00 30 82 01 80 06 0a 2b 06 01 04 01 d6 79 02 04 02 04 82 01 70 04 82 01 6c 01 6a 00 76 00 ad f7 be fa 7c ff 10 c8 8b 9d 3d 9c 1e 3e 18 6a b4 67 29 5d cf b1 0c 24 ca 85 86 34 eb dc 82 8a 00 00 01 7f d2 3a 5f da 00 00 04 03 00 47 30 45 02 20 1c f9 5c a6 52 4f 87 fc 38 0c b5 f2 98 91 40 f9 0d 88 de 07 b1 db a2 ba 76 ad 88 83 24 e5 82 25 02 21 00 d9 54 ef 0f 01 04 b2 92 89 04 31 ef b6 1a c5 4c d8 a1 7e a7 65 15 cb 6b d3 66 fc 01 93 6d 2a 11 00 77 00 35 cf 19 1b bf b1 6c 57 bf 0f ad 4c 6d 42 cb bb b6 27 20 26 51 ea 3f e1 2a ef a8 03 c3 3b d6 4c 00 00 01 7f d2 3a 5f bc 00 00 04 03 00 48 30 46 02 21 00 ea 21 fe 25 73 7b 78 d5 
[09:18:59]BSSL:CERT: 1e 6b 75 52 6a 84 8d ac 06 6f 72 85 52 e1 89 de aa fe 35 d0 a8 bc 94 f2 02 21 00 98 d1 87 ee 15 a1 c2 4e 5c 0f c6 81 bd a7 2e 92 83 7a 85 a3 7d b4 ff 1f 50 90 fc 65 17 1f a8 ff 00 77 00 b3 73 77 07 e1 84 50 f8 63 86 d6 05 a9 dc 11 09 4a 79 2d b1 67 0c 0b 87 dc f0 03 0e 79 36 a5 9a 00 00 01 7f d2 3a 5f da 00 00 04 03 00 48 30 46 02 21 00 e4 28 7f 56 3d af 12 26 cb b5 df b4 7e c3 f4 69 8f 3b 03 0d 4a 89 0e a9 f7 fb 2c 90 46 08 6e 55 02 21 00 c7 16 69 d2 5e bc 7b 19 ed 97 d3 2d 0b 30 e9 ef 23 3b 30 8e ed 33 0a dc 60 e2 f5 3a a3 31 71 cc 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b 05 00 03 82 01 01 00 00 40 6d e7 4d 4d e7 8f 52 18 65 7b 20 10 02 8c bd 46 b9 b3 77 60 96 69 26 e8 8b 53 11 65 4f c4 af 94 4a fd 7a c1 77 8e 29 f6 92 4d 79 dd d7 cd a5 3d 70 a3 9d 49 bc 51 
[... cut ...]
[09:19:00]User-Agent: esphome/backgarden-doorbell-1
[09:19:00]Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
[09:19:00]Connection: keep-alive
[09:19:00]Content-Type: application/json
[09:19:00]Content-Length: 68
[09:19:00]
[09:19:00]-----
[09:19:01][HTTP-Client][returnError] error(-5): connection lost
[09:19:01][VV][scheduler:026]: set_timeout(name='', timeout=3000)
[09:19:01][W][http_request:081]: HTTP Request failed; URL: https://hooks.slack.com/services/XXX/XXX; Error: connection lost
[09:19:01][HTTP-Client][end] tcp is closed
```

WDeenik, 2022-12-06:
I'm getting this exact same error when trying to POST to the Spotify API. Doing the exact same request (same headers, same body) through Postman works like a charm.

t-animal, 2023-01-23:
I also get this problem calling a slack API on a d1 mini.

t-animal, 2023-01-23:
Might be related to https://github.com/esphome/issues/issues/3893

flavorplus, 2024-03-11:
I have the same problem as well trying to connect to https://script.google.com. Other simple API services work fine. I even tried setting: power_save_mode: none in the wifi section but this doesn't seem to make any difference.

zaxaco, 2024-03-13:
same here!

```substitutions:
  device_name: luster3
  device_type: Light
  device_room: Hall
  static_ip_address: 192.168.1.132

esphome:
  name: ${device_name}
  platform: ESP8266
  board: d1_mini
  esp8266_restore_from_flash : True

wifi:
  ssid: !secret wifissid
  password: !secret wifipw
  id: ${device_name}_wifi_connection
  manual_ip:
    static_ip: ${static_ip_address}
    gateway: !secret gatewayip
    subnet: !secret subnetip
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Hotspot"
    password: !secret appw
    
captive_portal:

logger:
    level: DEBUG
# Enable Home Assistant API
api:
  password: !secret apipw
  reboot_timeout: 0s

# configuration entry OTA
ota:
  safe_mode: true
  password: !secret otapw

web_server:
  port: 80
  auth:
    username: !secret webuser
    password: !secret webpw

# Text Sensor with general information
text_sensor:
  - platform: version
    name: '${device_name} ESPHome Ver.'
  - platform: wifi_info
    ip_address:
      name: '${device_name} ip'
    ssid:
      name: '${device_name} ssid'

sensor:
  - platform: wifi_signal
    name: '${device_name} WiFi Signal'
    id: ${device_name}_wifi_signal
    update_interval: 60s
    accuracy_decimals: 0

switch:
  - platform: gpio
    pin: GPIO12
    name: "${device_name} lamp1"
    inverted: true
    restore_mode: RESTORE_DEFAULT_ON 
    
  - platform: gpio
    pin: GPIO14
    name: "${device_name} lamp2"
    inverted: true
    restore_mode: ALWAYS_ON

  - platform: gpio
    pin: GPIO016
    name: "${device_name} lamp3"
    inverted: true
    restore_mode: ALWAYS_OFF

  # The following can be omitted
  - platform: restart
    name: "${device_name} restart"

# Example configuration entry
http_request:
  useragent: esphome/${device_name}
  timeout: 5s
  
# Example configuration entry
interval:
  - interval: 1min
    then:
      # - script.execute: say_hello_server
      - http_request.get: 
          url: https://esphome.io
          verify_ssl: false
```

`[W][http_request:086]: HTTP Request failed; URL: https://esphome.io/; Error: connection failed; Duration: 0 ms`

my board is Wemos D1 mini

randybb, 2024-03-13:
@zaxaco This is not the same. Check your network configuration - without configured (and working) DNS it will be never able to connect to esphome.io as it needs to know its IP address.