esphome/issues #3378: Ultrasonic Sensor - No "NaN" or 'Null' value sent when in "Timeout" mode

MrSiO, 2022-06-17:
### The problem

I have a setup in which my UltraSonic sensor is most of the time in Timeout-Mode (room lenght more then 4 Meters).

From what I have read, when in Timeout-Mode (distance exceeding 4M in my configuration) ESPHome should send a "NaN" or invalid value.  But that is not what is happening.

- If force_update is disabled:  Nothing is sent to HA when in Timeout-Mode
- If force_update is enabled: The last valid value is sent continuously

In my setup I need to know then the sensor goes from "Timeout" to "valid distance sent" and most important "back to Timeout".

How can we get a NaN value when sensor is in timeout-mode?

Thx
 

### Which version of ESPHome has the issue?

V2022.6.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Home Assistant Core 2022.6.6

### What platform are you using?

ESP8266

### Board

NodeMCU CP2102

### Component causing the issue

Ultrasonic HC-SR04

### Example YAML snippet

```yaml
sensor:
  - platform: ultrasonic
    trigger_pin: GPIO5
    echo_pin: GPIO4
    name: "Ultrasonic Sensor"
    update_interval: 1s
    timeout: 4m         
    force_update: true    //or False
    filters:             
    - sliding_window_moving_average:
        window_size: 4  
        send_every: 4
```

### Anything in the logs that might be useful for us?

```txt
Below you can see I had a previous valid value of:  0.07880m
Then sensor gets into Timeout-mode, but you can see it keeps sending to HA the last valid value of 0.07880 instead of a NaN or NULL value.

[18:20:14][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:15][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:16][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:17][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:17][D][sensor:124]: 'Ultrasonic Sensor': Sending state 0.07880 m with 2 decimals of accuracy
[18:20:18][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:19][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:20][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:21][D][ultrasonic.sensor:036]: 'Ultrasonic Sensor' - Distance measurement timed out!
[18:20:21][D][sensor:124]: 'Ultrasonic Sensor': Sending state 0.07880 m with 2 decimals of accuracy
```

bkaufx, 2022-06-17:
I think it's the filters getting rid of the nan values.  Here is a similar problem someone else posted recently; https://github.com/esphome/issues/issues/3338

One workaround might be to create a template binary sensor and control it in an on_raw_value automation of the ultrasonic sensor so the binary sensor will be high if the raw value is nan.

ssieb, 2022-06-17:
NaN is very awkward in math calculations.  If you want to do something with it, you'll have to use a lambda filter before the average or use the on_raw_value trigger to catch it.

closed 2022-06-17 by ssieb

MrSiO, 2022-06-17:
Hi all.
Issue resolved thx to your help.

Final config is the following:

`    
sensor:
  - platform: ultrasonic
    trigger_pin: GPIO5
    echo_pin: GPIO4
    name: "Ultrasonic Sensor"
    update_interval: 3s
    timeout: 4m
    force_update: false
    filters:             
    - lambda: if (isnan(x)) {  return 999.0; } return x;     
    - sliding_window_moving_average:
          window_size: 3
          send_every: 3`

When in timeout mode, the returned value is 999.
That makes it possible for me to automate easily.

Thx to all.