esphome/issues #3379: SGP4x compenastion library compile issue

climeoDev1, 2022-06-17:
### The problem

The SGP4x sensor library fails during compilation:

Error:
src/esphome/components/sgp4x/sgp4x.cpp: In member function 'bool esphome::sgp4x::SGP4xComponent::measure_gas_indices_(int32_t&, int32_t&)':
src/esphome/components/sgp4x/sgp4x.cpp:172:37: error: invalid initialization of non-const reference of type 'float&' from an rvalue of type 'float'
     voc_algorithm_.get_states(this->voc_state0_, this->voc_state1_);
                                     ^
In file included from src/esphome/components/sgp4x/sgp4x.h:8:0,
                 from src/esphome/components/sgp4x/sgp4x.cpp:1:
.piolibdeps/room-master/Sensirion Gas Index Algorithm/src/VOCGasIndexAlgorithm.h:51:10: note:   initializing argument 1 of 'void VOCGasIndexAlgorithm::get_states(float&, float&)'
     void get_states(float& state0, float& state1);
          ^
*** [.pioenvs/room-master/src/esphome/components/sgp4x/sgp4x.cpp.o] Error 1

### Which version of ESPHome has the issue?

docker: esphome/esphome:latest

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

SGP4x

### Example YAML snippet

```yaml
sensor:
- platform: shtcx
  temperature:
    name: "sht40_temperature"
    id: "sht40_temperature"
  humidity:
    name: "sht40_humidity"
    id: "sht40_humidity"
  address: 0x70
  update_interval: 60s

- platform: sgp4x
  voc:
    name: "VOC Index"
    id: sgp41_voc_index
    algorithm_tuning:
      index_offset: 100
      learning_time_offset_hours: 12
      learning_time_gain_hours: 12
      gating_max_duration_minutes: 180
      std_initial: 50
      gain_factor: 230
  nox:
    name: "NOx"
    algorithm_tuning:
      index_offset: 100
      learning_time_offset_hours: 12
      learning_time_gain_hours: 12
      gating_max_duration_minutes: 180
      std_initial: 50
      gain_factor: 230
      
  compensation:
    humidity_source: sht40_humidity
    temperature_source: sht40_temperature
    
  update_interval: 10s
```

ssieb, 2022-06-17:
Duplicate of #3376.

closed 2022-06-17 by ssieb

climeoDev1, 2022-06-20:
The problem still exists. Passing the parameter as value is not compatible with passing the parameter as const reference
as defined in the .h file:

`
src/esphome/components/sgp4x/sgp4x.cpp: In member function 'bool esphome::sgp4x::SGP4xComponent::measure_gas_indices_(int32_t&, int32_t&)':
src/esphome/components/sgp4x/sgp4x.cpp:172:37: error: invalid initialization of non-const reference of type 'float&' from an rvalue of type 'float'
     voc_algorithm_.get_states(this->voc_state0_, this->voc_state1_);
                                     ^
In file included from src/esphome/components/sgp4x/sgp4x.h:8:0,
                 from src/esphome/components/sgp4x/sgp4x.cpp:1:
.piolibdeps/room-master-2/Sensirion Gas Index Algorithm/src/VOCGasIndexAlgorithm.h:51:10: note:   initializing argument 1 of 'void VOCGasIndexAlgorithm::get_states(float&, float&)'
     void get_states(float& state0, float& state1);
 `

In addition the test1 doesn't include the SGP4x library.