esphome/issues #3382: Compiler warning in spi code

UniKyrn, 2022-06-18:
### The problem

```
Compiling /data/esp32-tft/.pioenvs/esp32-tft/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling /data/esp32-tft/.pioenvs/esp32-tft/FrameworkArduino/esp32-hal-spi.c.o
Compiling /data/esp32-tft/.pioenvs/esp32-tft/FrameworkArduino/esp32-hal-time.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/esp32-tft/.pioenvs/esp32-tft/FrameworkArduino/esp32-hal-timer.c.o
```

### Which version of ESPHome has the issue?

2022.6.0 and earlier

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

spi

### Additional information

The software engineer in me hates seeing warnings like these, because you can't be sure there isn't a platform where they aren't actually the cause of an error.

mmakaay, 2022-06-23:
Those are warnings outside the ESPHome codebase, so we cannot fix those.

mmakaay, 2022-06-23:
Duplicate of #3162

closed 2022-06-23 by mmakaay

UniKyrn, 2022-09-28:
> Those are warnings outside the ESPHome codebase, so we _cannot_ fix those.

You mis-spelled "chose not to".  I had hoped your team would fix it and pass the fix on to the team you got the code from.

ssieb, 2022-09-28:
It's the Arduino ESP framework.  Feel free to submit a PR to them if you want.  It's not breaking anything, so not a priority.

ssieb, 2022-09-28:
Btw, we can't actually fix it because platformio uses the release that they provide.  We would have to fork the Arduino repo and somehow tell platformio to use our version instead.  Way too much work for no benefit.

UniKyrn, 2022-09-28:
Except that I didn't get the code from them, I got it from you.  I'm as far removed from knowing their team and how to connect with them as it can get.  At a bare minimum I'd have expected you to have passed the bug on to them and marked it as "In Progress" instead of just closing it.

mmakaay, 2022-09-29:
Sorry to have trampled your expectations. One's got to pick his battles, and since there was no functional problem here, people went for other battles.
The ESPHome Devs are as far removed from knowing that team as you are, so that wouldn't make a difference in who could have reported the issue.

Anyway, the issue has already been addressed upstream, so at some point the fix will enter the ESPHome ecosystem.

https://github.com/esphome/issues/issues/3382