esphome/issues #3383: BME280 suddenly starts reporting pressures around 700hPa instead of 1000hPa after deep sleep.

dbaarda, 2022-06-20:
### The problem

The BME280 sensor seems to suddenly start reporting pressures around 700-760hPa after waking from a deep sleep, after previously reporting normal atmospheric pressures of around 1005-1025hPa (I'm at 34m altitude above sea-level). It doesn't happen always, but after a couple of hours of wake-report-sleep cycles every minute it suddenly starts doing it and then only a full reboot will make it start reporting the right pressure.

### Which version of ESPHome has the issue?

2022.6.0b4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.6.6

### What platform are you using?

ESP32-IDF

### Board

Adafruit QT Py ESP32-C3 + PiicoDev Atmospheric Sensor BME280

### Component causing the issue

bme280

### Example YAML snippet

```yaml
esphome:
  name: underhouse
  platformio_options:
    board_build.flash_mode: dio

esp32:
# At some point the actual board might be supported.
#  board: adafruit_qt_py_esp32-c3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
  variant: ESP32C3

# Enable logging
logger:

ota:
  password: "<secret>"

mqtt:
  broker: 192.168.7.6
  username: "mqtt"
  password: "<secret>"
  discovery: true
  discovery_retain: true
  discovery_unique_id_generator: mac
  discovery_object_id_generator: device_name
  birth_message:
  will_message:
  shutdown_message:
  on_message:
    - topic: esp/sleep_mode
      payload: 'OFF'
      then:
        - deep_sleep.prevent: deep_sleep_1
    - topic: esp/sleep_mode
      payload: 'ON'
      then:
        - deep_sleep.allow: deep_sleep_1

wifi:
  ssid: "<secret>"
  password: "<secret>"
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Underhouse Fallback Hotspot"
    password: "<secret>"

i2c:
  sda: 5
  scl: 6
  scan: true
  id: bus_a

# We need to use globals to share sensor state without circular references.
globals:
   - id: temperature
     type: double
     restore_value: no
     initial_value: 'NAN'
   - id: pressure
     type: double
     restore_value: no
     initial_value: 'NAN'
   - id: humidity
     type: double
     restore_value: no
     initial_value: 'NAN'

# Example configuration entry
sensor:
  - platform: uptime
    name: "system uptime"

  - platform: wifi_signal
    name: "wifi strength"
    update_interval: 60s
    expire_after: 90s

  - platform: template
    name: "Dew Point Temperature"
    id: dewpoint_temperature
    filters:
      - lambda: |-
          return (243.5*(log(id(humidity)/100)+((17.67 * x)/
          (243.5 + x)))/(17.67-log(id(humidity)/100)-
          ((17.67 * x)/(243.5 + x))));
    icon: 'mdi:thermometer-alert'
    unit_of_measurement: °C
    update_interval: never
    expire_after: 90s

  - platform: template
    name: "Sea Level Pressure"
    id: sealevel_pressure
    filters:
      - lambda: |-
          const float STANDARD_ALTITUDE = 34.0; // in meters, see note
          return x / powf(1 - ((0.0065 * STANDARD_ALTITUDE) /
            (id(temperature) + (0.0065 * STANDARD_ALTITUDE) + 273.15)), 5.257); // in hPa
    unit_of_measurement: 'hPa'
    update_interval: never
    expire_after: 90s

  - platform: template
    name: "Absolute Humidity"
    id: absolute_humidity
    filters:
      - lambda: |-
          const float mw = 18.01534;    // molar mass of water g/mol
          const float r = 8.31447215;   // Universal gas constant J/mol/K
          return (6.112 * powf(2.718281828, (17.67 * id(temperature)) /
            (id(temperature) + 243.5)) * x * mw) /
            ((273.15 + id(temperature)) * r); // in grams/m^3
    accuracy_decimals: 2
    icon: 'mdi:water'
    unit_of_measurement: 'g/m³'
    update_interval: never
    expire_after: 90s

  - platform: bme280
    temperature:
      name: "Temperature"
      id: bme280_temperature
      oversampling: 4x
      expire_after: 90s
      on_value:
        then:
        - lambda: |-
            id(temperature) = x;
    pressure:
      name: "Pressure"
      id: bme280_pressure
      oversampling: 4x
      expire_after: 90s
      on_value:
        then:
        - lambda: |-
            id(pressure) = x;
    humidity:
      name: "Humidity"
      id: bme280_humidity
      oversampling: 4x
      expire_after: 90s
      on_value:
        then:
        - lambda: |-
            id(humidity) = x;
            id(dewpoint_temperature).publish_state(id(temperature));
            id(sealevel_pressure).publish_state(id(pressure));
            id(absolute_humidity).publish_state(id(humidity));
    address: 0x77
    update_interval: 60s

deep_sleep:
  id: deep_sleep_1
  run_duration: 1s
  sleep_duration: 54s
```

### Anything in the logs that might be useful for us?

```txt
Note that I've also very occasionally seen error messages logged about "failed to initialize bme280", but when that happens it completely fails to read values at all, and usually resolves itself in a following "wakeup". I'm not sure if these events happen before it starts reporting bad values. I don't think these are correlated, but they might both have the same cause.

Sorry I haven't got a saved sample.
```

### Additional information

This sounds suspiciously like the problem reported here that turned out to be using an uninitialised variable to set the control registers of the BMS280;

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME280-pressure-reading-wrong/td-p/6204

In Home Assistant, the graphs look like;

[image]

dbaarda, 2022-06-24:
I had a look at the code and I can't see any uninitialized variables. The following discussion looks more relevant;

https://github.com/rwaldron/johnny-five/issues/1274

It looks like they concluded that if they wait a bit before taking a reading, it fixes the problem.

This also mentions people having problems with this sensor, but for temperature this time. They fiddled with the I2C speed to try and fix it;

https://community.particle.io/t/bme280-sensor-problem/49627/29

I'm starting to form a theory: the setup for this device is racy. The current [ESPHome setup code](https://github.com/esphome/esphome/blob/28d2949ebef614240b0c400e15a3f710adbf799c/esphome/components/bme280/bme280.cpp#L100) reads the device id and then immediately launches into reading the calibration data off the device.

If we look at the official [Bosch driver code](https://github.com/BoschSensortec/BME280_driver/blob/c47f06eb44fc96970f0abfcc941ec16425b2a9e6/bme280.c#L421) it reads the device id, then issues a soft-reset before reading the calibration data. Importantly the [soft reset code](https://github.com/BoschSensortec/BME280_driver/blob/c47f06eb44fc96970f0abfcc941ec16425b2a9e6/bme280.c#L723) sends the soft reset command and the polls the status register until "NVM copying is done". The NVM is the calibration data. It also retries this whole process up to 5x before it gives up.

I suspect the magic bit is waiting until NVM copying is done, but the soft reset might also help. Note that when using deep_sleep the whole setup re-runs on wakeup like it was a reset, but the bme280 has not been reset at all, and has been in deep sleep after the last "forced" read. I'm wondering if it needs a soft reset to reload it's NVM correctly.

Even if the bme280 does reload it's NVM when you wake it up by talking to it again, perhaps it takes a little longer and we start reading out the calibration data before it's finished loading, ending up with invalid calibration data.

In any case, it sure likes like the bme280 setup code could be made more robust by copying the official Bosch driver a bit more closely.

dbaarda, 2022-06-29:
I've done some testing adding debugging to log the calibration constants and raw ADC read values. My theory that incomplete loading of the calibration constants might be the cause has been debunked. Instead, it appears that the raw ADC value being read are wrong.

The values logged when the reading is correctly in the ~1000hPa range;
```
[17:26:03][D][bme280.sensor:246]: RawT adc=520472
[17:26:03][D][bme280.sensor:267]: RawP adc=338040, tfine=110474
[17:26:03][D][bme280.sensor:304]: RawH adc=24982, tfine=110474
[17:26:03][V][bme280.sensor:232]: Got temperature=21.6°C pressure=1026.0hPa humidity=44.9%
[...]
[17:26:03][C][bme280.sensor:149]: BME280:
[17:26:03][C][bme280.sensor:150]:   Address: 0x77
[17:26:03][C][bme280.sensor:162]:   IIR Filter: OFF
[17:26:03][C][bme280.sensor:163]:   Update Interval: 60.0s
[17:26:03][C][bme280.sensor:165]:   Temperature 'Temperature'
[17:26:03][C][bme280.sensor:165]:     Device Class: 'temperature'
[17:26:03][C][bme280.sensor:165]:     State Class: 'measurement'
[17:26:03][C][bme280.sensor:165]:     Unit of Measurement: '°C'
[17:26:03][C][bme280.sensor:165]:     Accuracy Decimals: 1
[17:26:03][C][bme280.sensor:166]:     Oversampling: 2x
[17:26:03][C][bme280.sensor:170]:     Calibration: t1=28307, t2=26788, t3=50
[17:26:03][C][bme280.sensor:171]:   Pressure 'Pressure'
[17:26:03][C][bme280.sensor:171]:     Device Class: 'pressure'
[17:26:03][C][bme280.sensor:171]:     State Class: 'measurement'
[17:26:03][C][bme280.sensor:171]:     Unit of Measurement: 'hPa'
[17:26:03][C][bme280.sensor:171]:     Accuracy Decimals: 1
[17:26:03][C][bme280.sensor:172]:     Oversampling: 2x
[17:26:03][C][bme280.sensor:182]:     Calibration: p1=36105, p2=-10538, p3=3024, p4=7137, p5=-50, p6=-7, p7=9900, p8=-10230, p9=4285
[17:26:03][C][bme280.sensor:183]:   Humidity 'Humidity'
[17:26:03][C][bme280.sensor:183]:     Device Class: 'humidity'
[17:26:03][C][bme280.sensor:183]:     State Class: 'measurement'
[17:26:03][C][bme280.sensor:183]:     Unit of Measurement: '%'
[17:26:03][C][bme280.sensor:183]:     Accuracy Decimals: 1
[17:26:03][C][bme280.sensor:184]:     Oversampling: 2x
[17:26:03][C][bme280.sensor:191]:     Calibration: h1=75, h2=384, h3=0, h4=270, h5=50, h6=30
```

The values logged when the reading is incorrectly in the 700hPa range;
```
[17:23:27][D][bme280.sensor:246]: RawT adc=520120
[17:23:27][D][bme280.sensor:267]: RawP adc=524272, tfine=109898
[17:23:27][D][bme280.sensor:304]: RawH adc=24809, tfine=109898
[17:23:27][V][bme280.sensor:232]: Got temperature=21.5°C pressure=704.5hPa humidity=43.9%
[...]
[17:23:28][C][bme280.sensor:149]: BME280:
[17:23:28][C][bme280.sensor:150]:   Address: 0x77
[17:23:28][C][bme280.sensor:162]:   IIR Filter: OFF
[17:23:28][C][bme280.sensor:163]:   Update Interval: 60.0s
[17:23:28][C][bme280.sensor:165]:   Temperature 'Temperature'
[17:23:28][C][bme280.sensor:165]:     Device Class: 'temperature'
[17:23:28][C][bme280.sensor:165]:     State Class: 'measurement'
[17:23:28][C][bme280.sensor:165]:     Unit of Measurement: '°C'
[17:23:28][C][bme280.sensor:165]:     Accuracy Decimals: 1
[17:23:28][C][bme280.sensor:166]:     Oversampling: 2x
[17:23:28][C][bme280.sensor:170]:     Calibration: t1=28307, t2=26788, t3=50
[17:23:28][C][bme280.sensor:171]:   Pressure 'Pressure'
[17:23:28][C][bme280.sensor:171]:     Device Class: 'pressure'
[17:23:28][C][bme280.sensor:171]:     State Class: 'measurement'
[17:23:28][C][bme280.sensor:171]:     Unit of Measurement: 'hPa'
[17:23:28][C][bme280.sensor:171]:     Accuracy Decimals: 1
[17:23:28][C][bme280.sensor:172]:     Oversampling: 2x
[17:23:28][C][bme280.sensor:182]:     Calibration: p1=36105, p2=-10538, p3=3024, p4=7137, p5=-50, p6=-7, p7=9900, p8=-10230, p9=4285
[17:23:28][C][bme280.sensor:183]:   Humidity 'Humidity'
[17:23:28][C][bme280.sensor:183]:     Device Class: 'humidity'
[17:23:28][C][bme280.sensor:183]:     State Class: 'measurement'
[17:23:28][C][bme280.sensor:183]:     Unit of Measurement: '%'
[17:23:28][C][bme280.sensor:183]:     Accuracy Decimals: 1
[17:23:28][C][bme280.sensor:184]:     Oversampling: 2x
[17:23:28][C][bme280.sensor:191]:     Calibration: h1=75, h2=384, h3=0, h4=270, h5=50, h6=30
```
Note that all the calibration constants are identical, the temperature and humidity ADC values are around the same magnitude, but the pressure ADC value is much higher in the 700hPa case. Note the conversion of the raw ADC pressure value to hPa transforms it so that a high ADC value means a low pressure.

I started to then suspect that maybe the results were being read too soon after triggering the forced-mode sampling so I added some debugging to collect times using `millis()`. This showed we were waiting considerably longer than the wait time of ~16ms, probably due to normal scheduling overheads.
```
[21:13:46][V][bme280.sensor:201]: Sending conversion request...
[21:13:46][V][bme280.sensor:216]: Getting conversion result timeout=16.449999ms after millis=6602
[...]
[21:13:46][V][bme280.sensor:219]: Reading conversion result at millis=6692
[21:13:46][V][bme280.sensor:225]: Read completed at millis=6696
[21:13:46][D][bme280.sensor:250]: RawT adc=516672
[21:13:46][D][bme280.sensor:271]: RawP adc=524256, tfine=104259
[21:13:46][D][bme280.sensor:308]: RawH adc=25307, tfine=104259
[21:13:46][V][bme280.sensor:236]: Got temperature=20.4°C pressure=703.3hPa humidity=46.8%
```
[... cut ...] ater, and finished the read at 6696ms so it took 4ms. This is very generous timing for the adc conversion and read, so it shouldn't be causing problems.

I'm now suspecting that there is some kind of delay or grey-out on the sensor state after waking from sleep. I'm going to check its status register to see if there is any indication there that it's not quite in a right state, but I'm strongly suspecting that these guys just need a software reset to ensure they are configured right after wakeup.

dbaarda, 2022-06-29:
Looking at the status register it looks corect in both the 700hPa and 1000hPa cases. Bit 0 (loading NVM) is always zero after setup (I can't see it during/before setup because thats before mqtt is connected and thus before the logging I can see), and bit 3 toggles to one (measuring) immediately after issuing the forced conversion command, and zero before we read the result. Interestingly bit 2 seems to be always set in both cases, but there is nothing in the datasheet about what that bit is meant to mean.

So I guess I'm down to my "greyout after sleep, needs soft reset" theory. I'm going to add that now and see what happens.

dbaarda, 2022-06-30:
I've added the soft reset and it appears to fix the problem. Pull request on the way.

markusg1234, 2022-07-08:
Thanks for the PR. Has fixed it for me too.

closed 2022-08-07 by jesserockz