esphome/issues #3387: Setting the password for a Grow fingerprint sensor bricks the sensor

lancer73, 2022-06-21:
### The problem

When setting the fingerprint sensor password like this:

Add to the secrets.yaml file
`fingerprint_scanner: "0x01234567"`

And then in the code for the NodeMCU first:
`new_password: !secret fingerprint_scanner`

and then
`password: !secret fingerprint_scanner`

The logs tell you that the password is wrong. Upon debugging I see that the password used in the `password` clause is correct, but the password being set is dependent on all of the yaml code, so probably it is a memory address of a pointer instead of the actual password.

Furthermore, when setting the new password, I missed the password in the logs, because I had set the log level to WARN (did this at the end of debugging my code). My suggestion is to "upgrade" the loglevel of the "password is set to:" message from INFORM to WARN, as you don't want to miss this message ever. I initially bricked my sensor due to this (and I have learned a valuable lesson regarding log levels) 

### Which version of ESPHome has the issue?

2021.06.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.06.6

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

fingerprint_grow

### Example YAML snippet

```yaml
Secrets.yaml
`fingerprint_scanner: "0x01234567"`

In project yaml
`new_password: !secret fingerprint_scanner`
```

### Anything in the logs that might be useful for us?

```txt
[09:41:51][E][fingerprint_grow:198]: Wrong password
[09:41:51][E][component:112]: Component fingerprint_grow was marked as failed.
Serial port closed!
```

leranp, 2022-08-03:
same here :-(  any news about it?

lancer73, 2022-08-05:
I think lines 205 - 207 should be changed from:
```
  ESP_LOGI(TAG, "Setting new password: %d", *this->new_password_);
  this->data_ = {SET_PASSWORD, (uint8_t)(*this->new_password_ >> 24), (uint8_t)(*this->new_password_ >> 16),
                 (uint8_t)(*this->new_password_ >> 8), (uint8_t)(*this->new_password_ & 0xFF)};
```

to:
 ```
 ESP_LOGI(TAG, "Setting new password: %d", this->new_password_);
  this->data_ = {SET_PASSWORD, (uint8_t)(this->new_password_ >> 24), (uint8_t)(this->new_password_ >> 16),
                 (uint8_t)(this->new_password_ >> 8), (uint8_t)(this->new_password_ & 0xFF)};
```

What the code is doing is changing the password to the memore address where the password is stored. (Forgive me if I'm wrong, but it has been 20 years since I did something serious in C)

ofirnk, 2022-08-05:
Can someone run this code in debug with the actual set-password cmd commented out?
I've bricked my device and want to see if I can brute force a small memory range to guess the password.

So in other words:
1. Comment line 206 `this->data_ = {SET_PASSWORD, (uint8_t)(*this->new_password_ >> 24), (uint8_t)(*this->new_password_ >> 16),
                 (uint8_t)(*this->new_password_ >> 8), (uint8_t)(*this->  `
2. Make sure the log line stays the same, and that you run in debug log level
3. Run set password, few times
4. Paste the debug output with the password

Thank you in advance!

lancer73, 2022-08-05:
> Can someone run this code in debug with the actual set-password cmd commented out? I've bricked my device and want to see if I can brute force a small memory range to guess the password.
>
> So in [...]

See this topic in the HA fora:
https://community.home-assistant.io/t/cannot-communicate-with-grow-r503-fingerprint-sensor-after-setting-password/432628/5

The password will likely start with 0x3FFF3 (which leaves plenty to try)

closed 2022-08-17 by ssieb