esphome/issues #3388: Invalid preamble from SenseAir

kamaradclimber, 2022-06-21:
### The problem

I'm using a [SenseAir](https://esphome.io/components/sensor/senseair.html) component with an esp32.

I'm getting the following error:
> 08:26:15][W][senseair:031]: Invalid preamble from SenseAir! 00fe0400 000004e5 c600fe04 00

So far I've tried:
- with another senseair
- switching RX and TX in the UART configuration
- using different GPIOs for uart
- disabling logging (as suggested on https://github.com/esphome/issues/issues/488 even if it seems relevant when using specific GPIOs)

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

senseair

### Example YAML snippet

```yaml
esphome:
  name: co2-sensor

esp32:
  board: nodemcu-32s

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE

uart:
  rx_pin: GPIO27
  tx_pin: GPIO26
  baud_rate: 9600
  
web_server:
  port: 80

sensor:
  - platform: senseair
    co2:
      name: "SenseAir CO2 Value 1"
    update_interval: 10s
    
switch:
  - platform: restart
    name: "CO2 sensor restart"

# Enable Home Assistant API
api:
  encryption:
    key: "<redacted>"

[... 437 lines cut ...]
[08:24:35][V][senseair:037]:     ... e5
[08:24:35][V][senseair:037]:     ... c6
[08:24:37][VV][api.service:355]: on_ping_request: PingRequest {}
[08:24:37][VV][api.service:043]: send_ping_response: PingResponse {}
[08:24:39][VV][scheduler:196]: Running interval '' with interval=10000 last_execution=119972 (now=129972)
[08:24:45][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=126412 (now=136412)
[08:24:45][VV][uart.arduino_esp32:150]:     Flushing...
[08:24:45][E][uart:015]: Reading from UART timed out at byte 9!
[08:24:45][VV][uart.arduino_esp32:150]:     Flushing...
[08:24:45][W][senseair:023]: Reading data from SenseAir failed!
[08:24:45][V][component:199]: Component senseair.sensor took a long time for an operation (0.12 s).
[08:24:45][V][component:200]: Components should block for at most 20-30ms.
[08:24:49][VV][scheduler:196]: Running interval '' with interval=10000 last_execution=129972 (now=139975)
[08:24:52][VV][api.service:355]: on_ping_request: PingRequest {}
[08:24:52][VV][api.service:043]: send_ping_response: PingResponse {}
[08:24:55][VV][scheduler:196]: Running interval 'update' with interval=10000 last_execution=136412 (now=146419)
[08:24:55][VV][uart.arduino_esp32:150]:     Flushing...
[08:24:55][VV][uart.arduino_esp32:150]:     Flushing...
[08:24:55][W][senseair:031]: Invalid preamble from SenseAir! 00fe0400 000004e5 c600fe04 00
[08:24:55][V][senseair:037]:     ... 00
[08:24:55][V][senseair:037]:     ... 00
[08:24:55][V][senseair:037]:     ... 04
[08:24:55][V][senseair:037]:     ... e5
[08:24:55][V][senseair:037]:     ... c6
```

davidbb, 2022-08-02:
@kamaradclimber I've seen this on units I've soldered poorly or there is poor contact between TX/RX pins and the microcontroller board. I've also seen it sporadically when using low update intervals like what you have in your configuration sample (10s).

kamaradclimber, 2022-08-03:
Thanks for your feedback, I'll try to improve the contact by soldering the pins and the board + increase interval.

AndreasBVB, 2022-09-02:
I ran into a silly situation today where I drove the board and the sensor with two power sources and didn't level the GND. Every UART reading is a mess and is cured immediately when I started from the same GND.

blin100, 2022-09-29:
@AndreasBVB 
I'm running into the same problem as you! I just started driving the sensor using a dedicated power supply as some people mentioned that might cause issues, and now I'm getting gibberish in my UART readings. What exactly did you do to fix this? (I'm an absolute idiot and don't quite understand what you mean by leveling the GND)....
Thanks!!

AndreasBVB, 2022-09-29:
Fairly simple:
(1) disconnect all your loads: boards, sensors, and whatever you don't want to risk frying
(2) carefully connect the GND of the two power sources and see if they blow up LOL
(3) measure the voltages of the two power sources and make sure things didn't go crazy
(4) disconnect two power supplies (now GND leveled), reconnect your loads and restart

blin100, 2022-09-29:
Thanks!
Hmm, so the power is still nice at 5v, but I'm fluctuating between:
[15:15:49][E][uart:015]: Reading from UART timed out at byte 0!
[15:15:49][W][senseair:023]: Reading data from SenseAir failed!

and
[15:17:49][W][senseair:031]: Invalid preamble from SenseAir! 00000000 00000000 00000000 00

It's actually mostly just failing now, which is interesting...

AndreasBVB, 2022-09-29:
Maybe some faulty power supply? Try these steps:
(1) get a robust 5V from e.g. a well-made USB power
(2) power the sensor directly from that 5V (this sensor is picky, even 4.8v will cause problems)
(3) use a good buck/step-down to get 3.3v and power the ESP board directly

If these didn't work you may need to check connections/soldering.

blin100, 2022-09-29:
Hahaha, actually, this whole thing started because I was powering the senseair from the 5v line on the esp32, which was powered off a dedicated usb power brick.
Measured it, and the voltage coming to the sensor was low, like 4.2v, so I put in a second power brick just to the sensor.

When I use UART modbus to read from the sensor, I get:
[15:31:45][W][modbus:116]: Modbus CRC Check failed! F012!=03
[15:31:46][W][modbus:116]: Modbus CRC Check failed! F012!=03
[15:31:46][D][modbus_controller:032]: Modbus command to device=254 register=0x03 countdown=0 no response received - removed from send queue

Stumbled across the fact that there's a senseair integration, tried it, and got the invalid preamble.

It's probably bad soldering? Super new to this, so they are absolutely not good connections...
Thanks for your help though!

AndreasBVB, 2022-09-29:
The ESP board 5V is not always a true 5V. It's sometimes the USB power behind a diode or something. Putting the USB power directly to SenseAir and do your own buck for the 3.3v to ESP makes more sense.

blin100, 2022-09-29:
My god. apparently it was just the ORDER that i plugged the two things in. I did sensor, then microcontroller, and it works. I was always doing it the other way for some reason.

AndreasBVB, 2022-09-29:
This shouldn't be the case. I can start the ESP board first and plug in UART devices anytime after. UART devices will work like this (unlike I2C sensors).

blin100, 2022-09-29:
Bizarre. Not really sure why, though I will go and fix some soldering connections, it might just be something stupid making it seem like the order matters. In any case, connecting the two grounds was exactly what I needed, I think.

Thanks a ton for your help!

jbtwo, 2023-01-05:
I continued getting the `invalid preamble` error even after levelling GND and powering the sensor directly from 5V. For me the fix was to disable UART logging by adding this to the config:

```
logger:
  baud_rate: 0
```

OP already has this in their config, but just wanted to share incase it helps someone else, since this issue is the first Google result for "Senseair invalid preamble"

zatricky, 2024-05-31:
In case it is helpful for others I had a device with this issue due to a faulty USB hub that was connected to a PC. The hub was not providing it's own power despite having a dedicated power input plugged in. The easy quick fix was to plug it into another non-data USB power source.