esphome/issues #3389: Adopted devices prepared via web.esphome.io still showing after installation and rename with old name.

LayerCakeMakes, 2022-06-21:
### The problem

After preparing a device for adoption on web.esphome.io the adoption in the esphome addon on homeassistant worked fine it is running with the new name but there is still a device showing up as discovered ready to adapt with the old name though the device does not exist anymore and is not reachable under the same name in the network.

Other people are also having this issue but don't seem to have created a bug report here:
https://community.home-assistant.io/t/have-the-devices-working-with-new-names-but-i-am-still-seeing-adopt-panels-with-the-old-names/373199/2

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.6

### What platform are you using?

ESP8266

### Board

Sonoff RFBridgeR2

LeeXGreen, 2022-07-18:
I also had this issue, related to renaming devices, and what solved it for me was stopping and starting the ESPHome addon within Home Assistant.

csnyder616, 2022-07-19:
Restarting the add-on fixed this issue for me!

hado010, 2022-10-02:
Restart of the ESPhome add-on also fixed the problem for me. Thanks for the tip!

bcutter, 2022-12-30:
Restarting the addon does not "fix the issue", it's just a workaround. On the next adoption/onboarding the same problem will occur again, so the initial issue still exists.

See https://community.home-assistant.io/t/ignore-esphome-discovered-devices/401053 for a bunch of another folks being curious about this.

Especially for newbies this is a very confusing behaviour, as one might think "setup is obviously still not finished yet" and adopt the same ESP over and over again.

CMDR-Sloma, 2024-03-31:
Restarting ESPhome does nothing for me, rebooting Home Assistant which restarts also the addons does nothing for me.
Not only that, this device is showing as offline most of the time despite working fine, which I can verify by checking live logs.
[image]

I also have Everything Home Presence sensors which shows as offline in ESPhome all the time, but they are actually online and working fine in HA. I think these issue might be related.