esphome/issues #3394: Incompatible pointer warnings

tomlut, 2022-06-23:
### The problem

Incompatible pointer warnings in compile log. 

### Which version of ESPHome has the issue?

2022.6.2 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.6.7

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

esp32-hal-timer.c.o

### Example YAML snippet

```yaml
time:
  - platform: homeassistant
    id: homeassistant_time
```

### Anything in the logs that might be useful for us?

```txt
Compiling /data/rumpus-dehumidifier-conrol/.pioenvs/rumpus-dehumidifier-conrol/FrameworkArduino/esp32-hal-timer.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
```

mmakaay, 2022-06-23:
Those are warnings outside the ESPHome codebase, so we cannot fix those.

tomlut, 2022-06-23:
Ah I see. Sorry.

closed 2022-06-23 by tomlut

mmakaay, 2022-06-23:
Duplicate of #3162