esphome/issues #3396: ModuleNotFoundError: No module named 'platformio.commands.run'

nmg196, 2022-06-23:
### The problem

I'm trying to run `esphome run lr_mhi_ac_ctrl.yaml`  from this project https://github.com/ginkage/MHI-AC-Ctrl-ESPHome

However I get an error which says `ModuleNotFoundError: No module named 'platformio.commands.run'`

I noticed when installing platform IO I got a message in red which says: `esphome 2022.6.2 requires platformio==5.2.5, but you have platformio 6.0.2 which is incompatible`:

[image]

So do I somehow need to install an older version of platformio for esphome to work? Or is something else wrong?

Thanks!

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

D1 Mini

### Anything in the logs that might be useful for us?

```txt
C:\Projects\MHI-AC-Ctrl-ESPHome>esphome run lr_mhi_ac_ctrl.yaml
←[32mINFO Reading configuration lr_mhi_ac_ctrl.yaml...←[0m
←[33mWARNING 'lr_mhi_ac_ctrl': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name←[0m
←[32mINFO Generating C++ source...←[0m
←[32mINFO Compiling app...←[0m
Traceback (most recent call last):
  File "C:\Users\Nick Gilbert\.platformio\python3\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Nick Gilbert\.platformio\python3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Nick Gilbert\.platformio\penv\Scripts\esphome.exe\__main__.py", line 7, in <module>
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\__main__.py", line 918, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\__main__.py", line 373, in command_run
    exit_code = compile_program(args, config)
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\__main__.py", line 190, in compile_program
    rc = platformio_api.run_compile(config, CORE.verbose)
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\platformio_api.py", line 106, in run_compile
    return run_platformio_cli_run(config, verbose)
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\platformio_api.py", line 102, in run_platformio_cli_run
    return run_platformio_cli(*command, **kwargs)
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\platformio_api.py", line 93, in run_platformio_cli
    patch_structhash()
  File "C:\Users\Nick Gilbert\.platformio\penv\lib\site-packages\esphome\platformio_api.py", line 23, in patch_structhash
    from platformio.commands.run import helpers, command
ModuleNotFoundError: No module named 'platformio.commands.run'

C:\Projects\MHI-AC-Ctrl-ESPHome>
```

heschlie, 2022-06-23:
I ran into this yesterday, was just coming to make an issue to report on it! You need to downgrade your platformio install to 5.2.5, esphome will need to update their side to be compatible with 6.x+.

I did this by simply uninstalling platformio, and then pip installing with the specific version, I think `pip install -U platformio==5.2.5` will do it?

vvrockhound, 2022-06-23:
I just ran into this today. I tired backing out platformio to 5.2.5 and now I get another error.

`Error: Traceback (most recent call last):
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/jim/.local/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/jim/.local/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/jim/.local/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jim/.local/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/jim/.local/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/jim/.local/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/commands/run/command.py", line 134, in cli
    process_env(
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/commands/run/command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/commands/run/processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/commands/platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/platform/factory.py", line 52, in new
    pkg = PlatformPackageManager().get_package(pkg_or_spec)
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/package/manager/base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/package/meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/package/meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "/home/jim/.local/lib/python3.10/site-packages/platformio/package/meta.py", line 393, in load
    data["spec"] = PackageSpec(**data["spec"])
TypeError: PackageSpec.__init__() got an unexpected keyword argument 'uri'`

MikyM0use, 2022-06-24:
same problem here... update the platformio framework leads to "module not found" error, while downgrading to 5.2.5 shows me same error of @vvrockhound .

my workaround for the moment is to run with:
`docker run --rm -v "${PWD}":/config -it esphome/esphome run FILENAME.yaml`

mmakaay, 2022-06-24:
ESPHome currently uses platformio 5.2.5. The platformio 6 version changed its API in an incompatible way, which causes the first issue that you ran into.

The second issue is because of the downgrade which doesn't work fully. When I played with that, I simply deleted all files from `.platformio/` in my home directory, to get rid of any stuff that v6 might have left behind. After that, I could compile the firmware again using 5.2.5.

nmg196, 2022-06-24:
Thanks! That fixed the compilation.  It flashed once but had a load of errors (see below) so I though the board type was wrong in the yaml - I had d1_mini instead of d1_mini_pro (which I presume is compatible?).  But reflashing with d1_mini_pro set (no other changes) results in the board not being contactable on port 80.  It pings but the pings are very random and latency is high.  The serial logs (which only appeared the very first time) has loads of loop errors like this:

`[16:14:04][W][mhi_ac_ctrl:077]: mhi_ac_ctrl_core.loop error: -4`

[image]

I'm pretty close to throwing in the towel :(

mmakaay, 2022-06-24:
I'm not familiar with the specific component. It's loop code contains a whopping 400 lines of code. If there's code in there that is hogging the CPU for too long, it might very well interfere with the WiFi connection and webserver performance.

A quick glance at the code shows that the error indicates that the SPI SCK pin is stuck at HIGH. The device's loop code is waiting for that pin to go to a LOW state. I think that the code now is mainly waiting for the pin to go LOW, because the timeout for that is 100ms. When that timeout is reached, the error is logged and the device can continue running the main loop code to give other pieces of code some time to work (e.g. the webserver). But in the next round, the mhi_ac_ctrl loop will again block everything for 100ms.

One thing to check would be the wiring. If a pin is not dropping to LOW, maybe some wire connection is pulling it HIGH.

Since this is not a component that is provided by the ESPHome project directly, it's hard to give specific advice here. Maybe there are others that have experience with this hardware and implementation, but I think the best place to find support would be the original repository where this code came from.

nmg196, 2022-06-24:
There is no wiring :(   It's a bare board sat on my desk.    Perhaps it's faulty or damaged or something. Should a Wemos D1 mini pro theoretically work though? What boards do people most commonly use for ESPHome?   

I just wanted one which could take an external antenna.

mmakaay, 2022-06-24:
Yes, when there's no SPI device hooked up, then I can see why there are timeouts while talking to it.

There's not much to test the bare device. The only way to do this without installing the D1 mini, would be when there were a simulation device that you could hook up to the D1 mini and that could simulate various cases of the real unit. Not that feasible I presume.

gerdur, 2022-06-25:
As a esphome novice I was just running into the same problem (using new platformio version 6.0.2 and I will avoid to go back to an earlier version). Is there an official esphome update planned to support platformio v6.x.x ? Thanks

mmakaay, 2022-06-25:
Yeah, there is. See this pull request: https://github.com/esphome/esphome/pull/3566

Fretwire, 2022-07-03:
I was about to say downgrading to platformio 6.0.1 seems to work but majuss' suggestion to alter the line from .run to .lib does the trick with pio 6.0.2
Filename on Linux is ~/.local/lib/python3.10/site-packages/esphome/platformio_api.py

gerdur, 2022-07-03:
Sorry for my (maybe stupid) question: The last days I was working with ESPHome from HomeAssistant and everything worked fine. My first ESP32 Multi-Sensor device ist working properly together with HomeAssistant. 
Working with ESPHome on my Windows-PC from the CMD-Line is still a problem together with PlatformIO 6.0.2. 
What exactly do you mean with _"Just alter the line from ".run" to ".lib" everything works fine" ?
I'm an embedded developer since many years, but on a lot of other platforms. Until now, I only have minor experience  with ESP32. Many thanks for your clarification :-)

Fretwire, 2022-07-03:
I don't have access to a windows box so can't help with that specific OS but if you find the file "platformio_api.py" installed by ESPHome then on (or around) line 23 of that file you'll see:

from platformio.commands.run import helpers, command

Just change that to:

from platformio.commands.lib import helpers, command

mmakaay, 2022-07-03:
The actual issue is that ESPHome currently uses PlatformIO 5.2.5. This will be updated in a future release. If you're not using PlatformIO locally for other projects, it's probably easier to downgrade to that version of PlatformIO, instead of trying to hack the ESPHome code to talk correctly to PlatformIO 6.

majuss, 2022-07-04:
If you're using PIO 5 you're getting these errors: https://github.com/esphome/issues/issues/3357

majuss, 2022-07-04:
When I use PIO 5 I get all of these errors mentioned here with any board I try: https://github.com/esphome/issues/issues/3357

mmakaay, 2022-07-04:
In highly specific cases you are getting those errors. It's not a general use case.
Multiple changes will have to be done to get Adafruit Feather ESP32 V2 working.
And PRs for that are in the making.

In general I really don't recommend using PlatformIO v6, unless you really need it and are comfortable with hacking up the local installs for that.

mivade, 2022-07-06:
Regarding the `TypeError: PackageSpec.__init__() got an unexpected keyword argument 'uri'` message, that can happen if you have PlatformIO 6 installed (which some of us use e.g. via their IDE for other projects). A workaround so that you don't have to delete your `~/.platformio` directory is to set the `PLATFORMIO_CORE_DIR` environment variable in a Python virtual environment with `esphome` installed. For example:

```
$ PLATFORMIO_CORE_DIR=/tmp/platformio esphome run foobar.yaml
```

jesserockz, 2022-07-06:
I am closing this issue as platformio 6 is already merged into ESPHome **dev** branch. It will be released with ESPHome **2022.8.0**

See https://github.com/esphome/esphome/pull/3566

This error: `TypeError: PackageSpec.__init__() got an unexpected keyword argument 'uri'` happens when you try to run with platformio 6, which "upgrades" the local library cache and is not backwards compatible with 5.2.5

closed 2022-07-06 by jesserockz

ziolelle, 2022-07-11:
> PLATFORMIO_CORE_DIR

Great fix !!!! thank you !!!

I had the problem of having PlatformIO IDE in VSCode and esphome cli that up to now are incompatible.

Now without deleting the `~/platformio` I simply use a command like:
`PLATFORMIO_CORE_DIR=/Users/danielemetta/.platformiocli esphome compile esp32_climate.yaml`

and I can make both of them work !

really thank you

git-agil, 2022-08-08:
> `pip install -U platformio==5.2.5` will do it?
No, not for me.
I had to resolve all dependencies by downgrading all incompatible versions:
after running 
$ pip uninstall platformio; pip install platformio==5.2.5
I still had these:
_esphome 2022.5.1 requires colorama==0.4.4, but you have colorama 0.4.5 which is incompatible.
esphome 2022.5.1 requires zeroconf==0.38.4, but you have zeroconf 0.38.7 which is incompatible._
then, after executing
_$ pip uninstall colorama; pip install colorama==0.4.4
$ pip uninstall zeroconf; pip install zeroconf==0.38.4_
esphome compiled ok again :-)