esphome/issues #3397: ESP32_CAMERA_SIZE_2592X1944 in esp32 camera is missing

wildblue, 2022-06-24:
### The problem

Can not operate camera in full resolution

### Which version of ESPHome has the issue?

2022.06

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.06

### What platform are you using?

ESP32

### Board

Esp32cam

### Component causing the issue

Esp32-camera

### Example YAML snippet

```yaml
I try to attach a zip file...
```

wildblue, 2022-06-24:
Okay, file attached. These changes are working for me.

I created a version for you, so you can integrate the changes directly.

I also noted changes and included the old versions.

ssieb, 2022-10-25:
Why didn't you make a PR if you have the changes?