esphome/issues #3400: adc_gpio_init compilation error with 2022.7.0 and plateform 4.0.4

SeByDocKy, 2022-06-24:
### The problem

When upgrading both arduino lib to 2.0.3 and pio for 4.4.0 got a compilation error (adc_gpio_init error)  with any ADC sensor

### Which version of ESPHome has the issue?

2022.7.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.x

### What platform are you using?

ESP32

### Board

WROOM32

### Component causing the issue

ADC

### Example YAML snippet

```yaml
:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0  

sensor: 
   
  - platform: adc
    id: sensor_resistance
    pin: GPIO032
    update_interval: 10s
    attenuation: 11db
```

### Anything in the logs that might be useful for us?

```txt
Compiling app...
Processing efs (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 4.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- Wire @ 2.0.0
Compiling /data/efs/.pioenvs/efs/src/esphome/components/adc/adc_sensor.cpp.o
Compiling /data/efs/.pioenvs/efs/src/esphome/components/api/api_connection.cpp.o
Compiling /data/efs/.pioenvs/efs/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/efs/.pioenvs/efs/src/esphome/components/api/api_pb2.cpp.o
src/esphome/components/adc/adc_sensor.cpp: In member function 'virtual void esphome::adc::ADCSensor::setup()':
src/esphome/components/adc/adc_sensor.cpp:67:3: error: 'adc_gpio_init' was not declared in this scope
   adc_gpio_init(ADC_UNIT_1, (adc_channel_t) channel_);
   ^~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:67:3: note: suggested alternative: 'gpio_init'
   adc_gpio_init(ADC_UNIT_1, (adc_channel_t) channel_);
   ^~~~~~~~~~~~~
   gpio_init
*** [/data/efs/.pioenvs/efs/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
========================= [FAILED] Took 21.17 seconds =========================``
```

codersaur, 2022-06-29:
I'm seeing the same issue.

codersaur, 2022-07-06:
Courtesy of @ssieb, this can be fixed by editing adc_sensor.cpp. Just add  ` && !defined(USE_ESP32_VARIANT_ESP32S3)` to the line at the send of the setup() method:

`#if !defined(USE_ESP32_VARIANT_ESP32C3) && !defined(USE_ESP32_VARIANT_ESP32H2) && !defined(USE_ESP32_VARIANT_ESP32S2) **&& !defined(USE_ESP32_VARIANT_ESP32S3)**`

Unfortunately, I'm not well-versed with creating pull requests, so I'll leave it to someone more capable to do that bit...

ssieb, 2022-07-25:
From https://www.esp32.com/viewtopic.php?t=25244 it looks like that line can just be removed for all variants.
`adc1_config_channel_atten()` is always called.

closed 2022-07-31 by jesserockz