esphome/issues #3401: Dallas sensors no more scanned with 2022.7.0 and arduino 2.0.3 & platform 4.0.4

SeByDocKy, 2022-06-24:
### The problem

Compiling with this new setup, my dallas sensors are nor more scanned. Works perfectly with default options (arduino lib & platform)

### Which version of ESPHome has the issue?

2022.7.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.x

### What platform are you using?

ESP32

### Board

ESP32u

### Component causing the issue

dallas sensor

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0  

dallas:
  - pin: GPIO02
    update_interval: 10s

sensor:         
  - platform: dallas
    address: 0x05811851B3B2FF28
    name: "ecs_temperature"
    id: ecs_temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 2
    filters:
     - offset: 5.0
```
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/ecs.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Paris'
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing ecs (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 4.4.0)
[... 358 lines cut ...]
[18:49:31][D][sensor:125]: 'water_flow_meter_ecs': Sending state 0.00000 l with 2 decimals of accuracy
[18:49:31][D][sensor:125]: 'ecs_du_jour': Sending state 0.96411 l with 2 decimals of accuracy
[18:49:31][D][sensor:125]: 'WiFi puissance_ecs': Sending state -56.00000 dBm with 0 decimals of accuracy
[18:49:32][D][pulse_counter:162]: 'water_flow_meter_ecs': Retrieved counter: 0.00 pulses/min
[18:49:32][D][sensor:125]: 'water_flow_meter_ecs': Sending state 0.00000 l with 2 decimals of accuracy
[18:49:32][D][sensor:125]: 'ecs_du_jour': Sending state 0.96411 l with 2 decimals of accuracy
[18:49:33][D][pulse_counter:162]: 'water_flow_meter_ecs': Retrieved counter: 0.00 pulses/min
[18:49:33][D][sensor:125]: 'water_flow_meter_ecs': Sending state 0.00000 l with 2 decimals of accuracy
[18:49:33][D][sensor:125]: 'ecs_du_jour': Sending state 0.96411 l with 2 decimals of accuracy
[18:49:33][D][sensor:125]: 'temperature_sdb_haut': Sending state 31.40000 °C with 1 decimals of accuracy
[18:49:33][D][sensor:125]: 'pression_sdb_haut': Sending state 1.00027 bar with 3 decimals of accuracy
[18:49:33][D][sensor:125]: 'humidite_sdb_haut': Sending state 39.37598 % with 1 decimals of accuracy
[18:49:34][D][pulse_counter:162]: 'water_flow_meter_ecs': Retrieved counter: 0.00 pulses/min
[18:49:34][D][sensor:125]: 'water_flow_meter_ecs': Sending state 0.00000 l with 2 decimals of accuracy
[18:49:34][D][sensor:125]: 'ecs_du_jour': Sending state 0.96411 l with 2 decimals of accuracy
[18:49:34][D][sensor:125]: 'bec_puissance_cumulee': Sending state nan kWh with 1 decimals of accuracy
[18:49:35][D][pulse_counter:162]: 'water_flow_meter_ecs': Retrieved counter: 0.00 pulses/min
[18:49:35][D][sensor:125]: 'water_flow_meter_ecs': Sending state 0.00000 l with 2 decimals of accuracy
[18:49:35][D][sensor:125]: 'ecs_du_jour': Sending state 0.96411 l with 2 decimals of accuracy
[18:49:36][D][pulse_counter:162]: 'water_flow_meter_ecs': Retrieved counter: 0.00 pulses/min
[18:49:36][D][sensor:125]: 'water_flow_meter_ecs': Sending state 0.00000 l with 2 decimals of accuracy
[18:49:36][D][sensor:125]: 'ecs_du_jour': Sending state 0.96411 l with 2 decimals of accuracy
[18:49:37][W][dallas.sensor:137]: 'ecs_temperature' - Scratch pad checksum invalid!
```
```

jenscski, 2022-07-25:
I have the same problem, and a little investigating gave me that Arduino 2.0.2 works, 2.0.3 does not.
And probably it's this change in Arduino that's the cause, https://github.com/espressif/arduino-esp32/pull/6259

I'm guessing the one wire implementation in ESPHome need to be updated in some way.

I may have time to look into it, but probably way over my knowledge ...

jenscski, 2022-07-28:
I think I have found the problem. The code uses pinMode extensively to communicate with the one wire devices.

I created a test program using Platform IO and fount out that pinMode takes ~16us on Arduino 2.0.3 and later, but on previous version it only took ~4us. This is probably the reason for dallas not working with Arduino 2.0.3.

StijnVdd, 2022-08-30:
Having a similar issue. Sensor no longer found after upgrading from one ESPHome version to another

closed 2023-01-21 by github-actions[bot]