esphome/issues #3403: Simple esphome config, random light toggles?

dave, 2022-06-25:
### The problem

I have a M5Stack Atom Light connected to an M5Stack PWM module. I've flashed it with the simplest esphome config I can come up with, which should just toggle the light when I press the button on the Atom. It works, but randomly (approximately every half hour) it toggles the light without any action. 

Is the config file below correct? Is there any way it could be improved / simplified? How would I go about debugging this? How would I go about telling if it's a bug with the Atom or with esphome?

Do these M5Stack devices have a reputation for being flaky like this? I also have [this issue open](https://github.com/esphome/issues/issues/3341) where this device reboots when lots of MQTT messages are received at once... 🤷‍♂️

### Which version of ESPHome has the issue?

Version: 2022.5.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

M5Stack Atom

### Component causing the issue

not sure

### Example YAML snippet

```yaml
substitutions:
  device_name: light1
  friendly_name: Light 1

esphome:
  name: ${device_name}

esp32:
  board: m5stack-atom
  framework:
    type: arduino

output:
  - platform: ledc
    pin: GPIO22
    id: gpio_22
    frequency: 1220Hz

light:
  - platform: monochromatic
    output: gpio_22
    name: ${friendly_name} Light
    id: light_id

binary_sensor:
  - platform: gpio
    pin:
      number: 39
      inverted: true
    name: ${friendly_name} Button
    on_press:
      then:
        - light.toggle:
            id: light_id
            transition_length: 0s

logger:
  level: NONE

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback Hotspot"
    password: "hogxyvZGcsFa"

captive_portal:
```

bkaufx, 2022-06-25:
I would guess noise in the button.  Do you see the button being pressed?  Add a delayed_on filter on the binary sensor of around 50-100ms and see if that helps.   You might also add an uptime sensor to monitor whether the device is resetting itself.  Or leave logs up and you should be able to see what is happening.

https://esphome.io/components/binary_sensor/index.html#delayed-on
https://esphome.io/components/sensor/uptime.html

dave, 2022-06-25:
Hmm it's a clicky button that takes quite a push to actuate, so I would be surprised if it was triggering with noise... but perhaps. I'll enable the logs and see if it's registering a button press...

bkaufx, 2022-06-25:
It's not really about the physical characteristics of the button.  I'm not proposing that the button is physically pushed on its own.  The wire to the button acts like an antenna and picks up interference.  If that gets great enough the ESP can receive a logic level change even though nothing physically happened with the button.

Now that I"m looking at it, you probably just need to enable the pullup resistor in the input.  https://esphome.io/components/binary_sensor/gpio.html (search pullup: true)

dave, 2022-06-25:
OK I get it... I'll run some more tests.

AdamGit69, 2022-07-06:
Not sure if you ever worked this out however after lots of research, I found it was wifi, particularly when the M5Stack Atom Lite wifi goes to sleep, when it randomly wakes again it would send phantom button presses. Turning the light RGB light on or off or toggling anything that you had setup to work with the button press… It did my head in for ages and hours of google didn’t help much. Finally though after watching the logs and what the atom lite was doing on my network, I worked it out so I thought I’d come back here and share as this was the first post google brought me too…

To fix, add this line under your wifi settings to your esphome yaml:

power_save_mode: none

so your yaml would be something like below

```
wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
power_save_mode: none
```

and that’s it, no more phantom button presses. :slight_smile:

dave, 2022-07-06:
This is interesting... I actually upgraded to the latest esphome and set up logging to see if I could catch the phantom events, and they totally stopped. I was planning to downgrade esphome to the previous version to make sure I could isolate that as the problem but I haven't had any time to work on the project...

closed 2022-11-19 by github-actions[bot]