esphome/issues #3405: ESPHome missing support for important device_class

DrCoolzic, 2022-06-25:
### The problem

Currently ESPHome does not support two very important device classes: “apparent_power” and “reactive_power”. This is very annoying because currently **there is no way for a sensor to report values with VA or VAR** for unit_of_measurement. 
If using device_class=power these units of measurement are rejected for statistical data recording …

In the Sensor Component documentation, it is mentioned that:

> device_class (Optional, string): The device class for the sensor. See https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes for a list of available options.

But as mentioned two are missing: apparent_power and reactive_power

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.7

### What platform are you using?

ESP8266

### Board

irrelevant

### Component causing the issue

irrelevant

### Example YAML snippet

```yaml
- platform: teleinfo
    tag_name: "SINSTS"
    name: "linky_sinsts"
    device_class: "apparent_power"
    state_class: "measurement"
    unit_of_measurement: "VA"
```

### Anything in the logs that might be useful for us?

```txt
sensor.teleinfo: [source /config/esphome/linky-tic.yaml:161]
  platform: teleinfo
  tag_name: SINSTS
  name: linky_sinsts
  
  Unknown value 'apparent_power', valid options are '', 'aqi', 'battery', 'carbon_dioxide', 'carbon_monoxide', 'current', 'duration', 'energy', 'gas', 'humidity', 'illuminance', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'pm1', 'pm10', 'pm25', 'power', 'power_factor', 'pressure', 'signal_strength', 'sulphur_dioxide', 'temperature', 'timestamp', 'volatile_organic_compounds', 'voltage'.
  device_class: apparent_power
  state_class: measurement
  unit_of_measurement: VA
sensor.teleinfo: [source /config/esphome/linky-tic.yaml:168]
  platform: teleinfo
  tag_name: SINSTS1
  name: linky_sinsts1
```

### Additional information

If trying to use power instead of apparent_power
[image]

we get error
[image]

Please refer to
Add sensor device classes for apparent and reactive power #688 https://github.com/home-assistant/architecture/discussions/688

and 
Device class “power” should support “VA” and “VAR” units #57236 https://github.com/home-assistant/core/issues/57236

for more info

DrCoolzic, 2022-07-17:
Hey there anybody looking at this problem?
Badly needed for the Teleinfo component!
Thanks

DrCoolzic, 2022-08-09:
Can someone look at this problem. It should not be a big modification to add the two missing device classes :)

jesserockz, 2022-08-11:
These were added to `dev` in https://github.com/esphome/esphome/pull/3624

closed 2022-08-11 by jesserockz

DrCoolzic, 2022-08-11:
Many thanks. 
Is this update available on 2022.6.3 ?