esphome/issues #3406: mqtt on_message: handling misses retained messages on wakeup/reboot.

dbaarda, 2022-06-26:
### The problem

There are several problems that might be interacting.

1. ESPHome subscribes to the same topic multiple times when there are multiple `- topic:` entries for the same topic with different `payload:` matches. This triggers the mqtt server to send any retained mqtt message for that topic multiple times, once for each subscribe request.
2. Only the last `-topic:` entry receives and responds to retained messages on wakeup/reboot.

In this test case I've created 3 `-topic: ${devicename}/log_level` entries with different `payload:` entries that only do logging, and 2 `-topic: ${devicename}/sleep_mode` entries with different `payload:` entries for toggling deep_sleep enabled/disabled. I've then created retained messages for both topics that match one of the payload entries for each topic with (note my -h -u -P args are in my ~/.config/mosquito_pub);

```
$ mosquitto_pub -t underhouse/sleep_mode -m "OFF" -r
$ mosquitto_pub -t underhouse/log_level -m "DEBUG" -r
```

I then upload the included yaml file and collect logs. The logs show the multiple subscriptions for both topics, two responses to the last topic entry subscriptions, and no response to the earlier topic subscriptions.

### Which version of ESPHome has the issue?

2022.6.0b4

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

Adafruit QT Py ESP32-C3

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
substitutions:
  devicename: underhouse

esphome:
  name: ${devicename}
  platformio_options:
    board_build.flash_mode: dio

esp32:
# At some point the actual board might be supported.
#  board: adafruit_qt_py_esp32-c3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
  variant: ESP32C3

# Enable logging
logger:
  level: VERY_VERBOSE

ota:
  password: "1381b044bfad9aeca468373c8a1fa20b"

mqtt:
  id: mqtt_client
  broker: 192.168.7.6
  username: "mqtt"
  password: "mqtt iot!"
  discovery: true
  discovery_retain: true
  discovery_unique_id_generator: mac
[... 184 lines cut ...]
[16:49:25][VV][scheduler:196]: Running timeout '' with interval=1000 last_execution=6817 (now=7817)
[16:49:25][V][wifi_esp32:646]: Event: Got IPv6 address=fe80:0000:0000:0000:1291:a8ff:fe33:8184
[16:49:59][VV][scheduler:196]: Running interval '' with interval=60000 last_execution=4294949652 (now=42360)
[16:50:07][VV][scheduler:196]: Running interval 'update' with interval=60000 last_execution=4294956969 (now=49679)
[16:50:07][V][sensor:074]: 'system uptime': Received new state 49.681000
[16:50:07][D][sensor:125]: 'system uptime': Sending state 49.68100 s with 0 decimals of accuracy
[16:50:07][V][mqtt:415]: Publish(topic='underhouse/sensor/system_uptime/state' payload='50' retain=1)
[16:50:13][VV][scheduler:196]: Running interval 'update' with interval=60000 last_execution=4294962802 (now=55508)
[16:50:13][V][sensor:074]: 'wifi strength': Received new state -82.000000
[16:50:13][D][sensor:125]: 'wifi strength': Sending state -82.00000 dBm with 0 decimals of accuracy
[16:50:13][V][mqtt:415]: Publish(topic='underhouse/sensor/wifi_strength/state' payload='-82' retain=1)
[16:50:59][VV][scheduler:196]: Running interval '' with interval=60000 last_execution=42356 (now=102361)
[16:51:07][VV][scheduler:196]: Running interval 'update' with interval=60000 last_execution=49673 (now=109677)
[16:51:07][V][sensor:074]: 'system uptime': Received new state 109.680000
[16:51:07][D][sensor:125]: 'system uptime': Sending state 109.68000 s with 0 decimals of accuracy
[16:51:07][V][mqtt:415]: Publish(topic='underhouse/sensor/system_uptime/state' payload='110' retain=1)
[16:51:13][VV][scheduler:196]: Running interval 'update' with interval=60000 last_execution=55506 (now=115508)
[16:51:13][V][sensor:074]: 'wifi strength': Received new state -82.000000
[16:51:13][D][sensor:125]: 'wifi strength': Sending state -82.00000 dBm with 0 decimals of accuracy
[16:51:13][V][mqtt:415]: Publish(topic='underhouse/sensor/wifi_strength/state' payload='-82' retain=1)
```

### Additional information

Note these problems caused a fair bit of confusion and workarounds that I've documented at https://community.home-assistant.io/t/success-esphome-esp32-c3-bme280-mqtt-and-deep-sleep-with-homeassistant/432067/2?u=dbaarda

dbaarda, 2022-07-26:
Note in the logs there are three of these;
```
[16:49:24][V][mqtt.idf:111]: MQTT_EVENT_DATA underhouse/sleep_mod
```
Before two of these;
```
[16:49:24][V][mqtt.idf:111]: MQTT_EVENT_DATA underhouse/sleep_mode
```
Note the missing 'e' on the first ones. I don't fully understand how, but I suspect these first three are generated for responses to the 3 subscriptions to 'underhouse/log_level', but some how the messages are being badly parsed with corrupt topics that are somehow truncated version of the last 2 topics.

I've started digging into the code but I've not figured it out yet... the bit where these are logged and the on_message callback is made is here;

https://github.com/esphome/esphome/blob/c95acd25681c35ee765c4ae88f8ea131ff45e091/esphome/components/mqtt/mqtt_backend_idf.cpp#L111

It looks a bit suss at first glance, with the `static std:string topic'` only populated if the event.topic is set, but digging into the platformio code that defines `esp_mqtt_event_t` it seems that's because message data can be split over multiple events, and the topic is only included in the first one, with the subsequent events containing the rest of the data without topic set. The other dodgey looking bit is setting `data_len` to the `strlen(event.data)` if it's zero; I'm not sure why it does this, and wonder if its possible to legitimately have an event with zero data because its all coming in followup events or is genuinely empty. The rest of the code seems to think event data (or at least the topic) is not zero-terminated, so could this overrun or eat following data?

Other than that, it looks OK so next I'll dig around to see if invalid events could be getting fed into here.

dbaarda, 2022-07-26:
OH yeah, another interesting observation; all 5 of these `MQTT_EVENT_DATA` events are eventloop event_id=6, as indicated by the earlier line of logging before all of them;
```
[16:49:24][V][mqtt.idf:073]: Event dispatched from event loop event_id=6
```
Which makes me thing that all 5 of these are packed into the same incoming packet, increasing the possibility bad parsing could make them mess each other up.

dbaarda, 2022-07-26:
Ugh, the event_id is just the event type used to select the event handler in the case statement. It's not some kind of event count, so forget my last observation.

dbaarda, 2022-08-03:
Another interesting observation: The three first subscription `MQT_EVENT_DATA` events have topic `sleep_mod` which is the right length for the `log_level` subscription messages, but the wrong topic string. This suggests the event callback is being called with an invalid topic string pointer, which for some reason is not pointing at the `log_level` topic string for the first 3 messages, but the `sleep_mode` topic string for the last message.

dbaarda, 2022-12-03:
Err... This problem still exists. I might get a chance to debug deeper over the holidays.

aaliddell, 2022-12-29:
I have added a PR that I believe fixes this at https://github.com/esphome/esphome/pull/4239

If you could test and report if this fixes it for you, that'd be really useful. You should just be able to do:

```yaml
external_components:
  - source: github://pr#4239
    components: [ mqtt ]
```

dbaarda, 2023-01-01:
That esphome/esphome#4239 definitely fixes the problem of only the last `topic:` entry working on wakeup. This is nice because it means you don't have to turn off sleep_mode before you can toggle any other subscribed topics.

It doesn't fix the problem of multiple `topic:` entries for the same topic with different `payload:` matches giving multiple subscribe messages and responses for that topic. However, this is a relatively minor problem that just means extra mqtt chatter, and it can be worked around by only using one `topic:` entry per topic with conditional behaviour in a lambda.

aaliddell, 2023-01-01:
Great! I’ll see if I can address the other point too, it sounds like it’s just a case of deduplicating subscribes.

closed 2023-01-11 by jesserockz