esphome/issues #3408: ESP32C3 missing package

UniKyrn, 2022-06-26:
### The problem

When Trying to adopt a LilyGo ESP32C3 into ESPHome, it fails because of a missing package.

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

LILYGO® TTGO T-OI Plus ESP32-C3 RISC-V MCU Wireless Module Development Circuits Board

### Component causing the issue

toolchain

### Example YAML snippet

```yaml
esphome:
  name: esp32c3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp32c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32c3 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'
```

UniKyrn, 2022-06-28:
> This config got my Lilygo module working - had to install it plugged into
> the Pi's USB port at first, but now works OTA:
>
> substitutions:
>   display_name: 'ESP32 C3 Test'
>   device_description: 'ESP32 C3 Test for the Lilygo ESP01 board'
>   friendly_name: 'ESP32 C3 Test'
>   device_name: esp32c3-test-new
>   node_name: esp32c3_test_new
>
> esphome:
>   name: ${device_name}
>   platformio_options:
>     board_build.flash_mode: dio
>   comment: ${device_description}
>   on_boot:
>     - priority: 800.0
>       then:
>         - component.update: ${node_name}_uptime_raw
>         - component.update: wifisignal
>
> esp32:
>   board: esp32-c3-devkitm-1
>   framework:
>     type: esp-idf
> [... 95 lines cut ...]
>                 (days ? to_string(days) + "d " : "") +
>                 (hours ? to_string(hours) + "h " : "") +
>                 (minutes ? to_string(minutes) + "m " : "") +
>                 (to_string(seconds) + "s")
>               ).c_str();

The error changes, the key seems to be the Platform Version, but it still
is missing a package.  Are you on a Dev Branch by chance?

UniKyrn, 2022-06-29:
I'm trying to compile using ESPHome 2022.6.2 inside my HA install on a RasPi4.  I never get to the install stage, it won't compile to even try and install.  If it works for you, then your system must have the packages somehow, while mine doesn't.

chunkysteveo, 2022-06-29:
The only thing I can see that may be different is I am running on a 32bit RPi 3, whereas you are on a Pi 4 and on 64bit.... Maybe that is the key to the issue?

UniKyrn, 2022-06-29:
Could be, which is why I think it is an ESPHome issue.  The toolchains they want to install are not correct for all the various architecture types.

TylerKostuch, 2022-07-09:
Well, I've stumbled into the same exact issue. Changing the platform version to `3.3.2` gives me the same error @UniKyrn saw.

I'll be watching this issue for updates and if I discover anything, I'll post it here.

UniKyrn, 2022-07-09:
Thank's!  I still think it's just a config problem, the link to that arch
library using a Pi at the compile platform is just wrong.

TylerKostuch, 2022-07-09:
Well, I got pass that error, but I now have a new error 😅 

```
esphome:
  name: test-display
  platformio_options:
    platform_packages:
     - framework-espidf @ 3.40300.0

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    platform_version: 3.4.0
```

But now, I am getting the following:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:2713:60: error: invalid new-expression of abstract class type 'esphome::lilygo_t5_47_display::LilygoT547Display'
   t5_display = new lilygo_t5_47_display::LilygoT547Display();
                                                            ^
In file included from src/esphome.h:29,
                 from src/main.cpp:3:
src/esphome/components/lilygo_t5_47_display/LilygoT547Display.h:18:7: note:   because the following virtual functions are pure within 'esphome::lilygo_t5_47_display::LilygoT547Display':
 class LilygoT547Display : public PollingComponent, public display::DisplayBuffer {
       ^~~~~~~~~~~~~~~~~
In file included from src/esphome.h:21,
                 from src/main.cpp:3:
src/esphome/components/display/display_buffer.h:373:23: note: 	'virtual esphome::display::DisplayType esphome::display::DisplayBuffer::get_display_type()'
   virtual DisplayType get_display_type() = 0;
                       ^~~~~~~~~~~~~~~~
```

DanteBC, 2022-07-09:
> Well, I got pass that error [with this](https://github.com/esphome/issues/issues/3076#issuecomment-1179194694), but I now have a new error 😅
> 
> ```
> esphome:
>   name: test-display
>   platformio_options:
>     platform_packages:
>      - framework-espidf @ 3.40300.0
> 
> esp32:
>   board: esp-wrover-kit
>   framework:
>     type: esp-idf
>     platform_version: 3.4.0
> ```
> 
> But now, I am getting the following:
> 
> ```
> src/main.cpp: In function 'void setup()':
> src/main.cpp:2713:60: error: invalid new-expression of abstract class type 'esphome::lilygo_t5_47_display::LilygoT547Display'
>    t5_display = new lilygo_t5_47_display::LilygoT547Display();
>                                                             ^
> In file included from src/esphome.h:29,
>                  from src/main.cpp:3:
> src/esphome/components/lilygo_t5_47_display/LilygoT547Display.h:18:7: note:   because the following virtual functions are pure within 'esphome::lilygo_t5_47_display::LilygoT547Display':
>  class LilygoT547Display : public PollingComponent, public display::DisplayBuffer {
>        ^~~~~~~~~~~~~~~~~
> In file included from src/esphome.h:21,
>                  from src/main.cpp:3:
> src/esphome/components/display/display_buffer.h:373:23: note: 	'virtual esphome::display::DisplayType esphome::display::DisplayBuffer::get_display_type()'
>    virtual DisplayType get_display_type() = 0;
>                        ^~~~~~~~~~~~~~~~
> ```

Can you try commenting out anything that has to do with display in your config and try to compile again that way? just to see if it works.

ssieb, 2022-08-02:
The toolchain issue is because Espressif didn't provide a 64-bit ARM compile for that release.  @UniKyrn, that arch *is* correct for a pi if it's running a 64-bit OS.
The solution is what was found, you need to use a platform version that will get you a later toolchain release.

UniKyrn, 2022-08-02:
Version core-2022.7.7
Installation Type Home Assistant OS
Development false
Supervisor true
Docker true
User root
Virtual Environment false
Python Version 3.10.5
Operating System Family Linux
Operating System Version 5.15.32-v8
CPU Architecture aarch64
Timezone America/Los_Angeles
Configuration Directory /config
Ok, I'm on the 64 version ...

jysaloma, 2022-09-21:
I am having the same issue with ESPhome (2022.9.0) PI4 wireless updates

Processing espglow (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
INFO Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'

# Home Assistant Glow
#
# Read your electricity meter by means of the pulse LED on your 
# meter, useful if you do not have a serial port (P1).
#
substitutions:
  device_name: home-assistant-glow
  friendly_name: REMOVED
  project_version: "3.1.0"
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO9
  status_led: GPIO5

  # imp/kWh rate ⬇ #
  pulse_rate: '1000'

  # Webserver credentials ⬇ #
#  web_username: 'glow'
#  web_password: 'REMOVED'

dashboard_import:
  package_import_url: github://klaasnicolaas/home-assistant-glow/home_assistant_glow.yaml

esphome:
  name: espglow
  platformio_options:
    board_build.flash_mode: dio
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home-assistant-glow"
    version: "${project_version}"

esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf
    sdkconfig_options: 
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 120s
  
  manual_ip:
    static_ip: 10.1.1.133
    gateway: 10.1.1.254
    subnet: 255.255.255.0
    dns1: 10.1.1.254

#captive_portal:

# Enable logging
logger:

# Enable improv over serial
improv_serial:

# Enable Home Assistant API
api:
  services:
    - service: reset_total_energy
      then:
        - button.press:
            id: button_reset_total
  password: "REMOVED"

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5
  password: "REMOVED"

#web_server:
#  port: 80
#  auth:
#    username: '${web_username}'
#    password: '${web_password}'

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO4
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

button:
  - platform: restart
    name: "Restart - Glow"
  - platform: template
    name: "Reset - Total Energy"
    id: button_reset_total
    on_press:
      - pulse_meter.set_total_pulses:
          id: sensor_energy_pulse_meter
          value: 0

# Sensors for ESP version and WIFI information
text_sensor:
  # Installed version
  - platform: template
    name: "Glow - Installed version"
    id: glow_version
    icon: "mdi:label-outline"
    entity_category: diagnostic
    lambda: |-
      return {"${project_version}"};
    update_interval: 6h
  - platform: version
    hide_timestamp: true
    name: '${friendly_name} - ESPHome Version'
  - platform: wifi_info
    ip_address:
      name: '${friendly_name} - IP Address'
      icon: mdi:wifi
    ssid:
      name: '${friendly_name} - Connected SSID'
      icon: mdi:wifi-strength-2

sensor:
  # WiFi signal
  - platform: wifi_signal
    name: "${friendly_name} - WiFi Signal"
    update_interval: 120s

  # Pulse meter
  - platform: pulse_meter
    name: '${friendly_name} - Power Consumption'
    id: sensor_energy_pulse_meter
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    # internal_filter: 100ms
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);
    total:
      name: '${friendly_name} - Total Energy'
      id: sensor_total_energy
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});
  # Total day useage
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Energy'
    id: sensor_total_daily_energy
    power_id: sensor_energy_pulse_meter
    unit_of_measurement: 'kWh'
    icon: mdi:circle-slice-3
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001

# Enable time component to reset energy at midnight
# https://esphome.io/components/time.html#home-assistant-time-source
time:
  - platform: homeassistant
    id: homeassistant_time

ssieb, 2022-09-21:
@jysaloma there's no need to paste your config.  The explanation has already been given.  There's nothing esphome can do about this other than moving to a newer platformio and Arduino framework.  That is in progress, but is currently blocked by issues in the newer Arduino framework.

UniKyrn, 2022-09-28:
Then the solution to this issue seems obvious, ESPHome should not claim it
supports the board until such a time as it actually does.

chunkysteveo, 2022-09-28:
It does support the board though... I am on a 32-bit platform and have several ESP32C3 boards installed and working fine. 

I'm sure it used to state that C3, S2, S3 boards were experimental? Can't see that anymore 😕

UniKyrn, 2022-09-28:
Yeah, I remember that warning also.  Then it went away and a series of boards was added to the support list, so I pushed the button on a hardware order from AliExpress.

Which of course didn't end well.

ssieb, 2022-09-29:
> Then the solution to this issue seems obvious, ESPHome should not claim it supports the board until such a time as it actually does.

Where do you see any claim that the board is supported?
However, other people are using it.  You just have to specify the right platform options.  And maybe not use an ARM system for the compiling.

UniKyrn, 2022-09-29:
When I adopted the board, I was given a default configuration and asked to select the correct board type from a list.

https://registry.platformio.org/platforms/platformio/espressif32/boards?p=4

I was forwarded to that list by ESPHome, so I believed anything on that list was supported.  In deed, when I try to compile the sample, I don't get a board error, I get a toolchain error.  I have tried specifying the framework based on suggestions here, no joy.

[image]

This is, as I've said, on an RPi4 64bit.  It might work someplace, but it doesn't work everyplace and an RPi4 is a reasonably popular platform.

DanteBC, 2022-09-29:
```
esphome:
  name: esp01c3
  platformio_options:
    board_build.flash_mode: dio
    platform_packages:
     - toolchain-riscv32-esp @ 8.4.0+2021r1
     - toolchain-xtensa-esp32s2 @ 8.4.0+2021r1

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```

Try editing your config like this. 

> When I adopted the board, I was given a default configuration and asked to select the correct board type from a list.
>
> https://registry.platformio.org/platforms/platformio/espressif32/boards?p=4
>
> I [...]

UniKyrn, 2022-09-29:
INFO Reading configuration /config/esphome/esp32c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32c3 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
INFO Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1' requirements for your system 'linux_aarch64'

Nope, can't find that toolchain either.

Given that some boards/architectures are not working, my suggestion is that the popup asking the user to select a board type from the webpage linked to, should link to a webpage your team maintains, that lists what combinations you know do and don't work.  It shouldn't link straight to the platformio.org page, since you have no control over what boards might get listed there.

Give users a way to update your list with changes and at least they will have a way to know, before they buy hardware, whether it is expected to work.

chunkysteveo, 2022-09-29:
Compile the ESPhome firmware on a non-64 bit pi platform? My 32 bit on a pi 3 works perfectly fine with these C3 boards.

UniKyrn, 2022-09-30:
That's not a fix, it's a band-aid.  What happens when the next board only compiles on an x86 system?

ssieb, 2022-09-30:
> That's not a fix, it's a band-aid. What happens when the next board only compiles on an x86 system?

I guess you'll have to use an x86 system to compile it.  What do you expect us to do?  We can only use the toolchains that are provided.  We have no control over that.
And really, I'm done with this discussion now.

UniKyrn, 2022-09-30:
What triggers me about all of this is that when I reported an issue, I got a lot of excuses and reasons why it wouldn't/couldn't be fixed, but I heard no solutions as to how to avoid the same problem in the future,  That is why I suggested you link to your own site and curated list of boards/archs that you know work, or don't.  Your user base gets no warning, which bothers me.

As for using other boards, I already have lots of them, the C3 was a new addition.

closed 2022-09-30 by UniKyrn

evlo, 2022-12-15:
How to compile with arduino framework and not idf? I was able to do it before, but not anymore. As I was trying to put ble_proxy into the firmware I ended up updating and now i'm unable to make the board boot.

Thing is in the past I was somehow possible to do it before :(