esphome/issues #3415: Connectivity loss for just some ESPHome devices on WiFi network every GTK rekeying interval

climategadgets, 2022-07-01:
### The problem

### Expected Behavior
All ESPHome devices on the network stay connected.

### Actual Behavior
* Some (but not all) ESPHome devices on WiFi network lose connectivity every "GTK rekeying" interval set for that network;
* Different devices exhibit connectivity drops at different times, the usual trigger for the change being an arrival of a device previously not present on this network;
* Once the set of devices losing connectivity and staying connected gets determined, it can stay the same for extended periods of time (weeks to months).

### Charts
Two different time periods on the same set of devices with GTK rekeying intervals set to 3600 (default for the networking gear), and to 28800.
[image]
[image]

### Which version of ESPHome has the issue?

Last known affected version is 2021.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

Wemos D1

### Component causing the issue

wifi

### Example YAML snippet

```yaml
# Standard WiFi configuration with no gimmicks
---
esphome:
  name: <name
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: <ssid>
  password: <password>
```

climategadgets, 2022-07-01:
Possibly related: #1793

nagyrobi, 2022-07-01:
What AP do you use?

climategadgets, 2022-07-01:
UniFi nanoHD, with Linux controller.

climategadgets, 2022-11-02:
Bump. The issue is still present with even recent ESPHome versions.

climategadgets, 2023-03-28:
Another bump.

Both ESPHome and network hardware are at the last release, the behavior is still the same. Still no correlation to absolutely anything other than the GTK rekey interval, and still some devices exhibit this behavior, and some don't (obvious on the screenshots), with devices periodically moving from one group to another upon reboot.

The length of the blackout period is quite painful for applications where a 30 second overshoot can result in an x2 control signal provided.

[image]
[image]

climategadgets, 2023-04-02:
Actually, after a long and careful look, there *is* a very indirect correlation.

Blackouts on GTK rekey interval happen only on chips with 1-Wire devices present. However, not all chips with 1-Wire devices present suffer from blackouts, there are few of them that have spotless connectivity.

Setting the log level on those chips to VERY_VERBOSE, let's see if MQTT captures anything.