esphome/issues #3416: modbus function code 44 not working in 2022.6

heinekmadsen, 2022-07-01:
### The problem

I have a custom component for a Wavin ahc 9000 floor heating system.
In ESPhome 2022.5 the component works just fine, but after update to 2022.6 it's no longer working.

When using a number on the modbus_controller platform it fails to write the new value.

It uses function code 44 for writing the value. But in with this commit (i think) it makes it default to function code 6.
https://github.com/esphome/esphome/commit/4f52d43347c6b87529238d737473bed974b5705e

I do see that per https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf user defined function codes should be within some specific ranges.

But how will i get my number working again with 2022.6?

### Which version of ESPHome has the issue?

2022.6.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
number:
  - platform:               modbus_controller
    name:                   ${name} Target Temperature ${channel_01_friendly_name}_test
    id:                     ${device}_target_temp_${channel_01_id}_test
    modbus_controller_id:   ${device}_modbus_controller
    custom_command: 
      - 0x01
      - 0x43
      - 0x02
      - 0x00
      - 0x00 # channel 01 (00)
      - 0x01
    value_type: U_WORD
    unit_of_measurement: "°C"
    min_value: 6
    max_value: 40
    step: .5
    write_lambda: |-
      ESP_LOGD("main", "Trying to write new target temp: %f",x);
      uint16_t targettemp = x * 10;
      payload.push_back(0x0144);
      payload.push_back(0x0200);
      payload.push_back(0x0001); //0x0001 = channel 01 (00)
      payload.push_back(targettemp);
      return true;
    lambda: "return x*0.1;"
```

### Anything in the logs that might be useful for us?

```txt
[12:28:05][D][modbus.number:055]: Updating register: connected Sensor=Wavin Target Temperature Mille start address=0xBB0D register count=1 new value=23.00 (val=1.00)
[12:28:05][V][modbus_controller:036]: Sending next modbus command to device 1 register 0xBB0D count 1
[12:28:05][VV][uart.arduino_esp32:150]:     Flushing...
[12:28:05][V][modbus:184]: Modbus write: 01.06.BB.0D.01.44.3D.4E (8)
[12:28:05][V][modbus_controller:465]: Command sent 6 0xBB0D 1
[12:28:05][V][modbus:058]: Modbus received Byte  1 (0X1)
[12:28:05][V][modbus:058]: Modbus received Byte  134 (0X86)
[12:28:05][V][modbus:058]: Modbus received Byte  1 (0X1)
[12:28:05][V][modbus:058]: Modbus received Byte  131 (0X83)
[12:28:05][V][modbus:058]: Modbus received Byte  160 (0Xa0)
[12:28:05][D][modbus:105]: Modbus error function code: 0x86 exception: 1
[12:28:05][E][modbus_controller:068]: Modbus error function code: 0x6 exception: 1 
[12:28:05][E][modbus_controller:077]: Modbus error - last command: function code=0x6  register adddress = 0xBB0D  registers count=1 payload size=2
[12:28:07][VV][scheduler:186]: Running interval 'update' with interval=60000 last_execution=1438678 (now=1498678)
```

ssieb, 2022-10-31:
Do you have protocol documentation for your device?

heinekmadsen, 2022-10-31:
> Do you have protocol documentation for your device?

added the docs here
https://github.com/heinekmadsen/esphome_components/blob/dev/components/wavinahc9000v2/Wavin%20Modbus%20specification%20AHC9000%20direkte.pdf

heinekmadsen, 2022-11-30:
@ssieb did you have time to look into the documentation?