esphome/issues #3417: GPIO20 is now usable on some ESP32 chips and should be removed from validation list

critchards, 2022-07-01:
### The problem

ESP32 PICO Mini can use GPIO20. The Adafruit Feather 32 V2 uses this chip and GPIO20 is connected to the STEMMAQT port. Any peripherals that make use of STEMMAQT will not be accessible in ESPHOME.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.7

### What platform are you using?

ESP32

### Board

adafruit_feather_esp32_v2

### Component causing the issue

gpio

### Example YAML snippet

```yaml
i2c:
  sda: 22
  scl: 20
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/weather-feather.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

i2c: [source /config/esphome/weather-feather.yaml:37]
  - sda: 22
    
    The pin GPIO20 is not usable on ESP32s.
    scl: 20
```

### Additional information

https://github.com/esphome/esphome/tree/dev/esphome/components/esp32)/gpio_esp32.py
in esp_validate_gpio_pin...

```    
if value in (20, 24, 28, 29, 30, 31):
        # These pins are not exposed in GPIO mux (reason unknown)
        # but they're missing from IO_MUX list in datasheet
        raise cv.Invalid(f"The pin GPIO{value} is not usable on ESP32s.")
```

I dont know the best way to approach this, add pin 20 as a warning now that some chips use it maybe?

ashald, 2022-08-03:
Turns out, there is a workaround to use an external component to change validation rules, which you can do with:
```
external_components:
  - source: github://ashald/esphome@gpio20
    components:
      - esp32
```

That being said, I submitted a PR and 🤞 it will get into the next release.

kj4dxk, 2022-09-01:
Did you also have an issue with the ESP32 QT PY PICO which has a neopixel power on GPIO8?  Can we allow GPIO8 as well?

#esp32:
#  board: adafruit_qtpy_esp32
#  variant: esp32
#  framework:
#    type: arduino

esp32:
  board: esp32dev
  framework:
    type: arduino

output:
  - platform: gpio
    id: "neopixel_power"
    pin: GPIO8 # NEOPIXEL_POWER
    inverted: true