esphome/issues #3418: ili9341 with RGB color doesn't support

Aculeasis, 2022-07-01:
### The problem

My ili9341 2.4" TFT has an issue - red and blue colors are mixed up. I think it is because all ILI9341_MADCTL use BGR scheme, so I changed it to RGB (`0x48` to `0x40` in https://github.com/esphome/esphome/blob/2022.6.2/esphome/components/ili9341/ili9341_init.h#L50) and it worked. Maybe, somebody to add such a screen option (TFT 2.4R RGB) or bring the color scheme to the settings?

ALSO, I tested M5STACK and TFT 2.4 too, the first one has distorted colors and the second one has blinked. I couldn't fix they.

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.7

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

2.4" TFT 14P IC:ST7789V SPI 240(RGB) x 320 :10108

### Example YAML snippet

```yaml
font:
  - file: 'gfonts://Roboto'
    id: my_font
    size: 30
color:
  - id: red
    red: 100%
  - id: green
    green: 100%
  - id: blue
    blue: 100%
spi: 
  clk_pin: 25
  mosi_pin: 26
  miso_pin: 13
output:
  - platform: ledc
    pin: 32
    id: gpio_32_backlight_pwm
light:
  - platform: monochromatic
    output: gpio_32_backlight_pwm
    name: "ILI9341 Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON
display:
  - platform: ili9341
    model: TFT 2.4R
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33  
    lambda: |-
      it.fill(Color::BLACK);
      it.print(0, 0, id(my_font), id(red), TextAlign::TOP_LEFT, "Hello RED World!");
      it.print(0, 40, id(my_font), id(green), TextAlign::TOP_LEFT, "Hello GREEN World!");
      it.print(0, 80, id(my_font), id(blue), TextAlign::TOP_LEFT, "Hello BLUE World!");
```

### Additional information

If you have the same issue, you can temporarily fix it:
```
docker exec -it addon_<your_hash>_esphome /bin/bash
apt update && apt install nano
nano /esphome/esphome/components/ili9341/ili9341_init.h
```
Change `ILI9341_MADCTL  , 1, 0x48` to `ILI9341_MADCTL  , 1, 0x40` in `PROGMEM INITCMD_TFT`, save, clean build files in ESPHome console and rebuild firmware.

nielsnl68, 2022-10-31:
ping to unstall

nielsnl68, 2022-10-31:
Sorry for the late reaction could you tell where you ordered this display?

Aculeasis, 2022-11-01:
Here https://www.aliexpress.com/item/4000923807937.html
I bought it a year ago.

lukaszklapiszewski, 2022-12-13:
Hello,

any news about solution for this problem? I have the same situation.

ElektronikCZ, 2023-06-22:
I have the same issue

trppgg, 2023-07-21:
i'm having the same issue, red and blue are swapped

nielsnl68, 2023-07-21:
Can you all add an image with the following script:

```yaml
display:
  - platform: ili9xxx
    id: my_display
    . . .
    lambda: |-
      for(uint8_t i = 0; i<255; i++) {
          it.horizontal_line(5,i,100, my_red.fade_to_white(i));
          it.horizontal_line(110,i,100, my_green.fade_to_black(i));
          it.horizontal_line(215,i,100, my_blue.gradient(my_blue, i));
      }
        
color:
  - id: my_green
    green: 100%  
  - id: my_red
    red: 100%
  - id: color_blue 
    blue: 100%
```

This will give me an idea what is going on.

Aculeasis, 2023-07-21:
@nielsnl68
<details>
<summary>This is it</summary>

[image]
</details>

<details>
<summary>If I change `ILI9XXX_MADCTL  , 1, 0x48` to `0x40` in `INITCMD_ILI9341` in `/esphome/esphome/components/ili9xxx/ili9xxx_init.h` This starts to look right.</summary>

[image]
</details>

frustreermeneer, 2023-10-17:
This ILI9341 screen also has red and blue swapped:
https://www.aliexpress.com/item/1005005616073472.html

dumpfheimer, 2023-10-19:
This works as a workaround for me:
```
esphome:
  on_boot:
    - lambda: |-
        uint8_t val = 0x40;
        uint8_t len = 1;
        ili9xxx_ili9xxxdisplay->send_command(esphome::ili9xxx::ILI9XXX_MADCTL, &val, len);
```

Note, that if you use an ID for your display you might need to replace `ili9xxx_ili9xxxdisplay` with the ID of your display

vodidan, 2023-11-02:
> ```
> esphome:
>   on_boot:
>     - lambda: |-
>         uint8_t val = 0x40;
>         uint8_t len = 1;
>         ili9xxx_ili9xxxdisplay->send_command(esphome::ili9xxx::ILI9XXX_MADCTL, &val, len);
> ```

When using this workaround, display update takes a lot longer(half a second) and the touchscreen is not as responsive as normal(pretty much unusable).

with workaround:
```
[16:37:14][W][component:214]: Component ili9xxx.display took a long time for an operation (0.65 s).
[16:37:14][W][component:215]: Components should block for at most 20-30ms.
```
without workaround:
```
[16:39:28][W][component:214]: Component ili9xxx.display took a long time for an operation (0.17 s).
[16:39:28][W][component:215]: Components should block for at most 20-30ms.
```

nielsnl68, 2023-11-02:
You can ignore the warnings for now, sending the data to the display is just taking longer then esphome would love to see. And complains about it. There is noting wrong. 
At some time we hope to find a solution to speed up the transfer the display content tthe display.

clydebarrow, 2023-12-29:
> > ```
> > esphome:
> >   on_boot:
> >     - lambda: |-
> >         uint8_t val = 0x40;
> >         uint8_t len = 1;
> >         ili9xxx_ili9xxxdisplay->send_command(esphome::ili9xxx::ILI9XXX_MADCTL, &val, len);
> > ```

You can replace that with the `color_order:` config option now.

Aculeasis, 2023-12-30:
Yes, it's work now. I think this issue can be closed.
```
display:
  - platform: ili9xxx
    id: tft_display  
    model: TFT 2.4R
    color_order: RGB
```

closed 2023-12-30 by Aculeasis