esphome/issues #3419: ESP32-C3 & UART component error when compiling

bakermat, 2022-07-02:
### The problem

When adding the UART component for a Lolin C3 Mini board the firmware doesn't compile and throws an error. When removing the UART component from the YAML file it compiles correctly.

I believe any board using these build flags runs into the issue, in case this helps:
```
-DARDUINO_USB_CDC_ON_BOOT=1
-DARDUINO_USB_MODE=1
```

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2022.6.7

### What platform are you using?

ESP32

### Board

lolin_c3_mini

### Component causing the issue

uart

### Example YAML snippet

```yaml
esphome:
  name: testy-mctestface
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: lolin_c3_mini
  variant: esp32c3
  framework: 
    type: arduino
    version: 2.0.3
    platform_version: 4.3.0

uart:
  rx_pin:
    number: 10
  baud_rate: 115200
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/bla.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing testy-mctestface (board: lolin_c3_mini; framework: arduino; platform: platformio/espressif32 @ 4.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 2.0.0
|-- <FS> 2.0.0
|-- <Update> 2.0.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 2.0.0
|-- <ESPmDNS> 2.0.0
|-- <ArduinoJson> 6.18.5
|-- <Dsmr> 0.5.0
|-- <Crypto> 0.4.0
Compiling /data/testy-mctestface/.pioenvs/testy-mctestface/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o
src/esphome/components/uart/uart_component_esp32_arduino.cpp: In member function 'virtual void esphome::uart::ESP32ArduinoUARTComponent::setup()':
src/esphome/components/uart/uart_component_esp32_arduino.cpp:90:25: error: cannot convert 'HWCDC*' to 'HardwareSerial*' in assignment
     this->hw_serial_ = &Serial;
                         ^~~~~~
*** [/data/testy-mctestface/.pioenvs/testy-mctestface/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o] Error 1
========================== [FAILED] Took 4.59 seconds ==========================
```
```

Mosher23, 2022-07-20:
I have exactly same error here with adafruit_qt_py_c3 board

Prologer, 2022-08-31:
Facing the same issues with 2022.8.0 and 
`esp32:
  board: lolin_c3_mini
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1`

evlo, 2022-09-18:
ttgo t-01c3 also same 
```
    platform_version: 4.4.0
    version: 2.0.3
```

evlo, 2022-09-18:
```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```
did build for me

but again my board is actually `ttGO-T-01C3` and i did not look hot to make custom boards yet

PaulAntonDeen, 2022-10-13:
I get the same issue with this configuration:
`esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0`
I tried multiple version and platform_versions. Have not been able to find one that compiles.

TiddlyWiddly, 2022-11-09:
Same problem with this configuration

`esp32:
  board: lolin_c3_mini
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0`

joostbijl, 2022-11-27:
I'm having this issue too. It looks like something is off with the pin assignment on the C3 compared to the D1?

```
esphome:
  name: ts27meter
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: lolin_c3_mini
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0

uart:
  rx_pin: 20
  baud_rate: 115200
  rx_buffer_size: 1500
```
the error is:
```
Compiling .pioenvs/ts27meter/src/esphome/core/application.cpp.o
src/esphome/components/uart/uart_component_esp32_arduino.cpp: In member function 'virtual void esphome::uart::ESP32ArduinoUARTComponent::setup()':
src/esphome/components/uart/uart_component_esp32_arduino.cpp:90:25: error: cannot convert 'HWCDC*' to 'HardwareSerial*' in assignment
     this->hw_serial_ = &Serial;
                         ^~~~~~
*** [.pioenvs/ts27meter/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o] Error 1
```
When looking ath the uart_component_esp32_arduino.cpp file, it might be an issue with not correctly defining CONFIG_IDF_TARGET_ESP32C3 or incorrect assignment of pin numbers in the esphome yaml config?

evlo, 2022-11-27:
```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

this did not work for you?

evlo, 2022-11-27:
> I get the same issue with this configuration:
> 
> ```
> esp32:
>   board: lolin_s2_mini
>   variant: ESP32S2
>   framework:
>     type: arduino
>     version: 2.0.3
>     platform_version: 4.4.0
> ```
> 
> I tried multiple version and platform_versions. Have not been able to find one that compiles.

Why don't you use esp32-c3-devkitm-1 with the 2.03 and 4.4.0 ?

evlo, 2022-11-27:
```
uart:
  id: uart_bus
  tx_pin: GPIO21
  rx_pin: GPIO20
  rx_buffer_size: 2048  
  baud_rate: 300
  data_bits: 7
  stop_bits: 1
  parity: EVEN
  debug: 
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

joostbijl, 2022-11-27:
> would you mind trying building with
> 
> ```
> esp32:
>   board: esp32-c3-devkitm-1
>   framework:
>     type: arduino
>     version: 2.0.3
>     platform_version: 4.4.0
> uart:
>   tx_pin: GPIO21
>   rx_pin: GPIO20
> ```
> 
> I'm curious if it might be the port naming thing, altrought it should not be, it is now first and easiest thing I see.

Yes, this works. Thanks for the hint.

I did not think a second of trying another board than the one i actually have.

TiddlyWiddly, 2022-11-27:
Sorry, I got this working and never came back to post my fix. 

This worked for me: 

`esphome:
  name: test-c3-pico
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB
    board_build.variant: esp32c3

esp32:
  #board: lolin_c3_mini
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0

uart:
  rx_pin: GPIO20
  tx_pin: GPIO21
  baud_rate: 115200`

manuelsteiner, 2022-12-14:
Hi,

I have the adafruit_qt_py_s2 board and I am running in the same issue. When changing the underlying platformio framework to `esp-idf`, compilation is successful but either the ESP32 hangs or does not output any logs via UART on USB. I need to investigate further but I think it actually does not boot with a minimal config since no WIFI is joined or created.

When changing the board, logs are also not sent to the USB UART. Which is presumabily the case because the platformio config is wrong.

Has anyone had any progress enabling the UART and still getting the logger to output to USB with the arduino platform?

Any help would be appreciated.

evlo, 2022-12-15:
> Yes, this works. Thanks for the hint.
>
> I did not think a second of trying another board than the one i actually have.

Now with esphome 2022.12.0 i have no success building working firmware

LayerCakeMakes, 2022-12-19:
> Hi,
>
> I have the `adafruit_qt_py_s2` board and I am running in the same issue. When changing the underlying platformio framework to `esp-idf`, compilation is successful but either the ESP32 hangs or [...]

I'm having the same issues with UART but I also have boot issues with the arduino framework selected, it only joins every few boots ,so that might be a separate issue. If no network is joined the chip also gets quite hot.

ssieb, 2022-12-19:
> When changing the board, logs are also not sent to the USB UART. Which is presumabily the case because the platformio config is wrong.

See https://esphome.io/components/logger.html#hardware-uarts

manuelsteiner, 2022-12-19:
Thanks, 

I've been through that. I think it doesn't even boot with the esp-idf platform. But I'll investigate further to confirm that is actually the case.

When choosing a different board, I'd presumably have to re-configure a bunch of things, such as pins and so on to make things work?

LayerCakeMakes, 2022-12-19:
@manuelsteiner I've just tried with ESP-IDF and it works fine for me. Only problem is the neopixel LED is not supported in the ESP-IDF framework so if you need it you're still out of luck.

```
...
esp32:
  board: adafruit_qtpy_esp32s2
  framework:
    type: esp-idf
...
```

LayerCakeMakes, 2022-12-21:
> 

With a minimal config it boots and joins a network. I had to test the USB logging as I never use that feature, I just look wirelessly at the logs, I don't think it logs via USB by default, at least I did not get any messages.

manuelsteiner, 2022-12-21:
@LayerCakeMakes Thank you for confirming.

LayerCakeMakes, 2022-12-21:
@manuelsteiner USB logging works if you set `hardware_uart: USB_CDC`

evlo, 2022-12-21:
@LayerCakeMakes can you post the minimal config with uart that works for you?

LayerCakeMakes, 2022-12-21:
@evlo manuelsteiner and I are having the same issue but with the Adafruit QT Py S2 so this is probably not as helpful for you but here it is:
```
esphome:
  name: uartdemo

esp32:
  board: adafruit_qtpy_esp32s2
  framework:
    type: esp-idf

logger:
api:
ota:
  password: !secret esphome_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  rx_pin: RX
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
```

manuelsteiner, 2022-12-22:
That's weird. I tried again, even with a fresh board. Flashed fine but wouldn't come up and join a network or print logs on the serial port. Super strange that this works fine for you but I've tried with two boards now an neither feel like doing much.

LayerCakeMakes, 2022-12-22:
Hmm it seems to me that booting is hit or miss. Sometimes I need to
disconnect the power supply completely, the reset button isn't enough...
Strange for sure

manuelsteiner, 2022-12-22:
Ah interesting, thanks for that. Seems like there is something strange going on still and it's not just me.

tomav, 2023-01-17:
Same issue here when adding uart block in my config for a Lolin S2 Mini board.
Using v2022.12.3 Docker version. Tested on 2 machines.

```
uart:
  id: uart_bus
  rx_pin: GPIO11
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
```

```
Compiling .pioenvs/test/src/main.cpp.o
src/esphome/components/uart/uart_component_esp32_arduino.cpp: In member function 'virtual void esphome::uart::ESP32ArduinoUARTComponent::setup()':
src/esphome/components/uart/uart_component_esp32_arduino.cpp:90:25: error: cannot convert 'USBCDC*' to 'HardwareSerial*' in assignment
     this->hw_serial_ = &Serial;
                         ^~~~~~
*** [.pioenvs/test/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o] Error 1
========================== [FAILED] Took 6.79 seconds ==========================
```

As suggested, I was able to compile using `esp-idf`. Build seems to work (teleinfo module) but have a lot of errors like :

```
[23:12:12][E][teleinfo:133]: No group found
[23:12:16][D][sensor:127]: 'WiFi Signal dB': Sending state -61.00000 dBm with 0 decimals of accuracy
[23:12:43][E][teleinfo:038]: bad crc: got 60 except 56
[23:12:43][E][teleinfo:038]: bad crc: got 60 except 50
[23:12:43][D][sensor:127]: 'Linky Index': Sending state 35917.92578 kWh with 0 decimals of accuracy
[23:12:43][E][teleinfo:038]: bad crc: got 34 except 41
[23:12:43][E][teleinfo:168]: Invalid value for tag EASF02
[23:12:43][E][teleinfo:038]: bad crc: got 41 except 72
[23:12:43][E][teleinfo:038]: bad crc: got 41 except 35
[23:12:43][E][teleinfo:038]: bad crc: got 32 except 88
[23:12:43][E][teleinfo:038]: bad crc: got 32 except 95
```

I don't know if these errors are related to `teleinfo` or the build itself yet.

LayerCakeMakes, 2023-01-18:
As mentioned in #3988 I could get the UART to work with this minimal configuration (no log output via USB but I don't need that)

```
esphome:
  name: minimal-uart
  platformio_options:
    board_build.extra_flags:
      - "-DARDUINO_USB_CDC_ON_BOOT=0"

esp32:
  board: adafruit_qtpy_esp32s2
  framework:
    type: arduino

logger:
api:
ota:
  password: !secret esphome_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  rx_pin: RX
  tx_pin: TX
  baud_rate: 9600
```

The important part is the extra flags.

tomav, 2023-01-24:
Hi @LayerCakeMakes, I managed to compile a basic project with your suggestion and it works.
I'll try a full project as soon as I can to see if it behaves as expected. 
Thanks!

tomav, 2023-01-24:
Confirmed: compiles and works as expected. Thanks @LayerCakeMakes

piotrva, 2023-02-12:
I faced the same issue for dfrobot_beetle_esp32c3, but no issues for esp32-c3-devkitm-1.
It would however be nice to have an ability to use USB-CDC for a logger output.

nebulous, 2023-03-04:
For those experiencing intermittent boots - I've noticed that on the esp32-s3-box-lite at least, that booting will fail and reboot before the wifi connect phase _unless_ I'm monitoring the logs via usb. 🤔

screek-workshop, 2023-03-06:
> 

it's same from here.
which is very wired, only open the logs it can survive. 

i have a  lot c3 without ttl chip but only with usb-cdc.  
maybe is the logger wait something to open the port?

evlo, 2023-03-06:
For (even my) refference - this is working settings I do use that boot with no logger attached, `- "-DARDUINO_USB_CDC_ON_BOOT=0"` is the key i think, however the board DOES have usb com port emulator

```
esphome:
  name: ${hostname}
  comment: ${comment}
  platformio_options: 
    board_build.flash_mode: dio
    board_build.extra_flags:
      #- "-DARDUINO_USB_MODE=0"
      - "-DARDUINO_USB_CDC_ON_BOOT=0"
  on_boot:
    priority: 300
    then:
      - light.turn_on:
          id: onboard_rgb
          brightness: 20%
          red: 0%
          green: 100%
          blue: 0%        

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
```

I was never able to get it working with esp idf

screek-workshop, 2023-03-07:
> For those experiencing intermittent boots - I've noticed that on the esp32-s3-box-lite at least, that booting will fail and reboot before the wifi connect phase _unless_ I'm monitoring the logs via usb. 🤔

use two port typec esp32-c3 board, can track the reboot is cause by:

```bash
[08:23:26]E (15952) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[08:23:26]E (15952) task_wdt:  - loopTask (CPU 0)
[08:23:26]E (15952) task_wdt: Tasks currently running:
[08:23:26]E (15952) task_wdt: CPU 0: IDLE
[08:23:26]E (15952) task_wdt: Aborting.
[08:23:26]
[08:23:26]abort() was called at PC 0x42037ac9 on core 0
[08:23:26]Core  0 register dump:
[08:23:26]MEPC    : 0x40382588  RA      : 0x40388cda  SP      : 0x3fc94bd0  GP      : 0x3fc90400  
WARNING Decoded 0x40382588: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
WARNING Decoded 0x40382588: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
[08:23:26]TP      : 0x3fc6e124  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
[08:23:26]S0/FP   : 0x3fc94bfc  S1      : 0x3fc94bfc  A0      : 0x3fc94c08  A1      : 0x3fc94bea  
[08:23:26]A2      : 0x00000000  A3      : 0x3fc94c35  A4      : 0x00000001  A5      : 0x3fc9a000  
[08:23:26]A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc9a22c  S3      : 0x7fffffff  
[08:23:26]S4      : 0x3fc917dc  S5      : 0x3fc917cc  S6      : 0x3fc917d4  S7      : 0x3fc917cc  
[08:23:26]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
[08:23:26]T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
[08:23:26]MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
[08:23:26]MHARTID : 0x00000000  
[08:23:26]
[08:23:26]Stack memory:
[08:23:26]3fc94bd0: 0x00000000 0x00000000 0x3fc94be8 0x4038e566 0x3fc917d4 0x3fc917cc 0x3fc90030 0x3fc928d0
[08:23:26]3fc94bf0: 0x3fc94bfc 0x3fc928ec 0x3fc94be8 0x33303234 0x39636137 0x00000000 0x726f6261 0x20292874
[08:23:26]3fc94c10: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x37333032 0x20396361 0x63206e6f
[08:23:26]3fc94c30: 0x2065726f 0x00000030 0x3c130000 0x31832b3f 0x600c2000 0x3c13c000 0x3fc9a000 0x42037acc
[08:23:26]3fc94c50: 0x00000000 0x00000000 0x00000000 0x00003e50 0x00000004 0x00000003 0x3fcb4b94 0x00000000
[08:23:26]3fc94c70: 0x00000000 0x3fc9a000 0x3fc94edc 0x00000001 0x00001881 0x80000007 0x3fca1078 0x403801ee
[08:23:26]3fc94c90: 0x00000001 0x3fc94c98 0xffffffff 0x3fca1af0 0x3fca1af0 0x00000000 0x3fc94cac 0xffffffff
[08:23:26]3fc94cb0: 0x3fc94cac 0x3fc94cac 0x00000000 0x3fc94cc0 0xffffffff 0x3fc94cc0 0x3fc94cc0 0x00000000
[08:23:26]3fc94cd0: 0x3fc94cd4 0xffffffff 0x3fc94cd4 0x3fc94cd4 0x00000000 0x3fc94ce8 0xffffffff 0x3fc94ce8
[08:23:26]3fc94cf0: 0x3fc94ce8 0x00000000 0x3fc94cfc 0xffffffff 0x3fc94cfc 0x3fc94cfc 0x00000000 0x3fc94d10
[08:23:26]3fc94d10: 0xffffffff 0x3fc94d10 0x3fc94d10 0x00000000 0x3fc94d24 0xffffffff 0x3fc94d24 0x3fc94d24
[08:23:26]3fc94d30: 0x00000000 0x3fc94d38 0xffffffff 0x3fc94d38 0x3fc94d38 0x00000000 0x3fc94d4c 0xffffffff
[08:23:26]3fc94d50: 0x3fc94d4c 0x3fc94d4c 0x00000000 0x3fc94d60 0xffffffff 0x3fc94d60 0x3fc94d60 0x00000000
[08:23:26]3fc94d70: 0x3fc94d74 0xffffffff 0x3fc94d74 0x3fc94d74 0x00000000 0x3fc94d88 0xffffffff 0x3fc94d88
[08:23:26]3fc94d90: 0x3fc94d88 0x00000000 0x3fc94d9c 0xffffffff 0x3fc94d9c 0x3fc94d9c 0x00000000 0x3fc94db0
[08:23:26]3fc94db0: 0xffffffff 0x3fc94db0 0x3fc94db0 0x00000000 0x3fc94dc4 0xffffffff 0x3fc94dc4 0x3fc94dc4
[08:23:26]3fc94dd0: 0x00000000 0x3fc94dd8 0xffffffff 0x3fc94dd8 0x3fc94dd8 0x00000000 0x3fc94dec 0xffffffff
[08:23:26]3fc94df0: 0x3fc94dec 0x3fc94dec 0x00000000 0x3fc94e00 0xffffffff 0x3fc94e00 0x3fc94e00 0x00000000
[08:23:26]3fc94e10: 0x3fc94e14 0xffffffff 0x3fc94e14 0x3fc94e14 0x00000000 0x3fc94e28 0xffffffff 0x3fc94e28
[08:23:26]3fc94e30: 0x3fc94e28 0x00000000 0x3fc94e3c 0xffffffff 0x3fc94e3c 0x3fc94e3c 0x00000000 0x3fc94e50
[08:23:26]3fc94e50: 0xffffffff 0x3fc94e50 0x3fc94e50 0x00000000 0x3fc94e64 0xffffffff 0x3fc94e64 0x3fc94e64
[08:23:26]3fc94e70: 0x00000000 0x3fc94e78 0xffffffff 0x3fc94e78 0x3fc94e78 0x00000002 0x3fc94e8c 0xffffffff
[08:23:26]3fc94e90: 0x3fca43b0 0x3fcb60d4 0x00000000 0x3fc94ea0 0xffffffff 0x3fc94ea0 0x3fc94ea0 0x00000000
[08:23:26]3fc94eb0: 0x3fc94eb4 0xffffffff 0x3fc94eb4 0x3fc94eb4 0x00000008 0x3fc94ec8 0xffffffff 0x3fcad074
[08:23:26]3fc94ed0: 0x3fcbb20c 0x00000000 0x3fc94edc 0xffffffff 0x3fc94edc 0x3fc94edc 0x00000000 0x3fc94ee8
[08:23:26]3fc94ef0: 0x00000000 0x00000000 0x00000000 0x3fc94f00 0xffffffff 0x3fc94f00 0x3fc94f00 0x00000000
[08:23:26]3fc94f10: 0x3fc94f14 0xffffffff 0x3fc94f14 0x3fc94f14 0x00000001 0x00000001 0x00000000 0x0001ffff
[08:23:26]3fc94f30: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc94f3c 0x00000000 0x00000000 0x00000000
[08:23:26]3fc94f50: 0x3fc94f54 0xffffffff 0x3fc94f54 0x3fc94f54 0x00000000 0x3fc94f68 0xffffffff 0x3fc94f68
[08:23:26]3fc94f70: 0x3fc94f68 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
[08:23:26]3fc94f90: 0x00000000 0x3fc9e214 0x3fc9e27c 0x3fc9e2e4 0x00000000 0x00000000 0x00000001 0x00000000
[08:23:26]3fc94fb0: 0x00000000 0x00000000 0x421176f8 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[08:23:26]
[08:23:26]
[08:23:26]
[08:23:26]ELF file SHA256: 0000000000000000
[08:23:26]
[08:23:26]Rebooting...
[08:23:26]ESP-ROM:esp32c3-api1-20210207
[08:23:26]Build:Feb  7 2021
[08:23:26]rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[08:23:26]Saved PC:0x40382148
```

screek-workshop, 2023-03-07:
> For those experiencing intermittent boots - I've noticed that on the esp32-s3-box-lite at least, that booting will fail and reboot before the wifi connect phase _unless_ I'm monitoring the logs via usb. 🤔

i temporarily  fix this problem, by delay the ttl output.  
which need change the source of logger.cpp at logger component folder:

```c
void HOT Logger::log_message_(int level, const char *tag, int offset) {
  // remove trailing newline
  if (this->tx_buffer_[this->tx_buffer_at_ - 1] == '\n') {
    this->tx_buffer_at_--;
  }
  // make sure null terminator is present
  this->set_null_terminator_();

  const char *msg = this->tx_buffer_ + offset;
  if (this->baud_rate_ > 0) {
#ifdef USE_ARDUINO

    // 对于C3开启使用USB串口的模式下，如果要写出日志的情况，进行一个特殊补丁。(*23年3月7日*15时41分)
    #if defined(USE_ESP32_VARIANT_ESP32C3) && defined(ARDUINO_USB_CDC_ON_BOOT) && defined(ARDUINO_USB_MODE)

      if (millis() > 2000){ // 尝试延后启动？(*23年3月7日*15时26分)
        // LOG会有一些奇怪的问题，似乎在等待端口？(*23年3月7日*13时37分)
        // 适当延长这个启动间隔，能够保证串口输出，也能让wdt不触发。(*23年3月7日*15时41分)
        // 但是进一步和wifi的影响需要确定，比如如果wifi重连会怎么样？(*23年3月7日*15时41分)
        this->hw_serial_->println(msg);
      }

    #else

      this->hw_serial_->println(msg);

    #endif
#endif  // USE_ARDUINO
```

screek-workshop, 2023-03-07:
> For (my) refference - this is working settings I do use that boot with no logger attached, `- "-DARDUINO_USB_CDC_ON_BOOT=0"` is the key i think, however the board DOES have usb com port emulator.
> 
> Also in my case I do use D- and D+ of usb port for UART communication.
> 
> ```
> esphome:
>   name: ${hostname}
>   comment: ${comment}
>   platformio_options: 
>     board_build.flash_mode: dio
>     board_build.extra_flags:
>       #- "-DARDUINO_USB_MODE=0"
>       - "-DARDUINO_USB_CDC_ON_BOOT=0"
>   on_boot:
>     priority: 300
>     then:
>       - light.turn_on:
>           id: onboard_rgb
>           brightness: 20%
>           red: 0%
>           green: 100%
>           blue: 0%        
> 
> esp32:
>   board: esp32-c3-devkitm-1
>   framework:
>     type: arduino
> 
> 
> logger:
>   level: VERBOSE
>   #hardware_uart: USB_SERIAL_JTAG
>   hardware_uart: uart0
> 
> uart:  
>   - id: ac_port
>     baud_rate: 9600
>     tx_pin: 18
>     rx_pin: 19
> ```
> 
> I was never able to get it working with esp idf
> 
> So basically what @LayerCakeMakes did wrote here.
> 
> Regarding wifi issue, it might possibly be something regarding association, i did have that with esphome before, I changed 2.4 wifi to something like legacy N or something like that on the router and then it did boot fully, otherwise it would reset. Just a wild guess, not solution, but maybe worth a try? This happened to me long time ago and with 8266 I think.

`-DARDUINO_USB_CDC_ON_BOOT=0`, will lost all the uart port all, which can only into the dfu mode to reflash it.

screek-workshop, 2023-03-07:
> `-DARDUINO_USB_CDC_ON_BOOT=0`, will lost all the uart port all, which can only into the dfu mode to reflash it.

[image]
with a lot path, this can temp work with usb-cdc and uart at same time.