esphome/issues #3420: wifi_info Text Sensors do not work on ESP32; same code works on ESP8266

alexruffell, 2022-07-02:
### The problem

I am developing a salt level monitor and I like to display all the wifi connectivity parameters in the Diagnostic section of the device on HA. I am resusing code that works perfectly on an S31 outlet (ESP8266) but for some reason does not on an ESP32-WROOM-32D (as far as I can tell it is similar/same to nodeMCU ESP32 - nodemcu-32s)

This is what I get in HA:

[image]

There are no errors in the device log and the missing information shows just fine on the web page of the device itself (see below).

[image]

After numerous attempts, I have come to suspect that maybe there is something wrong with wifi_info text sensors on ESP32 specifically as they work just fine on ESP8266 in my ecosystem (therefore excluding a lot of other variables).

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7b2

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32D

### Component causing the issue

wifi_info

### Example YAML snippet

```yaml
substitutions:
  devicename: water-softener-monitor
  devicename_no_dashes: water_softener_monitor
  friendly_devicename: "Water Softener Monitor"
  device_description: "Water Softener Monitor"
  #Distance between bottom of tank and sensor in meters (m)
  distance_sensor_to_bottom: "0.90"

  
esphome:
  name: ${devicename}
  comment: ${device_description}  

esp32:
  board: nodemcu-32s
  framework:
    type: arduino
  

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicename} Hotspot"
    password: !secret iot_wifi_password
  
logger:
    baud_rate: 0 #disabled
    
  # Remove this line if you're not using Home Assistsant or your switch will restart every now and again
api:
  
ota:
  
web_server:
  port: 80
  include_internal: true
  
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${friendly_devicename}: IP"
      icon: "mdi:ip-outline"
      update_interval: 2s
    ssid:
      name: "${friendly_devicename}: SSID"
      icon: "mdi:wifi-settings"
      update_interval: 2s
    bssid:
      name: "${friendly_devicename}: BSSID"
      icon: "mdi:wifi-settings"
      update_interval: 2s
    mac_address:
      name: "${friendly_devicename}: MAC"
      icon: "mdi:network-outline"
    scan_results:
      name: "${friendly_devicename}: Wifi Scan"
      icon: "mdi:wifi-refresh"
      disabled_by_default: true
  
  # Device Specific Config
sensor:
  - platform: wifi_signal
    name: "${friendly_devicename}: WiFi Signal"
    update_interval: 60s
  - platform: ultrasonic
    trigger_pin: 23
    echo_pin: 22
    timeout: 4.0m     #This value should be close to actual limit of sensor
    update_interval: 5s
    name: "${friendly_devicename}: Sensor to salt"
    unit_of_measurement : "cm"
    filters:
    - lambda: return (x*100); #subtract distance measured to salt from total and convert from m to cm
    - filter_out: nan
    accuracy_decimals: 1
  - platform: ultrasonic
    trigger_pin: 23
    echo_pin: 22
    timeout: 4.0m     #This value should be close to actual limit of sensor
    update_interval: 5s
    name: "${friendly_devicename}: Salt Level"
    unit_of_measurement : "cm"
    filters:
    - lambda: return ((${distance_sensor_to_bottom} - x)*100); #subtract distance measured to salt from total and convert from m to cm
    - filter_out: nan
    accuracy_decimals: 1
  - platform: ultrasonic
    trigger_pin: 23
    echo_pin: 22
    timeout: 4.0m     #This value should be close to actual limit of sensor
    update_interval: 5s
    name: "${friendly_devicename}: Salt Level (%)"
    unit_of_measurement : "%"
    filters:
    - calibrate_linear:
        # Map 0.0 (from sensor) to 0.0 (true value)
        - 0.20 -> 100.0
        - 0.55 -> 50.0
        - 0.90 -> 0.0
    - lambda: return x;
    - filter_out: nan
    accuracy_decimals: 0

switch:
  - platform: restart
    name: "${friendly_devicename}: Restart"

#deep_sleep:
#  id: deep_sleep_1
#  sleep_duration: 30min
  
time:
  - platform: sntp
    id: ${devicename_no_dashes}_time

status_led:
  pin: GPIO2
```
```

### Anything in the logs that might be useful for us?

```txt
I compared logs of the ESP8266 to the logs of the ESP32 and they look the same. Both devices connect just fine to the network, and I can see all the details in the wifi section of the logs... just not in HA for the ESP32.
```

### Additional information

.

ssieb, 2022-08-02:
If the values are showing up in the logs and in the web server, then there is nothing wrong with esphome.  Try removing the device from the HA integration list and add it again.

closed 2022-09-16 by ssieb

MEOoms, 2022-12-29:
Same issue here, this has been closed but no resolution?