esphome/issues #3422: ili9341 triggers watchdog if nothing draws in the lambda

Aculeasis, 2022-07-03:
### The problem

I found that if nothing draws in the lambda or the lambda isn't declared, the watchdog timer will be triggered. I draw a dot to fix it. Is this a bug, isn't?

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.7

### What platform are you using?

ESP32

### Board

nodemcu esp32

### Component causing the issue

ili9341

### Example YAML snippet

```yaml
esphome:
  name: co2-station-test
  board: esp32dev
  platform: ESP32
  on_boot:
    priority: -100
    then:
      - lambda: |-
          id(tft_display).fill(Color::BLACK);
          id(tft_display).print(0, 0, id(my_font), id(red), TextAlign::TOP_LEFT, "Hello RED World!");
          id(tft_display).print(0, 40, id(my_font), id(green), TextAlign::TOP_LEFT, "Hello GREEN World!");
          id(tft_display).print(0, 80, id(my_font), id(blue), TextAlign::TOP_LEFT, "Hello BLUE World!");
logger:
api:
  password: !secret api_password
ota:
  password: !secret ota_password
wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: !secret wifi_ssid2
    password: !secret wifi_password2
  ap:
    ssid: "co2-station-test"
    password: !secret ap_password
captive_portal:
web_server:
  port: 80
  
font:
  - file: 'gfonts://Roboto'
    id: my_font
    size: 30

spi: 
  clk_pin: 25
  mosi_pin: 26
  miso_pin: 13

output:
  - platform: ledc
    pin: 32
    id: gpio_32_backlight_pwm

light:
  - platform: monochromatic
    output: gpio_32_backlight_pwm
    name: "ILI9341 Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON
    
color:
  - id: red
    red: 1
  - id: green
    green: 1
  - id: blue
    blue: 1
display:
  - platform: ili9341
    id: tft_display
    model: TFT 2.4R
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    update_interval: 10s
    auto_clear_enabled: false
    lambda: |-
      // It didn't help
      yield();
      // uncomment to fix watchdog
      // it.draw_pixel_at(0, 0, Color::BLACK);
      ESP_LOGD("TESTE", "OK");
```

### Anything in the logs that might be useful for us?

```txt
[00:02:50][D][TESTE:085]: OK
....
[00:02:58][D][TESTE:085]: OK
[00:03:08][D][TESTE:085]: OK
[00:03:13]E (24729) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[00:03:13]E (24729) task_wdt:  - loopTask (CPU 1)
[00:03:13]E (24729) task_wdt: Tasks currently running:
[00:03:13]E (24729) task_wdt: CPU 0: IDLE0
[00:03:13]E (24729) task_wdt: CPU 1: loopTask
[00:03:13]E (24729) task_wdt: Aborting.
[00:03:13]abort() was called at PC 0x4015760c on core 0
[00:03:13]
[00:03:13]ELF file SHA256: 0000000000000000
[00:03:13]
[00:03:13]Backtrace: 0x4008897c:0x3ffbf850 0x40088bf9:0x3ffbf870 0x4015760c:0x3ffbf890 0x40087091:0x3ffbf8b0 0x4016ed4b:0x3ffbc160 0x40158f17:0x3ffbc180 0x4008b3b9:0x3ffbc1a0 0x40089c0a:0x3ffbc1c0
[00:03:13]
[00:03:13]Rebooting...
[00:03:13]ets Jun  8 2016 00:22:57
[00:03:13]
[00:03:13]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[00:03:13]configsip: 0, SPIWP:0xee
[00:03:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[00:03:13]mode:DIO, clock div:2
[00:03:13]load:0x3fff0018,len:4
[00:03:13]load:0x3fff001c,len:1044
[00:03:13]load:0x40078000,len:8896
[00:03:13]load:0x40080400,len:5828
[00:03:13]entry 0x400806ac
[00:03:14][I][logger:243]: Log initialized
```

nielsnl68, 2022-09-14:
This will be fixed in as soon my PR is merge into the dev branch.
Or you can test out my reworked version on https://github.com/nielsnl68/switchplate by including:

```yaml
external_components:
    - source: github://nielsnl68/switchplate
      components: [ili9xxx]
      refresh: 10m

spi:
  mosi_pin: 23
  miso_pin: 19
  clk_pin: 18

display:
  - id: my_display
    platform: ili9xxx
    model: ili9341
    cs_pin: 5
    dc_pin: 16
    ...     
```

cheisig, 2022-09-25:
> This will be fixed in as soon my PR is merge into the dev branch. Or you can test out my reworked version on https://github.com/nielsnl68/switchplate by including:
>
> ```yaml
> external_components: [...]

@nielsnl68 
Is this fixing this as well?
https://github.com/esphome/issues/issues/3317

nielsnl68, 2022-09-25:
Yes, that is being fixed with my PR.

cheisig, 2022-09-25:
Lovely.
Thanks a lot!!

nielsnl68, 2022-09-25:
BTW i have also an enhanced version for the ili9xxx family drivers which can be found part of my switchplate project at:
https://github.com/nielsnl68/switchplate

you can easily load it as external_components:

```yaml
external_components:
    - source: githup://nielsnl68/switchplate
      components: [ili9xxx, ili9341]
      refresh: 10s
```
When you have time could you test it as well? Maybe you have other displays in the same family as well to test. I love to hear of ALL works well. But be aware it is still in development i could change thinks when needed.

Currently the driver component supports the ILI9341, ILI9342, ILI9481, ILI9486 and the ILI9488.

ircoopr, 2022-10-10:
@nielsnl68 - Do you have any knowledge of when your fix will be merged in to the DEV branch? I am having the following issue #3317 and hoping for your fix, but spotted there may be additional errors? (https://github.com/esphome/esphome/pull/3795#issuecomment-1270262065) 

Apologies, not super up on github and how these projects work (basically how and when a merge to DEV may happen), thanks for the hard work on this!

nielsnl68, 2022-10-10:
When you use the ili9341 then there is no easy, only when you want to use a bigger screen, then the memory issue will pop up.
I do not know when or if  @jesserockz will add my PR to dev. Better to ask @jesserockz on Discord.