esphome/issues #3423: ESP32: fatal error: IRsend.h: No such file or directory

ivankravets, 2022-07-04:
### The problem

Log: https://github.com/platformio/platformio-core/runs/7183702396?check_suite_focus=true

The `HeatpumpIR` library strictly depends on `IRremoteESP8266` and works only in ESP8266 environment https://github.com/ToniA/arduino-heatpumpir/blob/master/library.json#L22

There is 2.8 version which is compatible with ESP32/8266 https://registry.platformio.org/libraries/crankyoldgit/IRremoteESP8266

The solution is to strictly install `crankyoldgit/IRremoteESP8266@^2.8.2` on ESP32.

cc: @ToniA

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

pip

### What platform are you using?

ESP32

matphillips, 2022-08-19:
I get the same issue with esphone 2022.8.0

```
#include <IRsend.h>  // From IRremoteESP8266 library
                                                     ^
compilation terminated.
*** [.pioenvs/ir1/src/esphome/components/heatpumpir/heatpumpir.cpp.o] Error 1
In file included from src/esphome/components/heatpumpir/ir_sender_esphome.h:7:0,
                 from src/esphome/components/heatpumpir/ir_sender_esphome.cpp:1:
.piolibdeps/ir1/HeatpumpIR/IRSender.h:18:53: fatal error: IRsend.h: No such file or directory
```

YAML snippet:
```
substitutions:
  devicename: ir1
  upper_devicename: IR 1
  ip: 10.10.1.219

esphome:
  name: $devicename

esp32:
  board: wemos_d1_mini32

packages:
  standard: !include includes/standardConf.yaml

remote_transmitter:
  pin: GPIO32
  carrier_duty_percent: 50%

remote_receiver:
  id: reciever
  pin:
    number: GPIO17
    inverted: True
    mode: INPUT_PULLUP
  tolerance: 25%

#Air conditioner
climate:
  - platform: heatpumpir
    protocol: panasonic_dke
    name: "Bedroom AC"
    sensor: sensor_br

```

Cossid, 2022-10-23:
@rob-deutsch you're listed as the code owner for component heatpumpir.  It is currently failing test run by CI, causing all PRs to appear artificially broken.  Could you please take a look and get the tests updated as needed?

https://github.com/esphome/esphome/actions/runs/3307922703 is an example of CI failing because of heatpumpir code.