esphome/issues #3424: esp32-c3-devkitm-1 - arduino framework - wrong bootloader - "bootloader__80m.bin" nstead of "bootloader_dio_80m.bin"

mihsu81, 2022-07-05:
### The problem

The wrong bootloader is being selected when compiling for a lolin c3 mini for the android framework.
```
  framework:
    type: arduino
    version: dev
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

Error log:
```
Archiving /data/ble-gateway04/.pioenvs/ble-gateway04/libFrameworkArduino.a
Linking /data/ble-gateway04/.pioenvs/ble-gateway04/firmware.elf
RAM:   [==        ]  15.4% (used 50428 bytes from 327680 bytes)
Flash: [========  ]  76.1% (used 1396402 bytes from 1835008 bytes)
Building /data/ble-gateway04/.pioenvs/ble-gateway04/firmware.bin
esp32_create_combined_bin(["/data/ble-gateway04/.pioenvs/ble-gateway04/firmware.bin"], ["/data/ble-gateway04/.pioenvs/ble-gateway04/firmware.elf"])
usage: esptool merge_bin [-h] --output OUTPUT [--format {r,a,w}]
                         [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size FLASH_SIZE]
                         [--spi-connection SPI_CONNECTION]
                         [--target-offset TARGET_OFFSET]
                         [--fill-flash-size FILL_FLASH_SIZE]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: argument <address> <filename>: [Errno 2] No such file or directory: '/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/bin/bootloader__80m.bin'
*** [/data/ble-gateway04/.pioenvs/ble-gateway04/firmware.bin] Explicit exit, status 2
========================= [FAILED] Took 110.11 seconds =========================
```

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.7

### What platform are you using?

ESP32

### Board

lolin c3 mini

### Example YAML snippet

```yaml
esphome:
  name: ble-gateway04
  platformio_options:
    board_build.variant: esp32c3
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: dev
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

### Anything in the logs that might be useful for us?

```txt
Archiving /data/ble-gateway04/.pioenvs/ble-gateway04/libFrameworkArduino.a
Linking /data/ble-gateway04/.pioenvs/ble-gateway04/firmware.elf
RAM:   [==        ]  15.4% (used 50428 bytes from 327680 bytes)
Flash: [========  ]  76.1% (used 1396402 bytes from 1835008 bytes)
Building /data/ble-gateway04/.pioenvs/ble-gateway04/firmware.bin
esp32_create_combined_bin(["/data/ble-gateway04/.pioenvs/ble-gateway04/firmware.bin"], ["/data/ble-gateway04/.pioenvs/ble-gateway04/firmware.elf"])
usage: esptool merge_bin [-h] --output OUTPUT [--format {r,a,w}]
                         [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size FLASH_SIZE]
                         [--spi-connection SPI_CONNECTION]
                         [--target-offset TARGET_OFFSET]
                         [--fill-flash-size FILL_FLASH_SIZE]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: argument <address> <filename>: [Errno 2] No such file or directory: '/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/bin/bootloader__80m.bin'
*** [/data/ble-gateway04/.pioenvs/ble-gateway04/firmware.bin] Explicit exit, status 2
========================= [FAILED] Took 110.11 seconds =========================
```

nielsnl68, 2022-07-05:
could you try the following:
```yaml
esp32:
  board: lolin_c3_mini
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

This works for me, using the esphome *dev* branch.

closed 2022-07-05 by mihsu81

evlo, 2022-09-18:
When I tried
```
type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```
I did end up with 
```
error: cannot convert 'HWCDC*' to 'HardwareSerial*' in assignment
     this->hw_serial_ = &Serial;
```

mihsu81, 2022-09-18:
Your indentation is wrong.

Mine is:
```yaml
esp32:
  board: lolin_c3_mini
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

evlo, 2022-09-18:
I think that was just copy paste
```
esphome:
  name: elektromer
  #platform: esp32
#  platformio_options:
#    board_build.variant: esp32c3
#    board_build.flash_mode: dio
esp32:
  board: lolin_c3_mini
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
#esp32:
#  board: esp32-c3-devkitm-1
  #board: lolin_c3_mini
  # framework:
  #   type: arduino
  #   #version: dev
  #   #platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
  #   platform_version: 5.1.1 #4.4.0
  #   version: 2.0.4
#  framework:
#    type: esp-idf
#    version: 5.1.1
#  variant: esp32c3
#  framework:
#    type: esp-idf

wifi:
  use_address: 192.168.50.63
  ap:

api:
logger:
  baud_rate: 0
ota:
#web_server:
#captive_portal:

external_components:
  - source: github://evlo/esphome_ZPAZE312
    components: [obis]
    refresh: 0s

uart:
  id: uart_bus
  tx_pin: GPIO21
  rx_pin: GPIO20
  #baud_rate: 9600
  baud_rate: 300
  data_bits: 8
  stop_bits: 1
  parity: NONE
  debug: 
    direction: BOTH
    dummy_receiver: true
  #data_bits: 7
  #parity: ODD
  rx_buffer_size: 1700

obis:
  uart_id: uart_bus
  force_update:  # optional
    interval: 10 #60
    payload: "\x2F\x3F\x21\x0D\x0A"

sensor:
  - platform: obis
    channel: "21.8.0"
    name: "L1"
    unit_of_measurement: "kWh"
    accuracy_decimals: 3

  - platform: obis
    channel: "41.8.0"
    name: "L2"
    unit_of_measurement: "kWh"
    accuracy_decimals: 3

  - platform: obis
    channel: "61.8.0"
    name: "L3"
    unit_of_measurement: "kWh"
    accuracy_decimals: 3
```
Anyway now i'm not able to get to the bootloader error and always  end up with
```
Compiling /data/elektromer/.pioenvs/elektromer/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
src/esphome/components/uart/uart_component_esp32_arduino.cpp: In member function 'virtual void esphome::uart::ESP32ArduinoUARTComponent::setup()':
src/esphome/components/uart/uart_component_esp32_arduino.cpp:90:25: error: cannot convert 'HWCDC*' to 'HardwareSerial*' in assignment
     this->hw_serial_ = &Serial;
                         ^~~~~~
*** [/data/elektromer/.pioenvs/elektromer/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o] Error 1
```
even when i deleted whole .esphome folder

so I probably changed something etc.

So different issue