esphome/issues #3426: http_request feature is incompatible with esp-idf

464d41, 2022-07-07:
### The problem

Esphome config fails to compile as soon as http_request component appears in config.

```
ubuntu$ esphome run esp32c3.yml 
INFO Reading configuration esp32c3.yml...
Failed config

http_request: [source esp32c3.yml:46]
  
  This feature is incompatible with esp-idf.
  useragent: esphome/weightscale
  timeout: 5s
```

PS: I'm new to Esphome. Let me know if anything else is needed for debugging. Thanks!

### Which version of ESPHome has the issue?

Version: 2022.6.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

http_request

### Example YAML snippet

```yaml
esphome:
  name: esp32c3
  platformio_options:
    board_build.variant: esp32c3
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

http_request:
  useragent: esphome/weightscale
  timeout: 5s
```

### Anything in the logs that might be useful for us?

```txt
ubuntu$ esphome run esp32c3.yml 
INFO Reading configuration esp32c3.yml...
Failed config

http_request: [source esp32c3.yml:46]
  
  This feature is incompatible with esp-idf.
  useragent: esphome/weightscale
  timeout: 5s
```

tarte11, 2022-09-21:
I have a very similar issue with shelly plus i4 witch use esp32 board and the Configuration for generating Home Assistant Events and Multi-Click Actions from esphome example. As soon as I add http_request, compilation fail.

ashp8i, 2022-10-11:
you can compile esphome using arduino framework as this is a regular esp32 with one core disabled running at lower clock speed use the below: -

```
esphome:
  name: ${device_name}
  comment: ${device_description}
  platformio_options:
    platform_packages:
    - framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/2.0.5/framework-arduinoespressif32-solo1.zip
    board_build.f_cpu: 160000000L
  project:
    name: "${project_name}"
    version: "${project_version}"

esp32:
  board: esp32doit-devkit-v1
  variant: esp32
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0
```

That uses the arduino core from tasmota compiled for esp32-solo1

UnaiUribarri-TomTom, 2023-09-16:
Arduino is not compatible with other options, like sending serial to USB port in ESP32S2 to be able to use both ports in my application.

It will be great to have http_request support...

grischard, 2023-10-24:
Linking the relevant PR: https://github.com/esphome/esphome/pull/3256

ESP-IDF transition project board: https://github.com/orgs/esphome/projects/5/views/1

digaus, 2023-12-18:
@grischard 

I am not seeing OTA via web_server listed there? Will this come eventually?

grischard, 2023-12-18:
@digaus I think that's not on the roadmap actually - most people upload ota over the api, I guess? Feel free to open a ticket if you need it.

digaus, 2023-12-18:
> @digaus I think that's not on the roadmap actually - most people upload ota over the api, I guess? Feel free to open a ticket if you need it.

Will do. Problem is I want to deploy a device at friends house who does not run HA and for updates/changes I plan to compile the binary for him to update his device. 

Any other possible solutions for this case? (need ESP-IDF because of size limitation)

closed 2024-06-09 by kbx81