esphome/issues #3430: Specifying library source fails

aeonsablaze, 2022-07-08:
### The problem

The `name=source` syntax for specifying an alternative git source for a library is not working. Using the exact example from the "ESPHome Core Configuration" documentation page (i.e. `esphome/Improv=https://github.com/improv-wifi/sdk-cpp.git#v1.0.`) fails to compile. I have attached the full log output, but the critical piece seems to be:
`fatal: protocol 'esphome/Improv=https' is not supported`
which suggests that it is not properly splitting the string at the `=` to separate the name from the url.
[logs_livingroom-notifier_upload.txt](https://github.com/esphome/issues/files/9074517/logs_livingroom-notifier_upload.txt)
 

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.7

### What platform are you using?

ESP32

### Board

esp32 devkit v1

### Component causing the issue

core

### Example YAML snippet

```yaml
esphome:
  libraries:
    - esphome/Improv=https://github.com/improv-wifi/sdk-cpp.git#v1.0.0
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/livingroom-notifier.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing livingroom-notifier (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing git+esphome/Improv=https://github.com/improv-wifi/sdk-cpp.git#v1.0.0
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-nogml4yt'...
fatal: protocol 'esphome/Improv=https' is not supported
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'v1.0.0', 'esphome/Improv=https://github.com/improv-wifi/sdk-cpp.git', '/data/cache/platformio/cache/tmp/pkg-installing-nogml4yt']:
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/data/cache/platformio/platforms/espressif32/builder/main.py", line 233:
    target_elf = env.BuildProgram()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.3.0/SCons/Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/platformio.py", line 62:
    env.ProcessProjectDeps()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.3.0/SCons/Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/data/cache/platformio/packages/tool-scons/scons-local-4.3.0/SCons/Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/piolib.py", line 1104:
    project.install_dependencies()
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/piolib.py", line 940:
    lm.install(spec)
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_install.py", line 48:
    pkg = self._install(
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/library.py", line 93:
    return super(LibraryPackageManager, self)._install(
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/manager/_install.py", line 142:
    assert vcs.export()
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/vcsclient.py", line 200:
    assert self.run_cmd(args, cwd=os.getcwd())
  File "/usr/local/lib/python3.9/dist-packages/platformio/package/vcsclient.py", line 117:
    raise VCSBaseException("VCS: Could not process command %s" % e.cmd)
========================== [FAILED] Took 1.90 seconds ==========================
```

ssieb, 2022-08-02:
If a string contains a colon, you have to quote it, otherwise the yaml parser will have trouble with it.

aeonsablaze, 2022-08-02:
Quoting doesn't make any difference, I've tried both double and single quotes around the whole thing, around just the url and around just the name. After doing a bit of digging I believe the root of the problem is that the platformio library can't parse the "Custom Name" of the library if it has a slash in it. I'm not sure if this is a bug in that library or if it is intended behaviour. If the latter, then the config example at [ESPHome Core Configuration](https://esphome.io/components/esphome.html#libraries) would need to be changed accordingly.

hazi, 2022-08-11:
I have identified the same problem.

elafargue, 2022-11-12:
Same issue here, the example in documentation at https://esphome.io/components/esphome.html#libraries fails, I haven't been able to work out how to override a default library yet...

mzakharo, 2023-02-06:
This is still an issue for me. I can make things clone and build the lib by removing  everything before and including the first  `/` , for example:

this:
```
  libraries:
    - fastled/FastLED=https://github.com/mzakharo/FastLED.git
 ```
 
 becomes this:
 
 ```
  libraries:
    - FastLED=https://github.com/mzakharo/FastLED.git
 ```
 
 However, my build still fails due to duplicate implementation of functions, as it does not remove the  original `fastled/FastLED` dependency, provided by default esphome when using a `- platform: fastled_spi` light component. 

From the build log:
 ```
Dependency Graph
|-- FastLED @ 3.5.0+sha.3d2ab78
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- FastLED @ 3.3.2
```

Can anyone please help with this? What is missing?

mzakharo, 2023-02-07:
Looks like a possible solution is changing [core/__init__.py, add_library() function](https://github.com/esphome/esphome/blob/dev/esphome/core/__init__.py#L655) to this: 

```python
        for other in self.libraries[:]:
            if other.name is None or library.name is None:
                continue
            library_name = library.name if '/' not in library.name else library.name.split('/')[1]
            other_name = other.name if '/' not in other.name else other.name.split('/')[1]
            if other_name != library_name:
                continue
```

Thoughts?

nielsnl68, 2023-02-07:
Did you have a look at my PR of the FastLed fixup? https://github.com/esphome/esphome/pull/4257

mzakharo, 2023-02-07:
> Did you have a look at my PR of the FastLed fixup? [esphome/esphome#4257](https://github.com/esphome/esphome/pull/4257)

 ESPHome users should be allowed to override any library supplied by default installation.  In my specific case,  the above mentioned PR does not solve the problem due to https://github.com/FastLED/FastLED/issues/1475  I have a fork of the repo with the fix here: https://github.com/FastLED/FastLED/compare/master...mzakharo:FastLED:master . I would like to tell ESPHome to use this fork somehow.  Any ideas on how best to proceed?