esphome/issues #3432: Shelly EM 160°C

dwd1, 2022-07-09:
### The problem

hi,
i have a shelly em which reports 150°C with the following config.
anybody knows the correct config?
thank you

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
- platform: ntc
    sensor: shelly_em_temp_resistance_reading
    name: "Shelly EM Temperature"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
  - platform: resistance
    id: shelly_em_temp_resistance_reading
    sensor: shelly_em_temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
  - platform: adc
    id: shelly_em_temp_analog_reading
    pin: A0
```

pedroona, 2022-08-10:
Same problem here. Did you find a solution @dwd1 ?

jesserockz, 2022-08-11:
Closing as this is not really an "issue" with ESPHome, just a misconfiguration.

I personally do not know the correct configuration for these sensors sorry.

closed 2022-08-11 by jesserockz