esphome/issues #3433: Debug Component doesn't work on ESP32-S2 and ESP32-S3

codersaur, 2022-07-09:
### The problem

Compilation errors when trying to use the debug component on ESP32-S2 and ESP32-S3 devices.

### Which version of ESPHome has the issue?

2022.7.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

um_feathers3

### Component causing the issue

debug

### Example YAML snippet

```yaml
esphome:
  name: $device_name

esp32:
  board: um_feathers3
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.0.0

...

debug:
  update_interval: 5s

sensor:
  - platform: debug
    free:
      name: "${device_name}: Heap Free"
      entity_category: "diagnostic"
    # fragmentation:
    #   name: "Heap Fragmentation"
    block:
      name: "${device_name}: Heap Max Block"
      entity_category: "diagnostic"
    loop_time:
      name: "${device_name}: Loop Time"
      entity_category: "diagnostic"
```

### Anything in the logs that might be useful for us?

```txt
Compilation error message:

      In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_rom/include/esp32/rom/rtc.h:18,
                       from src/esphome/components/debug/debug_component.cpp:15:
      /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_rom/include/esp32/rom/ets_sys.h:29:2: error: #error "This header should only be included when building for ESP32"
       #error "This header should only be included when building for ESP32"
        ^~~~~
```

### Additional information

Solution: debug_component.cpp needs updating to include chip-specific rtc.h file (i.e. esp32s2/rom/rtc.h, esp32s3/rom/rtc.h).
Additionally, reset_reason and wakeup_reason switch statements need updating with new values.

spali, 2022-11-12:
Can confirm this on `2022.10.2` with `esp-idf` framework and another S3.

this works:
```yaml
esphome:
  name: test

esp32:
  board: adafruit_feather_esp32s3_nopsram
  variant: esp32s3
  framework:
    type: esp-idf
    version: 4.4.2
    platform_version: 5.2.0

logger:
```

this fails with the OP mentioned error:
```yaml
esphome:
  name: test

esp32:
  board: adafruit_feather_esp32s3_nopsram
  variant: esp32s3
  framework:
    type: esp-idf
    version: 4.4.2
    platform_version: 5.2.0

logger:
debug:
  update_interval: 5s
```

andorardo, 2023-01-03:
I'm seeing the same error on a standard ESP32-S3-DevKitC-1.

ra1der75, 2023-03-12:
I'm seeing the same error on a standard ESP32-S3

yousaf465, 2023-03-16:
same error on ESP2-S3

danutchereches, 2023-05-10:
I got the same error on a lolin_s3.

gpambrozio, 2023-08-20:
still happens.

closed 2023-08-20 by jesserockz

TimKraemer1, 2023-11-13:
Getting the same error on ESP32 C3 Development board when including "ets_sys.h" library in my main project file