esphome/issues #3435: Cannot install properly ESPHome on Windows 11

stefancvetkovic, 2022-07-09:
### The problem

I do a command `pip3 install esphome` and apparently it installs it.
However, when I try to run command `esphome run hallway_tagreader.yaml` it shows me an error
```
F:\esphome> esphome run .\tagreader_hallway.yaml
INFO Reading configuration .\tagreader_hallway.yaml...
WARNING 'hallway_tagreader': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Compiling app...
*******************************************************************************************************************************************************************************************************************************************************************************************************************************

Obsolete PIO Core v5.2.5 is used (previous was 6.0.2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
*******************************************************************************************************************************************************************************************************************************************************************************************************************************

Processing hallway_tagreader (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Error: Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Python310\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Python310\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Python310\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Python310\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Python310\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Python310\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Python310\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Python310\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Python310\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Python310\lib\site-packages\platformio\commands\run\processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "C:\Python310\lib\site-packages\platformio\commands\platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "C:\Python310\lib\site-packages\platformio\platform\factory.py", line 52, in new
    pkg = PlatformPackageManager().get_package(pkg_or_spec)
  File "C:\Python310\lib\site-packages\platformio\package\manager\base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "C:\Python310\lib\site-packages\platformio\package\manager\base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "C:\Python310\lib\site-packages\platformio\package\meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "C:\Python310\lib\site-packages\platformio\package\meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "C:\Python310\lib\site-packages\platformio\package\meta.py", line 393, in load
    data["spec"] = PackageSpec(**data["spec"])
TypeError: PackageSpec.__init__() got an unexpected keyword argument 'uri'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

none

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

CLI - command line interperetter

deanrock, 2022-07-10:
I have the same issue with the latest version of `esphome==2022.6.2`, installed in a fresh Python 3.9 virtual environment, on macOS Monterey.

deanrock, 2022-07-11:
Nevermind, I have solved my issue by removing `~/.platformio` folder.

Seems like I had globally installed `platformio` (probably the one used by VSCode), and it cannot handle having different versions, since they all use `~/.platformio`.

tmarkson, 2022-07-28:
Removing the `.platformio` folder worked for me too on Windows 10.  It's located at `C:\Users\user\.platformio`

ssieb, 2022-08-02:
Somehow you upgraded platformio and then when you use an older one with esphome, it has a problem with the cached data.  Clearing the .platformio directory is the solution.

closed 2022-08-02 by ssieb