esphome/issues #3440: ESP32 Touch + Deep Sleep not working, fully implemented?

AndreasBVB, 2022-07-13:
### The problem

Hi all,
New to the project but navigating with some basic C++ understandings.

I am trying to build a project using touch pins to detect moisture, while sending the device into deep sleep as much as possible so that it can be driven by batteries. What I have noticed is: any touch pin reading change, even though well above touch binary sensor thresholds, will wake up the device with a sensor OFF message and wake up reason 2: ESP_SLEEP_WAKEUP_EXT0, Wakeup caused by external signal using RTC_IO.

I checked the ESP32 implementation and found the "touchAttachInterrupt" needs to be called to specify the threshold, for example here:
https://randomnerdtutorials.com/esp32-touch-wake-up-deep-sleep/

I then searched around the entire ESPHome project but didn't find any call to this function. This makes me wonder if touch wake up from deep sleep has been fully implemented. Kindly suggest if I missed something. Thanks you!

### Which version of ESPHome has the issue?

2022.6.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

all ESP32 boards

### Example YAML snippet

```yaml
esphome:
  name: abcde-04aabb
  on_boot:
    priority: 900
    then:
      lambda: |-
        id(wake_up_reason) = esp_sleep_get_wakeup_cause();

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: NONE

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

deep_sleep:
  run_duration: 10s
  sleep_duration: 3590s
  touch_wakeup: true
  wakeup_pin: T4
  wakeup_pin_mode: KEEP_AWAKE

esp32_touch:
  iir_filter: 10ms
  setup_mode: false

binary_sensor:
  - platform: esp32_touch
    name: esphome-leak
    device_class: moisture
    pin: T4
    threshold: 100

globals:
  - id: wake_up_reason
    type: int
    restore_value: no
    initial_value: '0'
    
sensor:
  - platform: template
    name: "Wake-up Reason"
    accuracy_decimals: 0
    lambda: |-
      return (id(wake_up_reason));

mqtt:
  broker: *****
  username: ******
  password: ******
  birth_message:
   topic: /status
   payload: "Online"
  will_message:
   topic: /status
   payload: "Offline"
```

devoh747, 2022-07-13:
I noticed a warning in https://github.com/espressif/esp-idf/releases/tag/v4.4.1

On ESP32-S3, using touch sensor together with sleep modes is not supported yet. Please avoid using this feature in v4.4.1 release, as it may cause permanent damage to the chip.  It claims it should be fixed in 4.4.2   hmmm  but you are using just an ESP32 not a ESP32-S3.  Sorry this is probably not related.

machv, 2022-11-11:
Hello,
I just run into same issue with my project, where board is waking up all the time. I am using the latest ESPHome (2022.10.2). Have you come up with any workaround, please?

mauromorello, 2023-12-06:
Some news? I'm still hangin on a LOLIN32 Lite; with arduino code everything was ok, buto with ESPhome I'm struggling to get it working. It always wakes up milliseconds after entering deep sleep

lhac5vet, 2023-12-23:
I'm using a firebeetle 2 ESP32 board (ESP-WROOM-32E) and I'm facing the same problem when I try to setup the touch_wakeup feature in ESPHome. No way to wake up with a touch. Searching in the ESPHome component source code, I found the following in the **esp32_touch.cpp** starting line 59:

```
#if defined(USE_ESP32_VARIANT_ESP32S2) || defined(USE_ESP32_VARIANT_ESP32S3)
    touch_pad_config(child->get_touch_pad());
#else
    // Disable interrupt threshold
    touch_pad_config(child->get_touch_pad(), 0);
#endif
```
In case of the ESP32 standard variant (#else case), the "touch_pad_config()" has been forced with a threshold value 0. To confirm this, I set my board in deep_sleep and then I tied my GPIO pin shortly to GND to force a threshold value to 0.  And, as expected the system wakes up !
The next step is to test the modification in the source code and recompile it... if anyone can confirm it would be cool !

IMO, no need to differentiate if it is ESP32 or ESP32S2 and ESP32S3 anymore . The test on the "define" can be removed and the above 6 lines of code is reduced to only 1 line. 

`touch_pad_config(child->get_touch_pad());
`

Hope it helps.