esphome/issues #3443: Interlock Substitution

ChrisPotDev, 2022-07-15:
### The problem

Interlock does not support substitution

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.4

### What platform are you using?

ESP8266

### Board

Sonoff 4ch

### Component causing the issue

GPIO Switch Interlock

### Example YAML snippet

```yaml
interlock: &interlock_group [${deviceid}_${device_type}_switch_2, ${deviceid}_${device_type}_switch_3, ${deviceid}_${device_type}_switch_4]
```

### Anything in the logs that might be useful for us?

```txt
while parsing a flow sequence
  in "/config/esphome/device/4ch_fan/switch/switch_4ch_switch_2.yaml", line 12, column 29:
    interlock: &interlock_group [${deviceid}_${device_type}_swit ... 
                                ^
expected ',' or ']', but got '{'
  in "/config/esphome/device/4ch_fan/switch/switch_4ch_switch_2.yaml", line 12, column 31:
    interlock: &interlock_group [${deviceid}_${device_type}_switch ...
```

ssieb, 2022-11-13:
There's no benefit to using substitutions for ids anyway.  It just makes things messy.

ssieb, 2022-11-13:
It works if you use quotes around it.

bugeaud, 2025-01-09:
Substitution are required to centralize parameter, modularize and templatize & al that contains interlock (unless there is a better way to do component parameters at esphome). I don't see why the $ synthax is forbiden at any place. Is it possible to reconsider having substitution all around ? And as a start at interlock level ...