esphome/issues #3446: Shelly Dimmer 2 freezes and maintains light at full brightness after it's turned on

poldim, 2022-07-16:
### The problem

The dimmer connects to wifi and shows up in the dashboard as expected.  I can connect and see logs.  But if I turn it on, it goes to 100% brightness and then does not respond to any lighting changes.  IE trying to dim it, turn it off, etc.  The only thing I can do is reboot the ESP via a soft restart switch.

### Which version of ESPHome has the issue?

v2022.7.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.7.5

### What platform are you using?

ESP8266

### Board

shelly

### Component causing the issue

shelly_dimmer

### Example YAML snippet

```yaml
logger:
  baud_rate: 0

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 115200

light:
- platform: shelly_dimmer
  name: "${friendly_name}"
  id: this_light
  power:
    name: "${friendly_name} Power"
  voltage:
    name: "${friendly_name} Voltage"
  current:
    name: "${friendly_name} Current"
  leading_edge: true
  # warmup_brightness: 200
  # min_brightness: -100
  # max_brightness: 500
  firmware:
    version: "51.6"
    update: true
  # gamma_correct: 1
  # default_transition_length: 1s
  restore_mode: RESTORE_DEFAULT_OFF

sensor:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome-dining-room-chandelier.yaml...
INFO Starting log output from esphome-dining-room-chandelier.local using esphome API
INFO Successfully connected to esphome-dining-room-chandelier.local
[07:54:03][I][app:102]: ESPHome version 2022.7.0-dev compiled on Jul 14 2022, 23:54:07
[... 323 lines cut ...]
[07:54:34][D][shelly_dimmer:281]: Command sent, waiting for reply
[07:54:34][D][shelly_dimmer:391]: Got frame: 0x01
[07:54:34][D][shelly_dimmer:152]: Brightness update: 593 (raw: 0.593127)
[07:54:34][D][shelly_dimmer:269]: Sending command: 0x01 (2 bytes) payload 0x5102
[07:54:34][D][shelly_dimmer:281]: Command sent, waiting for reply
[07:54:34][D][shelly_dimmer:391]: Got frame: 0x01
[07:54:34][D][shelly_dimmer:152]: Brightness update: 548 (raw: 0.548662)
[07:54:34][D][shelly_dimmer:269]: Sending command: 0x01 (2 bytes) payload 0x2402
[07:54:34][D][shelly_dimmer:281]: Command sent, waiting for reply
[07:54:34][D][shelly_dimmer:391]: Got frame: 0x01
[07:54:34][D][shelly_dimmer:152]: Brightness update: 505 (raw: 0.505625)
[07:54:34][D][shelly_dimmer:269]: Sending command: 0x01 (2 bytes) payload 0xf901
[07:54:34][D][shelly_dimmer:281]: Command sent, waiting for reply
[07:54:34][D][shelly_dimmer:391]: Got frame: 0x01
[07:54:34][D][shelly_dimmer:152]: Brightness update: 460 (raw: 0.460671)
[07:54:34][D][shelly_dimmer:269]: Sending command: 0x01 (2 bytes) payload 0xcc01
[07:54:34][D][shelly_dimmer:281]: Command sent, waiting for reply
[07:54:34][D][shelly_dimmer:391]: Got frame: 0x01
[07:54:34][D][shelly_dimmer:152]: Brightnes
```
```

### Additional information

I previously had a shelly dimmer v1 installed on this circuit running original mgOS and it had been dimming the circuit without issues.  This eliminates any concern about installation wiring/bulbs.

poldim, 2022-07-29:
@edge90 @rnauber - since you added https://github.com/esphome/esphome/pull/2954 wondering if you have any thoughts here or have experienced this on your shelly's

scootec, 2022-08-02:
@poldim I had a similar issue with my recently flashed dimmer 2. Mine would turn on successfully, but when turned off the shelly would shoot to 100% and complete ignore any lighting commands until a restart.

In my case, adding a minimum brightness has prevented the issue from reoccurring so far. I gave the minimum brightness a go since my issue was only happened when the light running at a very low brightness during the turn on phase.

Not sure if it will help you, but mentioning here just in case!

edge90, 2022-09-22:
I will have a look as soon as I get some sparetime.

poldim, 2022-09-26:
For me, the problem was partially solved by rolling back the fw to `version: "51.5"`

Now I still get a "brightness pop" when turning the light off.  For example, if the light is dimmed to 50% and I send a turn off command in HA, it uses the esphome's transition setting to dim the light from 50% to 0%, but once it reaches at or near 0% it appears to have a momentary burst to something very bright (50-100%) and then immediately cuts to full off.

edge90, 2022-09-27:
What kind of bulb are you using, led? If so, try to change leading_edge to false.

harmptor, 2023-12-15:
I've had a very similar experience with my LED light from IKEA that says it's non-dimmable. (It can be dimmed, though with a quiet buzzing sound when between ~10% and ~90% brightness). When I set `leading_edge: true` and turned it on, I couldn't control my shelly until I flipped the breaker. With trailing edge and 51.5 it works fine though :)

edge90, 2023-12-28:
All lead lights should be trailing edge.

Sperryfreak01, 2024-07-01:
Had a similar issue for me it was two config settings.  The Gramma value = 0 and the min brightness level = 470, both of which were in the example code.  

These two settings combined caused the dimming curve to be so far up to effectively prevent dimming.  Setting min to 0 and gamma to 2.8 allowed the lights to be usable long enough to figure out the appropriate minimum brightness threshold.