esphome/issues #3448: Problems with updating Xiaomi Mi Desk Lamp 1S

clipse2004, 2022-07-17:
### The problem

Hello all, 

so far the update of the Xiaomi Mi Desk Lamp 1s has always worked without problems, also I have not changed anything in the configuration. 
Recently I can no longer do this, it always comes the message that no package for 'linux_x86_64' can be found. 

Can you help? 

Thx

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.5

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

tasmota/framework-arduinoespressif32 @ 3.10006.210420

### Example YAML snippet

```yaml
substitutions:
  friendly_name: Schreibtischlampe
  device_name: schreibtischlampe

esphome:
  name: ${device_name}
  comment: ${friendly_name}
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.5.0
    platform_packages:
      - tasmota/framework-arduinoespressif32 @ 3.10006.210420

packages:
  web_server: !include common/web_server.yaml
  api: !include common/api.yaml
  ota: !include common/ota.yaml
  time: !include common/time.yaml

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_password
  fast_connect: true
  ap:
    ssid: ${friendly_name} Hotspot
    password: !secret ap_password
    ap_timeout: 1min
  manual_ip:
    static_ip: 192.168.3.183
    gateway: 192.168.3.1
    subnet: 255.255.255.0
    
captive_portal:

logger:

# Text sensors with general information
text_sensor:
  - platform: version
    name: ${friendly_name} Version
    hide_timestamp: true

sensor:
  - platform: rotary_encoder
    id: rotation
    pin_a: GPIO27
    pin_b: GPIO26
    resolution: 2
    on_value:
      then:
        - if:
            condition:
              # Check if Button is pressed while rotating
              lambda: 'return id(button1).state;'
            then:
              # If Button is pressed, change CW/WW
              - lambda: |-
                  auto min_temp = id(light1).get_traits().get_min_mireds();
                  auto max_temp = id(light1).get_traits().get_max_mireds();
                  auto cur_temp = id(light1).current_values.get_color_temperature();
                  auto new_temp = max(min_temp, min(max_temp, cur_temp + (x*20)));
                  auto call = id(light1).turn_on();
                  call.set_color_temperature(new_temp);
                  call.perform();
            else:
              # If Button is not pressed, change brightness
              - light.dim_relative:
                  id: light1
                  relative_brightness: !lambda |-
                    return x / 10.0;
        # Reset Rotation to 0
        - sensor.rotary_encoder.set_value:
            id: rotation
            value: 0

binary_sensor:
  - platform: gpio
    id: button1
    pin:
      number: GPIO33
      inverted: True
      mode: INPUT_PULLDOWN
    on_click: 
      then:
      # use if-condition instead of toggle to set full brightness on turn_on
        - if:
            condition:
              light.is_on: light1
            then:
              - light.turn_off:
                  id: light1
            else:
              - light.turn_on:
                  id: light1
                  brightness: 100%
                  color_temperature: 2700 K

button:
  - <<: !include common/restart_button.config.yaml
#  - platform: restart
#    name: "Schreibtischlampe Restart"

output:
  - platform: ledc
    pin: GPIO2
    id: output_cw
    frequency: 40000Hz
    power_supply: power
  - platform: ledc
    pin: GPIO4
    id: output_ww
    power_supply: power
    frequency: 40000Hz
    
power_supply:
  - id: power
    pin: GPIO12
    enable_time: 0s
    keep_on_time: 0s

light:
  - platform: cwww
    id: light1
    default_transition_length: 0s
    constant_brightness: true
    name: ${friendly_name}
    cold_white: output_cw
    warm_white: output_ww
    cold_white_color_temperature: 4800 K
    warm_white_color_temperature: 2500 K #2500k is the original value of the lamp. To correct binning for 2700k to look more like 2700k use 2650k instead
    gamma_correct: 0
```

### Anything in the logs that might be useful for us?

```txt
Error: Could not find the package with 'tasmota/framework-arduinoespressif32 @ 3.10006.210420' requirements for your system 'linux_x86_64'
```

Raukze, 2022-09-08:
I get the same log output.

popy2k14, 2022-09-09:
Hey using another esphome node settings which works for my xiaomi desk lamp 1s:

```
esphome:
  name: midesklamp1s
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
    platform:
      - espressif32@1.11.0
  
esp32:
  board: esp32doit-devkit-v1
```

But sadly my OTAs wont work. They always fails with:

`OTA update results in failure: Error sending data: [Errno 104] Connection reset by peer`

Any hints?