esphome/issues #3450: SPI bus configuration fails on ESP32S2 devices

rosterloh, 2022-07-18:
### The problem

While trying to configure a SPI bus on an ESP32S2 using the Arduino framework the build fails due to VSPI not being defined for this platform. For ESP32S2 and ESP32S3 FSPI needs to be used

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

um_feathers2

### Component causing the issue

SPI

### Example YAML snippet

```yaml
esphome:
  name: spi-example

esp32:
  board: um_feathers2
  variant: esp32s2
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.0.0

spi:
  clk_pin: GPIO36
  mosi_pin: GPIO35
  miso_pin: GPIO37
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/office-monitor/src/esphome/components/spi/spi.cpp.o
Compiling .pioenvs/office-monitor/src/main.cpp.o
src/esphome/components/spi/spi.cpp: In member function 'virtual void esphome::spi::SPIComponent::setup()':
src/esphome/components/spi/spi.cpp:79:36: error: 'VSPI' was not declared in this scope
       this->hw_spi_ = new SPIClass(VSPI);  // NOLINT(cppcoreguidelines-owning-memory)
                                    ^~~~
src/esphome/components/spi/spi.cpp:79:36: note: suggested alternative: 'SPI'
       this->hw_spi_ = new SPIClass(VSPI);  // NOLINT(cppcoreguidelines-owning-memory)
                                    ^~~~
                                    SPI
```

### Additional information

# Useful Links
- [Example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino#L42)
- [Source](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.cpp#L346)

nielsnl68, 2022-07-18:
Hi, this is already an known issue. Somehow the VSPI define is not defined under ESP32-c3 and it seems now that it does not for the esp32-s2 as well.

on my github page at https://github.com/nielsnl68/switchplate i have added a temporary fix for it.

you can load it by adding:

```yaml
external_components:
  # use rtttl and dfplayer from ESPHome's dev branch in GitHub
  # equivalent shorthand for GitHub
  - source: github://nielsnl68/SwitchPlate
    components: [ spi ]
```
the only thing it does is replacing VSPI with 3. And that seems to work.

landonr, 2022-09-14:
I can't get the above to work on a T-Display S3 with an esp32s3. It say's spi component not found. Here's my current config 
```
esphome:
  name: s3

spi:
  clk_pin: GPIO12
  mosi_pin: GPIO11

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "123123"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "S3 Fallback Hotspot"
    password: "123"

color:
  - id: gray
    red: 50%
    green: 50%
    blue: 50%

font:
  - file: "fonts/iosevka.ttf"
    id: iosvetta
    size: 24

display:
  - platform: st7789v
    model: Custom
    offset_height: 0
    offset_width: 0
    height: 170
    width: 320
    backlight_pin: GPIO38
    cs_pin: GPIO6
    dc_pin: GPIO7
    reset_pin: GPIO5
    lambda: |-
      it.print(50, 50, id(iosvetta), "Hello World!");
      it.rectangle(64, 64, 192, 112, id(gray));
```

Jake-Jensen, 2023-08-11:
Issue is still present.

pierdr, 2024-03-06:
Any updates on this issue?

closed 2024-10-29 by clydebarrow