esphome/issues #3458: Updating a Device Wireless is not changing the runtime programm

majoto85, 2022-07-20:
### The problem

Hello,

If I'm changing some values and updating the device wireless, nothing is changing. Updating manually with ESP-Home-Flasher works.

For example. I have a light wich turns on after booting. I changed color to check if the new version is working and it's still the old color.

Thank you and great work,
Tobias 

### Which version of ESPHome has the issue?

2022.06.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.5

### What platform are you using?

ESP32

### Board

LaskaKit ESP-VINDRIKTNING

### Example YAML snippet

```yaml
Before Updating Wireless (LED is Blue):

  on_boot:
    then:
      - light.addressable_set:
          id: status
          range_from: 2
          range_to: 2
          color_brightness: 0.2
          red: 0%
          green: 0%
          blue: 100%

After Updating Wireless (LED is still Blue):

  on_boot:
    then:
      - light.addressable_set:
          id: status
          range_from: 2
          range_to: 2
          color_brightness: 0.2
          red: 0%
          green: 100%
          blue: 0%
```

ssieb, 2022-08-02:
Can you show logs of the wireless updating?

jaaka-se, 2022-08-05:
I have a similar problem with one of three esp32-cam and can supply a log.
the first part is from log-window in web-page "http://dolly3.local;80/"

16:14:07 | [D] | [esp32_camera:168] | Got Image: len=40996
--
16:14:16 | [D] | [ota:139] | Starting OTA Update from ::FFFF:C0A8:6D...
16:14:16 | [D] | [ota:308] | OTA in progress: 0.1%
16:14:17 | [D] | [ota:308] | OTA in progress: 24.4%
16:14:18 | [D] | [ota:308] | OTA in progress: 25.6%
16:14:19 | [D] | [ota:308] | OTA in progress: 26.0%
16:14:20 | [D] | [ota:308] | OTA in progress: 46.0%
16:14:21 | [D] | [ota:308] | OTA in progress: 70.3%
16:14:22 | [D] | [ota:308] | OTA in progress: 91.9%
16:14:24 | [I] | [ota:341] | OTA update finished!
16:14:24 | [I] | [app:133] | Rebooting safely...
16:14:24 | [D] | [esp32.preferences:114] | Saving preferences to flash...
16:15:14 | [D] | [esp32_camera:168] | Got Image: len=40424

The second part is from esphome docker 2022.6.2
INFO Generating C++ source...
INFO Compiling app...
Processing dolly3 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ESPmDNS> 1.0
|-- <ArduinoJson> 6.18.5
RAM:   [==        ]  17.6% (used 57600 bytes from 327680 bytes)
Flash: [======    ]  62.5% (used 1147026 bytes from 1835008 bytes)
========================= [SUCCESS] Took 5.16 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of dolly3.local
INFO  -> 192.168.0.101
INFO Uploading ./.esphome/build/dolly3/.pioenvs/dolly3/firmware.bin (1147136 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from dolly3.local using esphome API
WARNING Can't connect to ESPHome API for dolly3.local: Error resolving IP address: [Errno -3] Temporary failure in name resolution
INFO Trying to reconnect to dolly3.local in the background

I have a name "spelling error" on an output-switch that is corrected in the new build and after refrech the web page still shows the "old" name and it is also wrong text in the log after toggle switch on/off.

ssieb, 2022-08-05:
Can you get serial logs from the device?

closed 2022-12-19 by github-actions[bot]