esphome/issues #3462: ESPHome device webpage not updated when accessing through reverse-proxy

RoboMagus, 2022-07-23:
### The problem

I'm trying to access my ESPHome devices from outside my local network (I have to deal with multiple separate networks).
I have configured a reverse proxy to redirect trafic as such: `esphome.my.domain/devices/plug1` is redirected to `192.168.4.x` for device 'plug1'.

However, when looking into the JS that's loaded on the webpage, everything that is dynamically loaded comes from `/events`. As such it cannot deal with proxies. 😢 

### Which version of ESPHome has the issue?

2022.7.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/a

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

web_server

### Example YAML snippet

```yaml
esphome:
  name: plug1

esp8266:
  board: d1_mini
  restore_from_flash: true

wifi:
  id: wifiComponent
  ssid: !secret WiFi_ssid
  password: !secret WiFi_password
  fast_connect: on

ota:
  password: !secret ota_password

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_server_password
```

### Anything in the logs that might be useful for us?

```txt
Nope. Its not reaching the ESP device at all.
```

tefracky, 2022-09-24:
I have a similar problem and want to access the ESPHome-Device web-page over a public domain using a nginx reverse proxy. The page templated itselv is loading, but no entities or log. Is there anything new on it?

RoboMagus, 2022-09-24:
Nothing yet. I've made changes to the JS that I've successfully tested in my setup, but so far no response on the PR: esphome/esphome-webserver#9

Not sure if it's the right solution though. Someone with more insights in how the frontend is supposed to operate needs to weigh in on this.

tefracky, 2022-09-24:
Ah sorry, I didn't see the PR and it is sad, that they were/are not merged yet. Do you know if it is possible to compile it by ourself?

RoboMagus, 2022-09-25:
> Do you know if it is possible to compile it by ourself?

Yes, that is possible, as descibed by the [development](https://github.com/esphome/esphome-webserver#development) section of the readme on the esphome-webserver repo.

The resulting JS file can then be referred to as described by the [Here be Dragons](https://esphome.io/components/web_server.html#here-be-dragons) section of the webserver component of the EspHome Docs.

I've added the resulting `www.js` produced by the proposed changes in the attached zip: [www.zip](https://github.com/esphome/issues/files/9640423/www.zip).
Perhaps using the guide linked to above you could check if this solution works for your usecase and report back your findings
on PR esphome/esphome-webserver#9?

tefracky, 2022-09-25:
I use ESPHome in HomeAssistant and I am not sure how to use the `use_address` in the configuration. But however, I think it should work if I change `url = window.location.pathname;` to `url = my.domain/device;`, am I right?
What have I to use in the config-file?

```
js_include: "/config/esphome/.esphome/custom/www.js"
js_url: ""
```

Is this right?

Thanks a lot!

slovdahl, 2022-09-26:
> I have a similar problem and want to access the ESPHome-Device web-page over a public domain using a nginx reverse proxy. The page templated itselv is loading, but no entities or log. Is there anything new on it?

I also use nginx as a reverse proxy in front of ESPHome, and I had problems getting everything to work too. But I finally managed to fix all of it a few weeks ago (with a little bit of help from the browser's developer tools).

In my case, normal HTTP requests were proxied just fine, but the WebSocket connection did not work at all. My nginx listens on port 443 and is proxed to ESPHome running in Docker and bound to port 6052. This is my nginx config:

```nginx
http {
    upstream esphome {
        server 127.0.0.1:6052;
    }

    server {
        server_name <esphome.domain.com>;
        listen 443 ssl http2;

        proxy_buffering off;

        location / {
            proxy_pass http://esphome;
            proxy_set_header Host $host;
            proxy_redirect http:// https://;
            proxy_http_version 1.1;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection $connection_upgrade;
        }
    }
}
```

The problem for me was the `Host` header was wrong. For some reason I had it set to `proxy_set_header Host $host:$server_port;`. Hope this helps you out too.

closed 2022-10-10 by jesserockz